alienatedsec/yi-hack-v5 #130: 9CUS Outdoor: Freezing constantly.

wired-dev, 2021-12-09:
I'm using the 9CUS outdoor camera; I've been trying to get it stable for a few days now. I've seen it overload the swap file; I can't get snapshots to work semi reliably without disabling the RTSP server. 

Any assistance would be appreciated.

wired-dev, 2021-12-11:
heres the results I'm getting from top.

Mem: 29668K used, 1092K free, 372K shrd, 8K buff, 2988K cached
CPU:  8.3% usr 54.9% sys  0.0% nic  0.0% idle 36.0% io  0.0% irq  0.6% sirq
Load average: 49.92 31.10 15.16 5/143 7942
  PID  PPID USER     STAT   VSZ %VSZ CPU %CPU COMMAND
  964     1 root     S    26160 84.9   0 12.2 ./rmm
  469     2 root     DW       0  0.0   0  6.2 [mmcqd/1]
 1600     1 root     S     2696  8.7   0  2.2 rRTSPServer -r low -p 554
 1541     1 root     S     3968 12.8   0  2.0 ./mp4record
 1604     1 root     S     2488  8.0   0  1.8 h264grabber -r low -m yi_outdoor -f
 3407     2 root     SW       0  0.0   0  1.8 [RtmpTimerTask]
  780     1 root     D     4236 13.7   0  1.3 ./dispatch
 3408     2 root     SW       0  0.0   0  1.3 [RtmpMlmeTask]
 2787  2777 root     R     1492  4.8   0  1.0 top
 6783  6598 root     D     8964 29.1   0  0.8 imggrabber -m yi_outdoor -r high -w
  199     2 root     DW       0  0.0   0  0.8 [kswapd0]
 6305  6222 root     D    13004 42.2   0  0.7 imggrabber -m yi_outdoor -r high -w
 6821  6680 root     D     8964 29.1   0  0.7 imggrabber -m yi_outdoor -r high -w
 6820  6682 root     D     8964 29.1   0  0.7 imggrabber -m yi_outdoor -r high -w
 4994  4922 root     D     7496 24.3   0  0.7 imggrabber -m yi_outdoor -r high -w
 3421     1 root     S     1768  5.7   0  0.7 /home/base/tools/wpa_supplicant -c/tmp/wpa_supplicant.conf -g/var/run/wpa_supplicant-gl
 6304  6223 root     D    13004 42.2   0  0.6 imggrabber -m yi_outdoor -r high -w
 1559     1 root     S     1480  4.8   0  0.6 httpd -p 8080 -h /tmp/sd/yi-hack-v5/www/ -c /tmp/httpd.conf
 5506  5405 root     D    13468 43.7   0  0.5 imggrabber -m yi_outdoor -r high -w
 5401  5302 root     D    13356 43.3   0  0.5 imggrabber -m yi_outdoor -r high -w
 6114  5966 root     D    13072 42.4   0  0.5 imggrabber -m yi_outdoor -r high -w
 5292  5209 root     D    13512 43.8   0  0.4 imggrabber -m yi_outdoor -r high -w
 5566  5404 root     D    13492 43.8   0  0.4 imggrabber -m yi_outdoor -r high -w
 5581  5486 root     D    13492 43.8   0  0.4 imggrabber -m yi_outdoor -r high -w
 5392  5298 root     D    13356 43.3   0  0.4 imggrabber -m yi_outdoor -r high -w
 6191  6082 root     D    13072 42.4   0  0.4 imggrabber -m yi_outdoor -r high -w
 6110  5998 root     D    13072 42.4   0  0.4 imggrabber -m yi_outdoor -r high -w
 6192  5967 root     D    13072 42.4   0  0.4 imggrabber -m yi_outdoor -r high -w
 6469  6341 root     D     8608 27.9   0  0.4 imggrabber -m yi_outdoor -r high -w
 1581     1 root     S     1004  3.2   0  0.4 ipc_multiplexer
 7751  7738 root     D     1480  4.8   0  0.3 [snapshot.sh]
  779     1 root     S     1044  3.3   0  0.3 ./log_server
    3     2 root     SW       0  0.0   0  0.3 [ksoftirqd/0]
 7852  1559 root     S     1516  4.9   0  0.2 httpd -p 8080 -h /tmp/sd/yi-hack-v5/www/ -c /tmp/httpd.conf
 1565     1 root     S     1492  4.8   0  0.1 telnetd
  403     2 root     SW       0  0.0   0  0.1 [kworker/0:1]
  105     2 root     DW       0  0.0   0  0.1 [kusbotg]
 6664  1559 root     S     1516  4.9   0  0.0 httpd -p 8080 -h /tmp/sd/yi-hack-v5/www/ -c /tmp/httpd.conf
 6588  1559 root     S     1516  4.9   0  0.0 httpd -p 8080 -h /tmp/sd/yi-hack-v5/www/ -c /tmp/httpd.conf
 5296  1559 root     S     1516  4.9   0  0.0 httpd -p 8080 -h /tmp/sd/yi-hack-v5/www/ -c /tmp/httpd.conf
 6647  1559 root     S     1516  4.9   0  0.0 httpd -p 8080 -h /tmp/sd/yi-hack-v5/www/ -c /tmp/httpd.conf
 7888  1559 root     S     1516  4.9   0  0.0 httpd -p 8080 -h /tmp/sd/yi-hack-v5/www/ -c /tmp/httpd.conf
 5976  1559 root     S     1516  4.9   0  0.0 httpd -p 8080 -h /tmp/sd/yi-hack-v5/www/ -c /tmp/httpd.conf
 7527  1559 root     S     1516  4.9   0  0.0 [httpd]
 5448  1559 root     S     1516  4.9   0  0.0 httpd -p 8080 -h /tmp/sd/yi-hack-v5/www/ -c /tmp/httpd.conf
 6213  1559 root     S     1516  4.9   0  0.0 httpd -p 8080 -h /tmp/sd/yi-hack-v5/www/ -c /tmp/httpd.conf
 5393  1559 root     S     1516  4.9   0  0.0 httpd -p 8080 -h /tmp/sd/yi-hack-v5/www/ -c /tmp/httpd.conf
 7738  1559 root     S     1516  4.9   0  0.0 [httpd]

smnhdy, 2022-01-19:
I get the same on my 2 cameras.