alienatedsec/yi-hack-v5 #135: cannot connect to the web page (http://192.168.xxx.yyy:8080)

fred-c1, 2021-12-28:
Hello

camera = Yi Home 27US
yi-hack-v5 version = v0.3.2

Problem: I could not connect to the web page (http://192.168.xxx.yyy:8080)

Solution:
   step 1 = reset the camera (reset button on the back of the camera)
   step 2 = (re)connect the camera using the Yi Home (Android) application

Thanks!

fred-c1, 2022-01-06:
It worked a few days.
Now I cannot connect to the web page (http://192.168.xxx.yyy:8080)
but the Yi app. is able to connect to the camera.

I have tried several times to remove the hack and reinstall it.

Any idea?
Thanks

Sprocket02, 2022-01-07:
I had the same issue with my Yi Outdoor cam. Unfortunatly I had to go back to v4.

fred-c1, 2022-01-10:
@Sprocket02
Maybe some

fred-c1, 2022-01-10:
@Sprocket02
maybe some hope:
Just after my last comment, I decided to try a 2nd uSD card.
Both cards are uSD HC 32GB
With the 2nd card I didn't check the box "quick format" (=> about 1 ??? hour to format)
I have done several reboot (web page button) and power cycling.
So far the web page and the RTSP stream have always been OK.

fred-c1, 2022-01-10:
closed by mistake => reopen

fred-c1, 2022-01-21:
I installed v0.3.2 hack on another Yi Home 27US today.

It seems that there is definitely a sensitivity to the model of microSD card used:
1 model of 32GB microSD cards works
2 models of 32GB microSD cards don't works (even with a partition reduced to 4GB)
1 model of 8GB microSD cards works

4Xsample, 2022-01-28:
I found the same problem on a 47US.
I previously had installed `Yi-Hack-v4` on this cam.
I reseted the cam.
I copied the files `home_y18` and `rootfs_y18` to a clean SD card formated in Fat32 (I used several cards to try it, 8, 16, and 32 Gb).
I start the cam with the card inside.
I Followed the process from the iPhone App to conect the cam to my network and works 100% of the installations directly with the Xiaomi app but no web acces.
When I try to log in to `http://192.168.1.156:8080` I got the message from the browser where says I can't conect there.
[image]
And when I point to `http://192.168.1.156` directly with no ports I got a configuration website but with no content.
[image]
And when I try to follow any link of this website I just get no answer.
Just loading `http://192.168.1.156/?page=configurations` brings me no answer.

fred-c1, 2022-01-28:
@4Xsample :
The instructions are "Save both files and the folder on root path of microSD card."
Did you also copy the "folder" ?
This folder is important (and often forgotten)

4Xsample, 2022-01-28:
I did it a couple times but not with all the cards.
I'll try it now with the 8gb one!
Let's ser if is just this...
Thanks!

4Xsample, 2022-01-29:
Ok!
It worked for a few hours and then went off.. 
now It does the same... 
I could conect it to homebridge for couple hours and suddently the cam went offline.
It still can be accessed from yi home app but no rtsp and the website show's the header but no work (like the image I posted)

airart320, 2022-03-29:
Same issue here on the same hardware.
I can access the webinterface only with the initial FAT32 formatted SD. Once I format the card using the app - no access is available.

closed 2023-07-04 by github-actions[bot]