alienatedsec/yi-hack-v5 #137: Continuous camera_setting MQTT update

kyangk, 2021-12-29:
SW version: v0.3.2
Model: 27US

Noticed camera was continuously updating camera_setting MQTT. About one update every 1-2s.

Repro Steps
1. MQTT Camera Settings Advertise -> Retain = 1
2. Check frequency of `yicam/camera_setting`
2. Use HomeAssistant to change setting `yicam/camera_setting/LED/set`
3. See increase frequency of `yicam/camera_setting`

Workaround:
MQTT Camera Settings Advertise -> Retain = 0

kyangk, 2022-01-08:
My previous assessment was wrong. 

mosquitto_sub process conflicts with the mqttv4 + mqtt_config because they use the same client-id. Notice that as soon as I connect using mosquitto_sub, the mqttv4 + mqtt_config started connection looping.

This was causing the `status=online` to be continuously send.

```
tmp/sd #
/tmp/sd #
/tmp/sd # /tmp/sd/yi-hack-v5/bin/mosquitto_sub -i yi-cam -v -C 1 -h 192.168.1.203 -p 1883 -u yi_test2 -P DqLXZ46u22vpon6FLJpdi6Hx -t yi-cam/camera_setting
Trying to connect... Client yi-cam sending CONNECT
Client yi-cam received CONNACK (0)
Client yi-cam sending PUBLISH (d0, q1, r1, m2, 'yi-cam/status', ... (6 bytes))
. connected!
Client yi-cam received PUBACK (Mid: 2)
Trying to connect... Client yi-cam sending CONNECT
Client yi-cam received CONNACK (0)
Client yi-cam sending PUBLISH (d0, q1, r1, m3, 'yi-cam/status', ... (6 bytes))
. connected!
Client yi-cam received PUBACK (Mid: 3)
Trying to connect... Client yi-cam sending CONNECT
Client yi-cam received CONNACK (0)
Client yi-cam sending PUBLISH (d0, q1, r1, m4, 'yi-cam/status', ... (6 bytes))
. connected!
Client yi-cam received PUBACK (Mid: 4)
Trying to connect... Client yi-cam sending CONNECT
Client yi-cam received CONNACK (0)
Client yi-cam sending PUBLISH (d0, q1, r1, m5, 'yi-cam/status', ... (6 bytes))
. connected!
Client yi-cam received PUBACK (Mid: 5)
Trying to connect... Client yi-cam sending CONNECT
^C
/tmp/sd # Client yi-cam received CONNACK (0)
Client yi-cam sending PUBLISH (d0, q1, r1, m6, 'yi-cam/status', ... (6 bytes))
. connected!
Client yi-cam received PUBACK (Mid: 6)
```

As soon as I kill the sub process, the mqttv4 + mqtt_config becomes stable again.
```
/tmp/sd #
/tmp/sd #
/tmp/sd #
```

If I start mosquitto_sub using a different client id, the mqttv4 + mqtt_config is not affected.
```
/tmp/sd # /tmp/sd/yi-hack-v5/bin/mosquitto_sub -i yi-cam-sub -v -C 1 -h 192.168.1.203 -p 1883 -u yi_test2 -P DqLXZ46u22vpon6FLJpdi6Hx -t yi-cam/camera_setting
Client yi-cam sending PINGREQ
Client yi-cam received PINGRESP
```

This means `HOSTNAME` cannot be the same as `MQTT_CLIENT_ID`

kiruya1987, 2022-03-04:
Good point , thank you for the sharing

closed 2023-07-04 by github-actions[bot]