alienatedsec/yi-hack-v5 #140: SSID ?

aladin2000, 2022-01-09:
Hi
I do not have the Blue led after flashing the firmware. 

I wanted to know how to force the  wpa_supplicant  configuration 
ie SSID, password. 

regards

fred-c1, 2022-01-09:
I don't know if this is possible.
Did you try to use the Yi application (Android) to connect / provide SSID+password to the camera.

aladin2000, 2022-01-09:
No when I try to reset the cam, I hear  no "welcome sound"  after the reset.
I used to try the original firmware in the past but in that moment it does work anymore. So I wander if using the SDK, it could be possible to force the 'wpa' to reach my router ...

zachcheatham, 2022-06-05:
I'm also having this issue. Were you able to get it to work?

aladin2000, 2022-06-05:
> I'm also having this issue. Were you able to get it to work?

No it still be on my tobefix list

closed 2022-06-05 by aladin2000