alienatedsec/yi-hack-v5 #142: How to convert from MStar hack to yours?

idanny76, 2022-01-13:
Trying to flash your firmware over roleoroleo /yi-hack-MStar for my 1080P Yi Dome BFUS | 4.6.0. Copied your files to the sd card, but the firmware did not flash. Is is possible to overwite Yi-Hack from one builder to another?

idanny76, 2022-01-13:
> Did you copy over the folder from the release as well?

Yes the two files and the folder. It did not write over the current firmware.

idanny76, 2022-01-13:
The only way I can get it to read the new firmware is by renaming these files, but it will flash and then get stuck flashing. Then I have to re-flash the other firmware.
home_h201c to home_h20
sys_h201c to rootfs_h20

gelse, 2022-02-13:
i have the exact same behaviour as @idanny76

fred-c1, 2022-02-13:
Could be useful:
https://github.com/shadow-1/yi-hack-v3/issues/215
in particular:
"To REMOVE the hack, requires 2 files on the SDCard root with the following naming conventions:"

gelse, 2022-02-13:
@fred-c1 thanks, i already checked that - Yi seems to have removed the original firmware from their website.

alienatedsec, 2022-05-19:
@idanny76 this hack (v5) wont work on MStar cameras

closed 2022-05-19 by alienatedsec