alienatedsec/yi-hack-v5 #144: RTSP server video feed freezing

idanny76, 2022-01-20:
I was given an NVR and I was able to connect my Yi cameras. I have a group of Yi Dome 1080p and a single 720p. Since all the 1080p Dome were on the RoleoRoleo MStar Yi-Hack.... He was able to resolve the issue. From what I undertand he created a new RTSP server and now all the 1080p Dome are fixed. The problem I have now is that the 720p is still freezing after some time and I have to either reboot the camera or the NVR. I am attaching here the long chat I had with RoleoRoleo and how he was able to resolve it. 

Would you be willing to take a look at the conversation and see if you can implement what he did on the 720p Yi Home? Thank you
https://github.com/roleoroleo/yi-hack-MStar/issues/414

idanny76, 2022-01-23:
Roleroleo from another yi-hack was able to provide me with an updated rtsp server file for the this firmware and it works. The yo cameras now reconnect automatically when ever the live feed has issues.

https://github.com/roleoroleo/yi-hack-MStar/issues/414

fred-c1, 2022-01-30:
@alienatedsec
just in case you have some time to look into this

@idanny76: can you confirm: no more RTSP freeze on your Yi Home 27US since you are running the new RTSP sever ?

idanny76, 2022-01-30:
> @alienatedsec just in case you have some time to look into this
> 
> @idanny76: can you confirm: no more RTSP freeze on your Yi Home 27US since you are running the new RTSP sever ?

That's correct. No more Freeze/disconnects on the 27US. But I also have 1080P Dome and they all are reconnecting when ever the transmission gets interrupted with the new RTSP sever. What I have notice is that when for any reason there is an issue with the stream, it auto reconnects. Before if there was an issue with the stream, the connection would be blackout and it did not auto reconnect. I am not currently using that camera right now on the NVR, but while I had it with the new RTSP server... it was working better and reconnecting.

fred-c1, 2022-02-04:
@alienatedsec: This is what I did:
copy the file rtsp_server_yi on the microSD card (root path).
copy startup.sh in the yi-hack-v5 folder of the microSD card.
startup.sh contains:
"
killall wd_rtsp.sh
killall rRTSPServer
/tmp/sd/rtsp_server_yi -r high -a no &
"

Did I get this right?
It seems that since I have applied this patch on my Yi Home (720p) 27US, RTSP freezes are shorter.
I still get freezes that can last up to 30 seconds (mostly tested with Motioneye).
Freezes seems shorter with VLC than with Motioneye.

idanny76, 2022-02-04:
> 

Yes you have everything correct.... this part of the startup.sh "/tmp/sd/rtsp_server_yi -r high -a no &" you can edited for audio or video quality. I have mine on alaw sound, example: "/tmp/sd/rtsp_server_yi -r high -a alaw &". RoleoRoleo also told me to make sure the file startup.sh to make sure it was LF and not CFLF. I remember making that change helped.

idanny76, 2022-02-04:
> Thanks: good point. I will replace CRLF (Windows) end of line file encoding by LF (Linux)

Let me know if that helps. I remember when I edited the filed end of line it started working perfectly.

dawidmalina, 2022-02-16:
I can confirmed that latest file from https://github.com/roleoroleo/yi-hack-MStar/issues/414#issuecomment-1037953331 placed instead of rRTSPServer works very stable. Additionally I applied few changes to [system.sh](https://github.com/dawidmalina/yi-hack-v5/blob/rtsp/src/static/static/home/yi-hack-v5/script/system.sh) and [wd_rtsp.sh](https://github.com/dawidmalina/yi-hack-v5/blob/rtsp/src/static/static/home/yi-hack-v5/script/wd_rtsp.sh) in my fork that re-enable rtsp watchdog and fixes #153.

idanny76, 2022-02-18:
> I can confirmed that latest file from [roleoroleo/yi-hack-MStar#414 (comment)](https://github.com/roleoroleo/yi-hack-MStar/issues/414#issuecomment-1037953331) placed instead of rRTSPServer works very stable. Additionally I applied few changes to [system.sh](https://github.com/dawidmalina/yi-hack-v5/blob/rtsp/src/static/static/home/yi-hack-v5/script/system.sh) and [wd_rtsp.sh](https://github.com/dawidmalina/yi-hack-v5/blob/rtsp/src/static/static/home/yi-hack-v5/script/wd_rtsp.sh) in my fork that re-enable rtsp watchdog and fixes #153.

Did you rename the rtsp_server_yi to rRTSPServer and then replace the rRTSPServer file in /tmp/sd/yi-hack-v5/bin/? or Did you use the Startup.sh file to run the rtsp_server_yi? 

The two files you edited system.sh and wd_rtsp.sh were in the /home/yi-hack-v5/script/. But those two files are also in the SD card /tmp/sd/yi-hack-v5/script/. Do both need to be edited?

idanny76, 2022-02-18:
> Thanks: good point. I will replace CR+LF (Windows) end of line file encoding by LF (Linux)

Thanks to [dawidmalina](https://github.com/dawidmalina) recommendations. You can just rename the rtsp_server_yi to rRTSPServer and then rename rRTSPServer to rRTSPServer.old file in /tmp/sd/yi-hack-v5/bin/. There is no need for the startup.sh file anymore. Also make the changes to [system.sh](https://github.com/dawidmalina/yi-hack-v5/blob/rtsp/src/static/static/home/yi-hack-v5/script/system.sh) and [wd_rtsp.sh](https://github.com/dawidmalina/yi-hack-v5/blob/rtsp/src/static/static/home/yi-hack-v5/script/wd_rtsp.sh) re-enable rtsp watchdog. Now everything works mych better.

idanny76, 2022-02-23:
> Hi @idanny76 and @dawidmalina.
> 
> In which cameras are you using the rtsp_server_yi server? I have proceeded as described (backup old server, rename and upload rtsp_server_yi server and update scripts) but I get no image from the RTSP feed.
> 
> If I run the server manually through the command line and try to connect from, say, VLC, it does not say much: just that the client tries to connect and then disconnects.
> 
> I have reverted back to the original server and it is working (can not tell about stability because it has been running for just 10 minutes). We will see...
> 
> Cheers!

I am using mostly Yi Dome 1080p with RoleoRoleo Mstar v0.4.4 with the Alternate RTSP server and its working perfect. I also have two Yi Home cameras 720p and a 1080p with the V5 rtsp_server_yi RoleoRoleo provided and also edited the [system.sh](https://github.com/dawidmalina/yi-hack-v5/blob/rtsp/src/static/static/home/yi-hack-v5/script/system.sh) and [wd_rtsp.sh](https://github.com/dawidmalina/yi-hack-v5/blob/rtsp/src/static/static/home/yi-hack-v5/script/wd_rtsp.sh) and its working good, but not as good as the Yi Dome 1080. But the V5 of the rtsp_server_yi is way better than the stock RTSP server. 

Are running the Yi Outdoor camera with alienatedsec yi-hack-v5?

tonitonae, 2022-02-23:
Yes, I am running Yi Outdoor with alienatedsec yi-hack-v5.

Tried the rtsp_server_yi server provided by RoleoRoleo, but it did not work...

idanny76, 2022-02-23:
> Yes, I am running Yi Outdoor with alienatedsec yi-hack-v5.
> 
> Tried the rtsp_server_yi server provided by RoleoRoleo, but it did not work...

I am not sure on the outdoor, but the V5 cameras I have requires the installation files on the SD card. Do you still have those on the SD card?

tonitonae, 2022-02-24:
Yes, it is in the SD card, in the bin folder. It gets run, it is just that it does not accept connections, for some reason.

Anyway, I have been using the default RTPS server for 12 hours without much issue (just a few frames drop, which is reasonable given that it is a wifi outdoor camera). Let us see if it continues working.

Thanks!

alienatedsec, 2022-12-05:
Please upgrade to `0.3.6` to fix many other and similar issues

closed 2022-12-05 by alienatedsec