alienatedsec/yi-hack-v5 #147: How to automatically reboot cameras

kirilldem, 2022-01-30:
I would like to automatically reboot cameras once a week to make sure there are no glitches. I have tried to test the crontab with
 `46 20 * * * reboot now` 

[image]

After I reload the page only 46 is shown, so I don't think it is saved. Am I doing something wrong? Is there another way to reboot on schedule?

[image]

reef-actor, 2022-02-08:
Caused by issue #112 
You can change the set_configs.sh like in pending PR #113 to allow saving the full line

kirilldem, 2022-02-09:
Thanks for pointing me to the workaround.

kirilldem, 2022-03-11:
> Caused by issue #112 You can change the set_configs.sh like in pending PR #113 to allow saving the full line

How would I actually use the delimiter on the settings page?

I'd like to have this cron command `46 20 * * * reboot now`

reef-actor, 2022-03-11:
the changes in the pr fix the way the script saves text, allowing saving entries with spaces in them. so you should be able to save it just as you have typed

kirilldem, 2022-08-02:
> the changes in the pr fix the way the script saves text, allowing saving entries with spaces in them. so you should be able to save it just as you have typed

Hey reef-actor, 

I never actually managed to get this working. The PR you linked to has not been merged in the latest release right? 

As a workaround, I see that I can manually edit system.conf. I just edited it in WinSCP but the camera wouldn't boot so I assume  I need to add the cron command with inverted commas? Or perhaps it should be edited as sudo?

`CRONTAB=46 20 * * * reboot now`

Any help is greatly appreciated :)

closed 2023-06-29 by github-actions[bot]