alienatedsec/yi-hack-v5 #149: In private mode the events are not saved

eladiogalvez, 2022-02-12:
I’m trying to configure cam in private mode and when I have this options enabled the events are generated in home assistant but video files are not saved in records folder :

[image]

[image]

BytEvil, 2022-05-05:
confirmed. not worked.

alienatedsec, 2022-09-28:
@BytEvil @eladiogalvez 
Please try v0.3.3 and close this ticket if resolved

eladiogalvez, 2022-10-01:
**I'm sorry, but It's not working in last version v0.3.4 - beta.**

Once the option "Save video when a motion is detected" is enabled, when finish de current minute the next temporal file disappear and any generated event is recorded:

[image]

Date files in bash are in Zone time UTC+00:00, and name folders are UTC+02:00

kkdrz, 2022-12-04:
I have a similar issue. At first, I thought that it was because of the 'Disable Cloud', but in my case, it looks like the camera doesn't want to record video when it is in a wifi network without access to the internet. As soon as I change the network to the second one with internet access, the recording works perfectly, regardless of whether the option 'Disable Cloud' is enabled or disabled.

eladiogalvez, 2022-12-06:
@alienatedsec can you confirm where is the commit that fix it ?

alienatedsec, 2022-12-06:
@eladiogalvez 
https://github.com/alienatedsec/yi-hack-v5/commit/3df69afe417c24974bb30f177415594f47876200