alienatedsec/yi-hack-v5 #150: Optimize cron service and fix issue in cron file generation

eladiogalvez, 2022-02-12:
Possible will be better if try to start only one cron service and also when save configurations in web interface this generate duplicated schedulled tasks in file that generate stability issues.

[image]

eladiogalvez, 2022-02-12:
Possible option is in file system.sh:

**/tmp/sd/yi-hack-v5/script/system.sh**
```
echo "1 1 * * * $YI_HACK_PREFIX/script/check_update.sh" > /var/spool/cron/crontabs/root                     
if [ ! -z "$CRONTAB" ]; then                                                                                
 echo "$CRONTAB" > /var/spool/cron/crontabs/root                                                            
fi                                                                                                          
if [ "$FREE_SPACE" != "0" ]; then                                                                           
    echo "0 * * * * /tmp/sd/yi-hack-v5/script/clean_records.sh $FREE_SPACE" >> /var/spool/cron/crontabs/root
fi
...
#crond -c $YI_HACK_PREFIX/etc/crontabs                                                                      
```

eladiogalvez, 2022-02-12:
Possible can be interesting add to web disable/enable updates check's

lupo-ch, 2022-07-01:
Hello, thank you very much for this BUG report! 
This problem flooding my /var/spool/cron/crontabs/root with 20-30 and more one and the same lines caused my cameras to freeze when the  SD free space (%) is reached and the OS try to execute 20-30+ delete commands in the same second ... :( 

0 * * * * /tmp/sd/yi-hack-v5/script/clean_records.sh 1
0 * * * * /tmp/sd/yi-hack-v5/script/clean_records.sh 1
0 * * * * /tmp/sd/yi-hack-v5/script/clean_records.sh 1
....

I have fixed the problem as you suggested changing line 258 in /tmp/sd/yi-hack-v5/script/system.sh
from 
echo "0 * * * * /tmp/sd/yi-hack-v5/script/clean_records.sh $FREE_SPACE" >> /var/spool/cron/crontabs/root
to
echo "0 * * * * /tmp/sd/yi-hack-v5/script/clean_records.sh $FREE_SPACE" > /var/spool/cron/crontabs/root

Good Luck!
:) :) :)

closed 2022-12-05 by alienatedsec