alienatedsec/yi-hack-v5 #156: Can't access recorded videos in official Yi Camera app

ricardok-sp, 2022-02-21:
Installed yi-hack-v5 in five cameras at home, and since than, I can't access recorded videos on the app. Live broadcast works well, but I can not watch the SD card content. Is that an expected behavior? Does that happen to anyone else?

ricardok-sp, 2022-03-08:
The issue persists in all five cameras. I'd realy like to know if SD recordings work fine in the official app for everyone, or if it's broken just for me...

ricardok-sp, 2022-03-09:
It must be related to Issues #98 and #115

ViktorSchlaffer, 2022-03-09:
@ricardok-sp yeah, it sounds like the timezone needs to be set.
It looks like this for me:
/home/yi-hack-v5 # grep TIMEZONE /tmp/sd/yi-hack-v5/etc/system.conf
TIMEZONE=CET-1CEST,M3.5.0,M10.5.0/3

/home/yi-hack-v5 # ls -l /tmp/sd/record/ | tail -n 1
drwxr-xr-x    2 root     root         32768 Mar  9 14:04 2022Y03M09D15H

as you can see the timestamp of the file and the name of the directory are having different timestamps.

ViktorSchlaffer, 2022-03-09:
What I meant to show with my example is this:
drwxr-xr-x 2 root root 32768 Mar 9 **14**:04 2022Y03M09D**15H**

The directory timestamp is in my local timezone, the create time of the directory is in UTC.
So in my case, setting the timezone works.

ricardok-sp, 2022-03-09:
The blank timezone did the trick for me! Now I can watch the SD content, even in the up-to-date app. And the content previous to the timezone change is also accessible (although with a -3h delay). Thanks, @ViktorSchlaffer for the tip and support!

kapabel, 2022-06-29:
I had the same issue but I had a corrupt SD card. Maybe this information will help someone.

closed 2022-12-05 by alienatedsec