alienatedsec/yi-hack-v5 #157: Yi Outdoor 1080p - Flashing process doesn't work

Chapshan, 2022-02-25:
Hi,

Sorry for the noob question.
I was trying to install the yi-hack-v5 on my new yi outdoor 1080p camera.
I first configured the camera with the app and disconnected it from power.
I then created the SD card with files from this link: https://github.com/alienatedsec/yi-hack-v5/releases/download/0.3.2/yi_outdoor_0.3.2.tgz
This is how my SD card content looks like:
```
PS L:\> ls
    Directory: L:\
Mode                 LastWriteTime         Length Name
----                 -------------         ------ ----
d-----        02/25/2022     16:08                yi-hack-v5
-a----        10/19/2021     16:58        7792364 home_h30
-a----        10/19/2021     16:58        1015264 rootfs_h30
```

Then I inserted the SD card to camera and plugged it back into power.

I was expecting to see the yellow led flashing for a while, but it was flashing for only 2-3 seconds. In app, all seems to be working as usual. I have found camera's IP and tried to log in via web-browser (to <ip>:8080), but it seems this web page is not available (not that I'm so surprised, as flashing process seems to be not working properly).

I was trying the above steps with two different SD cards with different sizes (one of 64GB and the other with 4GB).

Am I missing something with the steps to flash the yi-hack-v5 firmware into the camera?

Thanks,

MichlW, 2022-02-25:
Did you checked if your Outdoor 1080p model is compatible with this yi-hack-v5 ?
There are other yi-hack custom firmware mods that supports several versions of the Yi Outdoor 1080p cams.

e.g. https://github.com/roleoroleo/yi-hack-Allwinner-v2#supported-cameras

Chapshan, 2022-02-27:
Thanks @MichlW  for the answer.

Actually, I'm completely noob with this, and don't know the answer to your question.
I was under the impression that for yi outdoor 1080p, there's only one model:
[image]

How do I check what model do I have? And how do I check which one of them the yi-hack-v5 supports?

MichlW, 2022-02-28:
You should have a look at these instructions: https://github.com/roleoroleo/yi-hack-Allwinner-v2#is-my-cam-supported

Maybe this can be tested identically with the other firmwares listed there.

iointerrupt, 2022-03-03:
If your camera is supported, try formatting the sd card first using the Yi App and then placing the two firmware binary's first and then the folder into the sd card.  I did this with an old 512mb sd card and a 4gb and it worked after multiple fails.

The app formats it as FAT32 with dos partition (not gpt) and first sector at 16.  When formatting using PCs usually the first sector is formatted at 2,048.  Im on linux and I couldnt force it to format the first sector to anything below 2048 using fdisk or any other tool.

Chapshan, 2022-03-29:
Hi,

thanks for the answers and sorry for taking me time to come back in here. I had very little time in the past month and only recently I had some time to come back to this project.

@MichlW - thanks for the link. It took me a while to figure out what is my camera model. I always looked at the box's serial, and couldn't figure out how it fits to the format mentioned on the list. Eventually, I figured out I need to look at the serial number written on the camera itself, and this aligned more with the list.
It seems my camera model is IFUSY. I see only IFUS in the list. Is that mean my camera doesn't fit to any of the projects?

Meanwhile, I tried my luck with few other tries:

1. I tried @iointerrupt suggestion, but it didn't work either (I'm not sure if this because I'm on windows system or not. Windows tend to write few system files on each new drive it sees, so if I understand correctly the intention behind your suggestion, then I can't write to the first sector as windows will always "beat me" with the race of being the first to write to it).
2. I tried using other SD card. This time a larger one (as I only have a single 4GB SD card and all the rest are larger than it). I used the instructions from yihack5 page format it correctly (creating a single 4GB partition). Again - no luck.
3. I tried re-installing my camera and then went over the previous tries. No luck with that either.

Any other suggestions (or maybe, as mentioned, my camera model doesn't fit to this hack)?

Thanks,

fred-c1, 2022-03-29:
did you try this: "try formatting the sd card first using the Yi App" ?

Chapshan, 2022-03-30:
@fred-c1, yes. Tried it. This is what I meant on #1 above, because after formatting the card with the app, I need to write the hack files into it somehow. Since I'm using a windows machine, it writes some system files into the drive once connected to the pc. So my guess is that it might be a potential root cause for the failure (i.e., those system files are written to the drive before I get the chance to write the hack files to it).
I don't have non-windows machine, so I can't try writing those files differently and can't confirm if this is the root cause or not (though I doubt it is the root cause. official guide also shows how to format SD card with windows, so process is expected to work even though eventually windows will write those system files to the drive first).

fred-c1, 2022-04-01:
OK. On my side, formatting the sd card first using the Yi App seems to help.
I used a windows 10 PC to copy the files on the SD card.

Chapshan, 2022-04-02:
@fred-c1 - Can you confirm your model starts with IFUSY, like mine?

fred-c1, 2022-04-03:
I have some yi home 720p - model 27US
and some yi home 1080p - model BFUS

about "Windows tend to write few system files on each new drive it sees": this sounds strange
unless there is a virus in your computer ????

Chapshan, 2022-04-03:
@fred-c1 - No virus. I'm referring to the 'System Volume Information' directory that windows writes to each drive it sees. This is a hidden directory, so if you're not set to see hidden files/directories in windows, you won't see this directory (but still it'll will be there):

[image]

In any case, you're having different model than I have, so it might be related to that.

Chapshan, 2022-04-06:
Sorry for bumping.
Can any one tell if IFUS also includes the IFUSY?
If not - is there other solution for IFUSY?

Thanks,

sfilinto, 2022-04-27:
same issue here. i got this working on a Yi Home, so i bought the Yi outdoor camera. unable to flash the Yi outdoor camera. I've tried out all the suggestions listed above without any luck. also used a 16Gb micro SD card to no luck.

The Serial number starts with IFUSY, however in the below link i read that its the first 4 characters which matter, so assuming its a supported model.
https://github.com/roleoroleo/yi-hack-Allwinner-v2#is-my-cam-supported

hoping someone has a solution to flash this camera.

sfilinto, 2022-04-27:
I referred to this issue https://github.com/alienatedsec/yi-hack-v5/issues/60 and accordingly flashed this https://github.com/roleoroleo/yi-hack-Allwinner-v2 firmware sucessfully.

Bust011r, 2022-12-04:
Hi, i've the same problem with the outdoor camera but my camera code start with RFUS and the flash didn't work.

Is too new and not suppoted by yi hack?

closed 2022-12-05 by alienatedsec