alienatedsec/yi-hack-v5 #159: mqtt isn't working on yi home 1080p

fidesachates, 2022-03-07:
I enabled the mqtt server on my yi home 1080p, but I can't seem to seem to connect. The server port doesn't seem open. When I ssh into the system, this is what I see. 

```
/home/yi-hack-v5 # ps -ef | fgrep mq
 1586 root      0:00 mqttv4
 1587 root      0:00 mqtt-config
 2145 root      0:00 fgrep mq
```

```/home/yi-hack-v5 # lsof | fgrep mq
1586	/tmp/sd/yi-hack-v5/bin/mqttv4	/dev/null
1586	/tmp/sd/yi-hack-v5/bin/mqttv4	/dev/console
1586	/tmp/sd/yi-hack-v5/bin/mqttv4	/dev/console
1586	/tmp/sd/yi-hack-v5/bin/mqttv4	/tmp/sd/yi-hack-v5/etc/mqttv4.conf
1586	/tmp/sd/yi-hack-v5/bin/mqttv4	socket:[2697]
1586	/tmp/sd/yi-hack-v5/bin/mqttv4	socket:[2698]
1586	/tmp/sd/yi-hack-v5/bin/mqttv4	socket:[2699]
```

After a while mqtt-config stopped showing up in the ps list. 

I didn't change any part of the config except for enable it at first; I also tried setting the ip address and a client id to no avail.

MichlW, 2022-03-08:
AFAIK the MQTT in yi-hack-v5 is a client, not a server.
You need to setup a MQTT server separately and connect to it from the MQTT client that you have activated in yi-hack-v5.

fidesachates, 2022-03-09:
Oh.... man... I assumed that it was a server because the default was 0.0.0.0 which is usually the binding address for a server. Thanks, that explains a few things.

closed 2022-03-09 by fidesachates