alienatedsec/yi-hack-v5 #162: FTP Upload not working

iointerrupt, 2022-04-07:
I have 2 Yi Home 720p cameras with the hack working well. I can view RTSP streams and get MQTT motion notifications.  I have cloud access disabled.  I have a remote FTP server set up receive videos from the cameras and enabled "FTP Upload" enabled under events but it appears nothing happens with this.  No videos are sent to the FTP server nor are stored on the SD cards on the cameras.  The cameras are 47US.

Additionally, is there a way to lower the motion off notification to anything under one minute?

joseantgv, 2022-08-02:
Any update on this?

alienatedsec, 2022-09-28:
@joseantgv @iointerrupt 
My 47US is storing videos on the SD CARD with no problem. I will look closer to FTP upload.

alienatedsec, 2022-09-28:
Working with v0.3.3
If your cameras are not saving videos, please format the card in the YI app and put back the `yi-hack-v5` folder, restart and check again.

closed 2022-09-28 by alienatedsec