alienatedsec/yi-hack-v5 #166: YI Outdoor Snapshot schedule

DEMs86, 2022-04-25:
Hello!
Is it possible to make snapshots by preconfigured scheme. For example, twice in a minute. And save this picture with timestamp in filename on SDcard. 
It's very helpful feature for making timelapse of some activity in frame. (for me it's a process of building). 

I'm trying to do this by CRON but don't know how to add date and tine to file name
0 0/5 8-17 * * ? /home/yi-hack/www/cgi-bin/snapshot.sh
For example, everyday, each five minutes from 8 to 17 hours.

reef-actor, 2022-05-06:
Perhaps you could periodically download the snapshot (http://[ip-address]/cgi-bin/snapshot.sh?res=high&watermark=yes) using [this ancient tool](https://www.nirsoft.net/utils/seqdownload.html). It seems to have the options you need (sequential file naming, configurable period and time range).

DEMs86, 2022-05-15:
Thanks! Very good tool! And it's working!
But i have one another problem with snapshot, it works only in low resolution. When i try to using "res=high" option snapshot in not opening or open only once in 10 trying. 
May be some one know what;s the problem of this?

DEMs86, 2022-05-15:
```
/tmp/sd/yi-hack-v5 # imggrabber -m yi_outdoor -r high --debug
Debug on
Starting program
Resolution high
1652611663500 - mapping file /tmp/view, size 1586752, to b6d8b000
1652611663518 - closing the file /tmp/view
1652611663519 - found latest frame: id 118, frame_counter 57281
1652611663519 - processing frame 118
1652611663556 - processing frame 118
1652611663582 - processing frame 118
1652611663606 - processing frame 118
1652611663628 - processing frame 118
1652611663646 - processing frame 119
1652611663666 - processing frame 119
1652611663686 - processing frame 120
1652611663706 - processing frame 120
1652611663728 - processing frame 120
1652611663747 - processing frame 121
1652611663766 - processing frame 121
1652611663786 - processing frame 122
1652611663806 - processing frame 122
1652611663827 - processing frame 122
1652611663847 - processing frame 123
1652611663882 - processing frame 123
1652611663906 - processing frame 124
1652611663928 - processing frame 124
1652611663947 - processing frame 125
1652611663966 - processing frame 125
1652611663986 - processing frame 126
1652611664008 - processing frame 126
1652611664029 - processing frame 126
1652611664046 - processing frame 127
1652611664066 - processing frame 127
1652611664086 - processing frame 128
1652611664086 - writing frame: frame_offset 491844, frame_ptr b6e0c784, frame_length 37
1652611664106 - processing frame 129
1652611664106 - writing frame: frame_offset 491888, frame_ptr b6e0c7b0, frame_length 8
1652611664129 - processing frame 130
1652611664129 - writing frame: frame_offset 491900, frame_ptr b6e0c7bc, frame_length 162870
7411223767353936 - frame found, exit loop
Decoding h264 frame
Starting decode
Decode frame
Writing yuv buffer
Killed
/tmp/sd/yi-hack-v5 # imggrabber -m yi_outdoor -r high --debug
Debug on
Starting program
Resolution high
1652614471139 - mapping file /tmp/view, size 1586752, to b6cfc000
1652614471140 - closing the file /tmp/view
1652614471140 - found latest frame: id 106, frame_counter 50233
1652614471141 - processing frame 106
1652614471163 - processing frame 106
1652614471187 - processing frame 106
1652614471204 - processing frame 106
1652614471223 - processing frame 106
1652614471243 - processing frame 107
1652614471264 - processing frame 107
1652614471285 - processing frame 107
1652614471303 - processing frame 108
1652614471323 - processing frame 108
1652614471343 - processing frame 109
1652614471364 - processing frame 109
1652614471385 - processing frame 109
1652614471403 - processing frame 110
1652614471424 - processing frame 110
1652614471444 - processing frame 111
1652614471464 - processing frame 111
1652614471484 - processing frame 111
1652614471515 - processing frame 112
1652614471535 - processing frame 112
1652614471553 - processing frame 113
1652614471573 - processing frame 113
1652614471593 - processing frame 114
1652614471614 - processing frame 114
1652614471634 - processing frame 114
1652614471654 - processing frame 115
1652614471673 - processing frame 115
1652614471693 - processing frame 116
1652614471713 - processing frame 116
1652614471735 - processing frame 116
1652614471754 - processing frame 117
1652614471773 - processing frame 117
1652614471794 - processing frame 118
1652614471814 - processing frame 118
1652614471835 - processing frame 118
1652614471853 - processing frame 119
1652614471873 - processing frame 119
1652614471899 - processing frame 120
1652614471914 - processing frame 120
1652614471935 - processing frame 120
1652614471953 - processing frame 121
1652614471973 - processing frame 121
1652614471994 - processing frame 122
1652614472014 - processing frame 122
1652614472034 - processing frame 122
1652614472054 - processing frame 123
1652614472073 - processing frame 123
1652614472093 - processing frame 124
1652614472094 - writing frame: frame_offset 741852, frame_ptr b6dba81c, frame_length 37
1652614472114 - processing frame 125
1652614472115 - writing frame: frame_offset 741896, frame_ptr b6dba848, frame_length 8
1652614472135 - processing frame 126
1652614472135 - writing frame: frame_offset 741908, frame_ptr b6dba854, frame_length 127078
7411223767353936 - frame found, exit loop
Decoding h264 frame
Starting decode
Decode frame
Writing yuv buffer
Killed
/tmp/sd/yi-hack-v5 #
```

DEMs86, 2022-05-15:
@alienatedsec may be you can help with [this ](https://github.com/alienatedsec/yi-hack-v5/issues/166#issuecomment-1126917352)problem?

WalkingDot, 2022-05-18:
> For example, everyday, each five minutes from 8 to 17 hours.

these cron options should work:
`*/5 8-17 * * * imggrabber -m yi_outdoor -r low > /tmp/sd/$(date +"%Y%m%d%H%M").jpg`

closed 2022-12-05 by alienatedsec