alienatedsec/yi-hack-v5 #179: Yi 1080p Home 48US system.conf empty after reboot?

MACscr, 2022-06-17:
I have two of these webcams, but the second one doesnt appear to keep the system.conf settings after reboot and the end result is a blank system.conf. Both were bought at the same time and came as a pair. First one works fine, but the second one has this issue. Any suggestions? Other configs seem to be saving ok and retaining their settings.

MACscr, 2022-06-17:
hmm, i lied. Other settings arent saving. weird on this particular camera either.

MACscr, 2022-07-08:
Weird. The files are even getting reset to stock even if there isnt a read only issue. I can verify through ssh that the hostname file at /tmp/sd/yi-hack-v5/etc/hostname has the correct hostname, but after reboot, it and other files like TZ are completely blank. system.conf is back to stock.

alienatedsec, 2022-12-05:
I got the same camera `48US` and it does not have any issues

@MACscr please start from scratch - or maybe you already fixed the problem

closed 2022-12-05 by alienatedsec