alienatedsec/yi-hack-v5 #180: No sound in RTSP stream and playback in Yi Home app no longer working

kapabel, 2022-06-17:
Hello!
The reason why I flashed the firmware of my YI Outdoor 1080P is to get the RTSP stream out so I can integrate the stream in my Homebridge. However, I do like to use also the Yi Home app for easy playback from the SD card. I do not use their cloud service.

I have noticed that there is no sound included in the RTSP stream and also that the playback function in the Yi Home App is no longer working. I can see there is recording but when trying to play back it will snap back to the end with the current time stamp.

Is this normal behavior? Or are there some setting to tweak on my end?

Thanks!

kapabel, 2022-06-28:
Edit: I figured out what the problem was with the yi home app (corrupt sd card). But I still can not get the sound to work. It does work in the yi home app but I can not get it on the rtsp stream :(

kapabel, 2022-07-08:
Thanks. I have now seen that there is another issue with using the original app. As soon as the card is full it is supposed to delete the oldest recording to make some space. But this is not happening.. let’s see what I can find out!

alienatedsec, 2022-11-25:
Audio will be in the next release - the branch is already there 
#74 
#116

closed 2022-12-05 by alienatedsec