alienatedsec/yi-hack-v5 #186: Change WiFI SSID does not work

Cisor, 2022-07-31:
Steps to reproduce issue:

- http://camaddress:8080/?page=wifi
- Select network from "WiFi network ESSID"
- Enter password 
- Enter password confirmation
- Open browser console to Network tab
- Save
- http://172.16.0.200:8080/?page=reboot
- "Reboot Camera"

Behaviour:
- "Saved" on wifi page
- Camera returns - 
```
{
"error":"false"
}
```
- Camera reboots attached to previous network

Expected:
- Camera reboots attached to new network

alienatedsec, 2022-09-17:
@Cisor Which camera model do you have?

Cisor, 2022-09-17:
I knew I forgot something!
It is a Yi Dome

alienatedsec, 2022-12-06:
There have been a few changes since you raised this issue. Please can you try `0.3.7`?

Cisor, 2022-12-07:
I appreciate the follow up, thanks. I will test this, although it's not likely to be within a fortnight

closed 2023-06-26 by github-actions[bot]