alienatedsec/yi-hack-v5 #193: [Feature Request] Updating wifi credentials via file on SD card root

iointerrupt, 2022-09-01:
Current I believe the ability to change the wifi credentials exists via by either shelling into the camera via SSH/telnet or web ui.

Would it be possible to update the startup script to include a check for a "configure_wifi.cfg" on the root of the sd card or somewhere on the sd card to allow changing the password that way.  Obviously the script would also delete the configure_wifi.cfg afterwards to prevent a loop.

Justification: If we have to revert to a bricked firmware, sometimes the cams have to be reset to stock and yi home cam app has to be used to register the cam.  My cams usually are on a locked down vlan with no internet access but to register, they have to have internet access to I have to first put them on my main network, register, updating firmware to yi-hack, login to yi-hack and change the wifi creds.  Being able to do it via a file on the sd card would automate that process.  Just a thought.

alienatedsec, 2022-09-17:
@iointerrupt thanks - noted

alienatedsec, 2022-09-28:
@iointerrupt 
Can you reference any solutions already available?

iointerrupt, 2022-09-29:
> @iointerrupt Can you reference any solutions already available?

The original post above shows the piece of code added to the system.sh that could be a solution.

alienatedsec, 2022-09-29:
@iointerrupt done

The format needs to be as follows:
`wifi_ssid=[your_wifi_essid]`
`wifi_psk=[your_wifi_password]`

I will release another version soon

alienatedsec, 2022-09-29:
Done v0.3.4 released

closed 2022-11-29 by alienatedsec

alienatedsec, 2024-02-19:
This issue continues in #350 and describes the problem in more detail. 

In short: 
- the template file [configure_wifi.zip](https://github.com/alienatedsec/yi-hack-v5/files/14325076/configure_wifi.zip) needs to be edited with your WiFi details in the format as is, and no `Enter's` or lines should be added or removed.
- copy over to your SD card `/recover` folder - this is `/tmp/sd/recover` when in the console on your camera
- reboot