alienatedsec/yi-hack-v5 #195: Help with identifying the camera

NiiiYaa, 2022-09-16:
Help with identifying the camera
I wish to apply the right 'hack' so I need to identify the camera.
The camera was bought on AliExpress - "no brand".
Works with the "YI IOT" app.
On GoogleHome identifies itself as: "xiaoyi".
Firmware: 6.0.05. 10_202112100938
ID: "A1769004DDT2Q6220729"

Motherboard:
Looks like this:
image
[image]

Chip 1:
ANYKA
AK3918EN080
V200
CDSJ21F22

Chip 2:
SV6256P
TAC2204
IR498B5
-- I might be 1
-- B might be 8

Chip3:
OK291323
ULN2803VS
-- 8 might be B

Camera looks like this:
[image]

haps-basset, 2022-09-24:
Hello
I have got a similar camera as yours and I would like to investigate the internals of it 
Could you describe the dissembling procedure of your camera?
Haps

haps-basset, 2022-09-30:
The camera shown in the picture is a new design based on YI IoT software and ANYKA AK3918 SoC. 
I have just hacked access to the camera via Wi-Fi but because the camera does not Xiaomi Hi3518ev200 Chipset, thus I am not sure that you would be interested in the tarball of the camera system files. Please let me know 
More information about AK3918 development, you can find in the following git repositories:
•	ricardojlrufino/anyka_v380ipcam_experiments
•	mucephi/anyka_ak3918_kernel
•	ricardojlrufino/arm-anykav200-crosstool

closed 2022-11-29 by alienatedsec

rleyden559, 2023-03-01:
I have a very similar IP Cam.  The layout of the PCB is similar but has a different rev # and date.  I had been able to access it by FTP and telnet.  Telnet required a simple hack,  
Startup text:
 Linux version 3.4.35 (zhaofeihong@dell-PowerEdge-R740) (gcc version 4.8.5 (anyka (gcc-4.8.5 + binutils-2.24 + ulcibc-0.9.33.2)(20170223)) ) #1 Mon Aug 1 19:43:47 CST 2022
CPU: ARM926EJ-S [41069265] revision 5 (ARMv5TEJ), cr=00053177.

The flash is read only except for a small jffs2 partition that stores  local WIFI settings, etc. 
I backed up the partitions to a SD card.  But, I'm unsure if I can change U-Boot from telnet to boot from the SD card, thereby making the file system R/W.  There is an update script but I'm not sure how to exploit it.
The PCB does not have UART RX and TX labeled.   UART seems the standard why of accessing U-Boot.
I'd be happy to share what I have and would be interested in any tips.  My main goals is to stream to a web browser instead of the Android app.
[image]

haps-basset, 2023-03-01:
I have also hacked into the camera and I am also interested to use open protocols onvif, rtps  etc.  I built for the camera toochain, SDK etc.  I was able to recompile all yi-hacks-v5 apps but..  but I struggle with the camera video kernel driver etc. If you are interested I am happy to share my "hacks". 
Perhaps we should start a new project?

filder35, 2023-03-28:
Hello. During the firmware, the electricity was turned off. The firmware is no longer running. Connected by UART. I tried to download the firmware from the same equipment. Flash memory was soldered and flashed on the progromator. The launch is made by the bootloader loads the kernel but then goes into reboot. Something is missing. I'm trying to build my own firmware.
[image]
[image]
[image]
[image]
[image]

filder35, 2023-03-28:
Need a dump from a similar device to build your firmware

aaronm120, 2023-06-12:
Was anyone ever able to find out anymore on this camera, if custom firmware can be used?

filder35, 2023-06-12:
here we try
https://github.com/e27-camera-hack/E27-Camera-Hack/discussions/1#discussioncomment-6106066

Bigfatalex, 2023-08-13:
Hey, did you geht IT running?

d0gleg, 2023-08-22:
Hello, I've got another variant of the Anyka AK3918 board. Would anybody know if there's an open source firmware out there for this board.

[image]
[image]

filder35, 2023-08-22:
https://github.com/e27-camera-hack/E27-Camera-Hack/discussions/1#discussioncomment-6106066

filder35, 2023-08-22:
[d0gleg](https://github.com/d0gleg)
Is this board running?
 is it possible to take a dump?
Here we continue.
https://github.com/e27-camera-hack/E27-Camera-Hack/discussions/1#discussioncomment-6106066

d0gleg, 2023-08-22:
yeh the board is running. I'm quite new at this. Can you tell me how to get a firmware dump?

filder35, 2023-08-22:
Here is a good blog for beginners with instructions. Read will ask questions.
https://gitea.raspiweb.com:2053/Gerge/Anyka_ak3918_hacking_journey

AILIFE4798, 2023-08-24:
> [d0gleg](https://github.com/d0gleg) Is this board running? is it possible to take a dump? Here we continue. [e27-camera-hack/E27-Camera-Hack#1 (reply in thread)](https://github.com/e27-camera-hack/E27-Camera-Hack/discussions/1#discussioncomment-6106066)

I have a very different camera but with same chipset and are willing to take a full dump if u want it I have ch341a so I can get the best out of it
Openipc support for this soc is still bad so I'm here
[image]

filder35, 2023-08-24:
[AILIFE4798](https://github.com/AILIFE4798)
Processor versions are different v200 and v300.
in such subtleties, be careful you can ruin the camera.
[d0gleg](https://github.com/d0gleg)
You have probably studied the materials on the links above. Everything you do with the camera you do at your own peril and risk. Read the instructions very carefully. Wrong actions can kill the camera. And for the restoration will need special equipment. That's why I wrote, first you need to take a dump from the working camera so that in case of failure you can restore the original state of the camera.

filder35, 2023-08-24:
For what. The terminal mode is convenient for debugging, diagnosing work processes and loading single-board devices. An error in the bootloader, an error when initializing the drivers, errors in the operation of the equipment, or you just need to conveniently configure something or quickly check it on a single-board device while working from your work computer - in all these situations it is convenient to connect in the so-called terminal mode, which is both a full-fledged console for executing commands and a standard output port for all information about the processor. Of course, this is convenient to check the correct operation of the device, to check how the download process is going at all its stages, etc. and so on. This is not just useful, but practically vital and is a must-have tool for all system developers of the so-called embedded system.
To get started, we need the PuTTY program and a USB to TTL converter with a cable.
USB to TTL converter popular on two chips CH340G and PL2303HX
How to install drivers This information on the Internet is enough.
If all the steps for installing the driver are performed correctly, then in the “Device Manager” after connecting the converter to the computer (via USB port), your converter will appear in the Ports (COM and LPT) group.
UART has two main data lines
RXD (RX) - receiving line
TXD (TX) - transmission line
UART device connected crosswise
We go to the “Device Manager” and determine which COM port our converter has
Mine is COM3, you may have a different port number.
We launch the PuTTY program and in the window that opens, select “Serial” in the “Serial line” field, enter our port, in my case it is “COM3”, and in the “Speed” field we enter the connection speed “115200” TTL converter.
We press the “Open” button and a black terminal window will open, at the moment it will be empty without messages and should not pop up a connection error. Now we can power up. I just turn on the device.

AILIFE4798, 2023-08-24:
i know the basics of uart debug i came from openwrt so what do i need to provide to get a custom firmware
i have nothing to lose really i dont even know what app this camera use and have no intrest using it i want just a stream url that i can add to home assistant and maybe control the motor with mqtt will be very helpful
i have alr desoldered the flash to take a full dump after this ill put it back on and get the bootlog

AILIFE4798, 2023-08-24:
heres the full dump

[AK3918EV300_full_dump.zip](https://github.com/alienatedsec/yi-hack-v5/files/12425047/AK3918EV300_full_dump.zip)

d0gleg, 2023-08-24:
I noticed on my board that there's 3 solder pads just above the AK3918 chip. I'll try to wire a USB serial port to these pins. Can I download the firmware using Putty?

d0gleg, 2023-08-24:
I don't think I'm brave enough to desolder the flash

AILIFE4798, 2023-08-24:
> I don't think I'm brave enough to desolder the flash

its safer to do it that way
yes u can dump it with uboot if u r camera have ethernet but its definately easier to do with a flash programmere

AILIFE4798, 2023-08-24:
my bootlog
[ak3918ev300_bootlog.txt](https://github.com/alienatedsec/yi-hack-v5/files/12425155/ak3918ev300_bootlog.txt)

AILIFE4798, 2023-08-24:
this version of uboot dont seem to support boot from sd card like other camera soc uboot dose so i think i either have to use tftp which i really dont want or write the firmware with flash programmer again

AILIFE4798, 2023-08-24:
[image]
[image]
[image]

filder35, 2023-08-24:
We are also trying to bring the stream to the network here:
https://github.com/e27-camera-hack/E27-Camera-Hack/discussions/1#discussioncomment-6807171
And for your v300 version, there is this:
https://github.com/helloworld-spec/qiwen/tree/main/anycloud39ev300
I'm also studying the sources, I'm trying to collect the firmware, but so far to no avail.

filder35, 2023-08-24:
So far for my camera I'm trying this:
https://github.com/jhusak/Lamobo-D1-stabilityfixes.git

AILIFE4798, 2023-08-24:
> We are also trying to bring the stream to the network here: [e27-camera-hack/E27-Camera-Hack#1 (reply in thread)](https://github.com/e27-camera-hack/E27-Camera-Hack/discussions/1#discussioncomment-6807171) And for your v300 version, there is this: https://github.com/helloworld-spec/qiwen/tree/main/anycloud39ev300 I'm also studying the sources, I'm trying to collect the firmware, but so far to no avail.

even as someone who can read chinese i still dunno what im looking at
im probably not skilled enough to try develop my own firmware i cant even get another camera with premade firmware working(hi3518)
openipc dose have a firmware for this soc(ev300)but im sure it cant go smoothly either the image sensor not supported or wifi not work u name it and so i better leave these for the professional i ll provide as much info as i can

d0gleg, 2023-08-24:
OK, I'm happy to move over to 
https://github.com/e27-camera-hack/E27-Camera-Hack/discussions/1#discussioncomment-6807171

My firmware is: 6.0.05.10_202301061607, I can try to extract it on the weekend.

filder35, 2023-08-24:
[AILIFE4798](https://github.com/AILIFE4798)
Is your camera working? Or not? If it works, then you can try to disable the extra and leave ONVIF / RSTP for the network. If not, then try to restore it. I'm not an expert either, but I learned a lot during my time with the camera. I'll disassemble your firmware and lay out how I did it, try it yourself. For v380 there is a lot of information on the net.
Just need to look at the version of the processor I have v200
here is my download log you can compare
[dump_v380_log_mod.txt](https://github.com/alienatedsec/yi-hack-v5/files/12425492/dump_v380_log_mod.txt)
.

filder35, 2023-08-24:
[AILIFE4798](https://github.com/AILIFE4798)
here is your firmware unpacked
if you have any questions ask
[out_v380_v300.zip](https://github.com/alienatedsec/yi-hack-v5/files/12425700/out_v380_v300.zip)

filder35, 2023-08-24:
you telnet ftp without root
admin password:
username=admin
Password=vip1128
it's probably a ro network connection

AILIFE4798, 2023-08-24:
i will check later i think the camera is working but because of no app idk how can i connect wifi so i havent tested yet but i can use serial or ethernet to ssh

filder35, 2023-08-24:
uart connection settings everything in the jffs2 section can be edited and will add the launch of an ftp server

filder35, 2023-08-24:
by uart
username=root
Password=
and command output
ps
dmesg
ifchonfig
netstat

AILIFE4798, 2023-08-25:
> you telnet ftp without root admin password: username=admin Password=vip1128 it's probably a ro network connection

i have tried to connect ethernet to the camera and it did get a ip address but i have done a port scan with fing but it have no open port i tried to ssh anyways and ofc it didnt work
it seems to try connect to 112.124.12.122 but the server is down which is good
heres a bootlog including the ethernet part
[ak3918ev300_bootlog_ethernet.txt](https://github.com/alienatedsec/yi-hack-v5/files/12434730/ak3918ev300_bootlog_ethernet.txt)

filder35, 2023-08-25:
The camera seems to be working. Try to connect through the app on your phone. The application must have settings for connecting home wifi. v380 is called

AILIFE4798, 2023-08-25:
> The camera seems to be working. Try to connect through the app on your phone. The application must have settings for connecting home wifi. v380 is called

i have tried to use the app com.macrovideo.v380 but it keep saying the app need root acess even when i have gave it it still wont proceed and why would ip cam viewer need root its quite weirdnone the less
and ofc for long term im gonna need to find a way to remove depandancy from the app anyways
if just need to connect wifi i think its possible to do with serial command or just use ethernet and call it a day yesterday i keep working on the other camera but lack of ethernet really make life allot harder then it needs to be so back to this camera

filder35, 2023-08-25:
try :  /mnt/mtd/wificonf/wifi_softap start
  let's see what happens should start the wifi access point and you need to connect to it. If not, I will later throw off the commands as I manually connected the wifi.

AILIFE4798, 2023-08-25:
the script worked and created a ap with ssid "MV39595680"and no password
but im unable to connect to it

filder35, 2023-08-25:
according to the idea without a password, you must connect. What does it show in the log?

filder35, 2023-08-25:
Does the phone see a new network? MV39595680

filder35, 2023-08-25:
command ifconfig what shows

filder35, 2023-08-25:
try to connect through the app v380

AILIFE4798, 2023-08-25:
> according to the idea without a password, you must connect. What does it show in the log?

[image]

> Does the phone see a new network? MV39595680

yes but only for a few minute before it disappear and i have to press reset button on camera
while its on i have done intranet scan and no open port

> command ifconfig what shows

[image]

> try to connect through the app v380

id imagine itll work but the app just dont work for me

 
> i have tried to use the app com.macrovideo.v380 but it keep saying the app need root acess even when i have gave it it still wont proceed and why would ip cam viewer need root its quite weirdnone [...]

filder35, 2023-08-25:
i ran like this
wpa_supplicant -B -c/mnt/mtd/wificonf/wpa_supplicant.conf -iwlan0
wpa_cli
>add_network
> set_network 0 ssid "MYSSID" -- your ssid
> set_network 0 key_mgmt WPA-PSK
> set_network 0 pairwise CCMP
> set_network 0 psk "passphrase" - - your network password
> enable_network 0
quit
The only thing I changed was the wpa_supplicant.conf file with my own.
Try it might run.
Here is some more information for your reference:
https://wiki.archlinux.org/title/wpa_supplicant

AILIFE4798, 2023-08-25:
do u know how to disable debug message cuz it keep sending error every second i cant use the cmd properly

haps-basset, 2023-08-26:
Assuming that your "debug" uses a stderr pipe then redirect output to null device. e.g.
some_cmd 2>/dev/null  
Note: Some experts pump debug messages via stderr and stdout so if your "debug" messages are printed via stdout then this method will not help you.
You could redirect all outputs to null but then you will see nothing     > /dev/null 2>&1 
More info about redirections can be found in Linux "cmd"  apps called sh, bash man pages.

d0gleg, 2023-08-26:
Hi, I'm running a camera with AK3918E V200 chip and I've set up the console via a USB serial port. I've got the same issue with output to the console making it difficult to use the console. I was trying redirection using "command 2> /dev/null" when I saw your message, but it's not working for me. I also tried to stop things by killing various processes, but no luck.

haps-basset, 2023-08-26:
Ok, can you be more precise about what "don't work for me"?   I wrote that redirection stderr to null works only when the designer of the app understands Linux principles and will not s... debug messages to stdout. If they  mix and send debug messages to stdout and stderr then you have got the following options:
1) redirect stdout and stderr to null  cmd  > /dev/null 2>&1   but then YOU WILL  see nothing 
2) Send an email to developers asking them to redesign the code so that error or debug messages will be only printed when some argument/key will be used to switch mode typically for Unix is -v or -d. Good luck to you if they will implement your request  :)

d0gleg, 2023-08-26:
Yeh, I was trying to redirect stderr to /dev/null when I saw your message. Just a coincidence :)
I was hoping that if I type "command 2> /dev/null" at the command prompt then it would work. I'm not sure if I'm using the correct syntax at the command prompt. I was just reporting that it was not working for me. I understand and agree with what you had said.

My linux days are long time over. I at best have dabbled with linux, and unix before that :)

haps-basset, 2023-08-26:
I am surprised because the concept of pipe is not new. It was used in VMS, Unix even Windows/DOS so I am surprised that it does not work.   If you redirect the stderr snd stdout  as follows cmd  > /dev/null 2>&1  then you will not see any messages.
Just to verify, what I am saying. I have created the script sending every second hello message to std and err pipes. Then It sent it to the background and the messages popup even prompt was ready to accept the new command.  A repeated the script piping to null and nothing was on the screen.
I am guessing  that you are thinking about redirection std/err of running images, yes You can but is a little more complex
https://stackoverflow.com/questions/593724/redirect-stderr-stdout-of-a-process-after-its-been-started-using-command-lin
Next time please be a little precise about what you want to achieve because many who will try to help don't know your level of Unix knowledge.

haps-basset, 2023-08-26:
I need your help, I bought a new camera that supports ONVIF and RTPS services. I would like to dump flash partitions to a file in order to share with you but the camera uses telnet (which is enabled)  with username and password. I have tried to use Hydra to crack the password using default usernames and passwords downloaded from the Internet but after 2 days  Hydra gave up. Obviously, I could use "a hammer or stone" approach to open it and solder USB UART but  I am not sure that concept makes any sense. 
Anyka camera init script has commented out log-in via getty and as such you can connect via UART otherwise I am s.....
See source code in rcS 
#ttySAK0::respawn:/sbin/getty -L ttySAK0 115200 vt100 # GENERIC_SERIAL      #v
ttySAK0::respawn:-/bin/sh  
The camera uses CareCamPro software. The sticker says that the username and password is admin/admin123456  but those credentials are valid for ONVIFS, not generic Telenet login.

d0gleg, 2023-08-26:
Thanks haps-basset, noted. Yeh, I also noticed that the command prompt had disappeared. I've used redirection very rarely over the years. I was hoping for a quick and nasty way of removing the messages from the console. I had the impression that AILIFE4798 had a similar issue to me. The other option I was hoping to explore was to kill off some of the processes.

BTW I've got a lot to learn about internet etiquette. Thanks for your kind words. I see you mentioned VMS. In case you're talking about the old VAX computers, I worked on VAXs on and off for 20 years. Before that I spent a few years on Gould minicomputers.

d0gleg, 2023-08-26:
I see your last message. In case you're talking to me, I'll have a look at what I can find out. But at my rate somebody else may (hopefully) jump in....

haps-basset, 2023-08-26:
I love VMS, It was designed by engineers for engineers, with a limited number of well-documented commends, not Unix BS but that is history. 
Sorry for the misunderstanding, I addressed my previous post to all members, readers etc. hoping that we would be able to crack the b.. and get a workable solution. I have installed a few of those cameras in-house and they work well but I don't like Chinese spyware  :(.
I am not going to force or tell you what to do but consider usage strace for browsing what's happened during an app execution. You will be able to find what the app uses during the run e.g. drivers, memory usage etc.
Based on my studies of some  SDKs parts because I was never able to find a full set of SDKs, I end-up with some primitive picture of our camera architecture. 
Since lsmod shows installed kernel modules, I am guessing that one is for the control sensor; frame oriented and the second to stream frames.  In a nutshell, the stream-oriented driver pushes the frames and the "sensor" driver allows to pull frames from the sensor by applications.   I am sorry that I am telling you obvious BS  because as a VMS dude, you know what is pull/push device :)
My problem is that I don't know how to initiate or use the bloody stream device. Having this component under control we would be able via v4l mechanism connect the camera's sensor stream to any applications. I am not keen to replicate or develop a new streaming device that would not be compatible with most media apps. 
Anyway, I am so happy that I met another DEC head although some spelled it in a different way :)

AILIFE4798, 2023-08-26:
> I had the impression that AILIFE4798 had a similar issue to me. The other option I was hoping to explore was to kill off some of the processes.
yes i do.and yes you can kill the main app to stop the message but for someone using windows im not skilled enough to do so
ill try the redirection later and hope for the best

haps-basset, 2023-08-26:
Just a few hints:
1) To find what is running use ps . Unfortunately, Anyka ps is very limited. 
    An alternative is top, but I am not sure if the top is part of the release or I compiled it 4 myself.
2) to kill an application you can use the following command  kill -SIGNAL app ID, 
   where id  is the id of the app that you found using ps , signal use -9
The alternative is pkill  name of the app
Note: Some apps for example anyka_ipc spawn a sub-process with watchdog which will re-spawn it  so it is not that easy to kill it if you do not kill them in particular

d0gleg, 2023-08-26:
I use "ps -aef" out of habit. It'll show you a little more than "ps". Yeh, I spent a good hour randomly killing processes with all sorts of unpredictable results including respawning.

[ps -aef.txt](https://github.com/alienatedsec/yi-hack-v5/files/12445102/ps.-aef.txt)

d0gleg, 2023-08-26:
I might have already killed off a couple of processes in that listing

[198 more replies]