alienatedsec/yi-hack-v5 #196: confusion/clarification regarding modified firmware

SimonMcN, 2022-10-06:
Hi,
My cameras seem to reflash on every single reboot.  I would have thought this was suboptimal, as you're just overwriting a known position ? (once hacked, always hacked ?)

Is it possible to get the firmware files moved to a backup directory post flash to prevent wear and tear ?

Thx
Simon

alienatedsec, 2022-10-06:
That should not be the case, and the hack does not drive it. I suggest deleting the root and home files to prevent the loop you described.

SimonMcN, 2022-10-06:
Thanks for responding.  Any idea why it would occur please ?  I've been doing a lot of wiping microSD's flashing and testing, and they always seem to get left there ?  The microsd are known good, high quality and experiencing no issues ?
As an aside, I have been manually moving the files to a firmware folder.

Would it be worth doing a check on start up and moving them just in case ?

SimonMcN, 2022-10-06:
The camera I have running the [yi-hack-Allwinner](https://github.com/roleoroleo/yi-hack-Allwinner) firmware renames it to .done ?

SimonMcN, 2022-10-06:
> > The camera I have running the [yi-hack-Allwinner](https://github.com/roleoroleo/yi-hack-Allwinner) firmware renames it to .done ?
> 
> That is the wrong place to ask. Over here, we have nothing to do with the Allwinner chipset.

I understand that.  Apologies for my miscommunication.  My inference, or query was more along the lines of, if that project renames the file, then perhaps it is because they have seen the same thing occur ?

closed 2023-06-24 by github-actions[bot]