alienatedsec/yi-hack-v5 #200: Yi Dome 1080P 15US - recording broken on 0.3.4

maciekn, 2022-10-28:
I've noticed that recording files on SD card are broken after using 0.3.4 firmware version- some files are missing (like they were overwritten).
Using 0.3.3 works like a charm

alienatedsec, 2022-10-29:
More details are needed, @maciekn - is it that you cannot see those files through the web or on the SD card? Have you tried starting from scratch? The only thing on my mind now is the timezone, as I had similar issues during testing.

Regardless, a proper timezone setting can resolve your issue, but I am unsure how it will affect historical files.

young747, 2022-11-19:
Perhaps the cause is that two "mp4record" are running.
look at ps tree.
kill one of  two "mp4record".

alienatedsec, 2022-11-28:
> Perhaps the cause is that two "mp4record" are running. look at ps tree. kill one of the two "mp4record". I wonder why the two are runned.

@young747 you are right - It looks like its something to do with the `privacy` - I missed this commit from @roleoroleo #commit ae5333ca13cd148adb9d3b3e2e86513557129ade

alienatedsec, 2022-12-05:
> Perhaps the cause is that two "mp4record" are running. look at ps tree. kill one of the two "mp4record". I wonder why the two are runned.

I have done some testing, and it still happens. The easiest is to kill one of the processes as suggested above.

[image]

I still need to find the reason for it.

alienatedsec, 2022-12-05:
fixed in `0.3.5`

closed 2022-12-05 by alienatedsec