alienatedsec/yi-hack-v5 #202: High CPU usage on `rmm` with slow responsiveness

t0ny-peng, 2022-11-01:
Hi there. Thanks for the project. I flashed my Yi Dome camera yesterday, however the web ui is not very responsive, e.g., it takes many seconds to load the content. I then SSH to the camera and found that `./rmm` is constantly using ~40% of the CPU. May I know what does it do and if it's possible to stop it?

```
Mem: 17764K used, 828K free, 572K shrd, 8K buff, 2496K cached
CPU:  4.6% usr 93.5% sys  0.0% nic  0.0% idle  0.6% io  0.0% irq  1.2% sirq
Load average: 17.61 14.55 11.61 5/241 22676
  PID  PPID USER     STAT   VSZ %VSZ CPU %CPU COMMAND
  941     1 root     S    19088102.5   0 22.9 ./rmm
22607     1 root     S     7988 42.9   0 12.2 ./p2p_tnp
  570     2 root     RW       0  0.0   0 10.2 [mmcqd/1]
```
Disk usage:
```
/home/yi-hack-v5 # df -h
Filesystem                Size      Used Available Use% Mounted on
/dev/root                 1.3M      1.2M     84.0K  93% /
tmpfs                     9.1M      4.0K      9.1M   0% /dev
/dev/mtdblock5           12.7M      7.8M      4.8M  62% /home
tmpfs                    16.0M    812.0K     15.2M   5% /tmp
/dev/mmcblk0p1           14.8G      3.8G     11.0G  26% /tmp/sd
/dev/mmcblk0p1           14.8G      3.8G     11.0G  26% /tmp/sd/yi-hack-v5/www/record
```

Memory usage:

```
/home/yi-hack-v5 # free
              total        used        free      shared  buff/cache   available
Mem:          18592       15500         604           0        2488           0
-/+ buffers/cache:        15500        3092
Swap:         65532        3472       62060
```
Thanks!

alienatedsec, 2022-11-28:
@t0ny-peng `./rmm` is better not stopped as it would break your camera recording - it is like the main camera application to run all the time. I suspect `yi_dome` and its `./rmm` does not like the later version of busybox compiled with this hack.

alienatedsec, 2022-12-05:
Please try `0.3.5`

The issue with two `mp4record` processes should fix many other performance issues

closed 2022-12-05 by alienatedsec