alienatedsec/yi-hack-v5 #203: Cameras start to reboot in about one month of use

ricardok-sp, 2022-11-05:
I have got 5 Yi-Home 1.080p cameras with 128gb SD Cards, and successfully applied the firmware in all of them. Everything seemed to be working fine but...

Once in a while (about once a month) one of the cameras starts to keep rebooting and rebooting, and never passes the yellow light. The way I found easier to work it out was to reapply the firmware in them. So I have a 16Gb SD ready with the hacked firmware, and when a camera starts to reboot, I switch the SD for this one, wait some minutes and switch them again. It works all the times. 

Ok, that is working for me. But the big question: WHY? Does anyone have any ideas? What can I do to test or to find out what the problem must be?

Thanks!

waleed-alharthi, 2022-11-23:
I confirm running into the same problem. Had to go back yi-hack-v4 which I hate, because I love free RTSP! any workarounds are appreciated.

Minims, 2022-11-23:
Quite the same issue on a Yi Outdoor. It works without issue for about 1 month. 
Then the camera start crashing and does not respond anymore.
I make an hardware reboot, It works a bit, 1 or 2 min. Then it Crash again. 
After a while an hardware reboot does not work anymore.

I need to re flash with yi-hack-v4 to make it works again. hopping a fix too. but don't understand why it crash after 1 month.

ricardok-sp, 2022-11-23:
Still happening here. Last week my internet failed for a day, and after that I had to reflash two of the cameras.

alienatedsec, 2022-11-24:
@ricardok-sp, what exactly do you have running? RTSP? what else?

ricardok-sp, 2022-11-25:
Just RTSP (LR/HR/LR stream), SSH, FTP, HTTPD, Swap File.

alienatedsec, 2022-11-28:
@ricardok-sp in contrast to yours, I have 20% on my cameras for free SD space. I will let it run, but it could be a month before we find out the result

ricardok-sp, 2022-11-30:
Ok, the problem may be this one (just an SD with no free space). One of my cameras went rebooting again, and instead of reapplying the firmware, I just erased some recording files. It worked. I'll set the 20% on the remaining cameras and follow-up...

Maybe it could be useful to set a default value of 10% or 20% for this field on the next versions.

alienatedsec, 2022-11-30:
@ricardok-sp noted. Thanks

closed 2022-12-05 by alienatedsec