alienatedsec/yi-hack-v5 #204: No more WiFi connection Dome camera 1080p (log_no_wifi_dev)

Veritas999, 2022-11-09:
Hello.
I have 2 Dome 1080p  cameras.
After applying the settings in the web interface, one camera lost Wi-Fi connection.
The camera can be flashed in stock and yi-hacks v4 and v5. But it cannot connect to wifi.
When I turn it on, I hear "Waiting to connect". Then I select the network, the qr code is generated, the camera reads it without problems: "QR code scan, connectet to WI-FI", the blue diode blinks, then lights up yellow and reboots. And so endlessly.
The loops are ok. I tried to swap boards, the problem is the same on this board with different cameras.
The login and password are correct in wpa_supplicant.conf.
What do these lines mean in log.txt?
```
[./rmm][1/19/6:41:58:942]: got RMM_SPEAK_CONNECTTING[./dispatch][1/19/6:41:59:595]: wifi disconnected, now reconnect wifi
[./watch_process][1/19/6:42:1:235]: cloud crashed![./watch_process][1/19/6:42:1:236]: p2p_tnp crashed![./watch_process][1/19/6:42:1:237]: arp_test crashed![./dispatch][1/19/6:42:8:277]: wpa may crashed, now reset
[./dispatch][1/19/6:42:8:329]: no usb dev
```

alienatedsec, 2022-12-06:
What is the Hardware ID? The first four digits are at the back of your cameras.

Veritas999, 2023-01-11:
Any solution? Maybe there is a special firmware with a rollback to the factory?

alienatedsec, 2023-01-12:
Have you tried manual wifi connectivity?

Also, when you flash the latest firmware (hack or whatever) please ensure you have a FAT32 filesystem.

Veritas999, 2023-01-12:
> Have you tried manual wifi connectivity?

What does it mean

alienatedsec, 2023-01-12:
https://github.com/alienatedsec/yi-hack-v5/issues/193 and https://github.com/alienatedsec/yi-hack-v5/commit/98f3db1ca38dcb3a81526412ded8103d8c2eb43d

Let me know how it went

Veritas999, 2023-01-12:
I correctly understood, creat configure_wifi.cfg file at the root of the SD card, and write two lines with my data?
wifi_ssid=[your_wifi_essid]
wifi_psk=[your_wifi_password]

Does not help. It tries to connect again, reads the qr code, reboots and starts all over again.

alienatedsec, 2023-01-12:
> I correctly understood, creat configure_wifi.cfg file at the root of the SD card

you will need to place the `configure_wifi.cfg` file in the `/tmp/sd/recover/` folder and restart the camera

Those two lines should be in the file, yes:
```
wifi_ssid=[your_wifi_essid]
wifi_psk=[your_wifi_password]
```

Replace `[your_wifi_essid]` and `[your_wifi_password]` with details matching your WiFi network. DO NOT configure anything using the QR code.

Veritas999, 2023-01-12:
No sir. All the same when power is applied "Waiting to connect". Perhaps the mechanical damage just happened. Thank you!

alienatedsec, 2023-01-12:
> Perhaps the mechanical damage just happened.

no chance

Try recovery firmware, e.g., one available here:
`https://app.box.com/s/cibs7n1mgvhqaqjlidtveegu1uajt5yr/folder/32280574575`

Veritas999, 2023-01-12:
Yes, I tried. And not only this one. With any firmware, the camera can be flashed. Those the yellow diode is on for about a minute, then it blinks and reboots. But after always "Waiting to connect".

alienatedsec, 2023-01-12:
Try this file

[home_h20m.zip](https://github.com/alienatedsec/yi-hack-v5/files/10405043/home_h20m.zip)

extract and place in the root of the SD card (it has to be FAT32)

Veritas999, 2023-01-12:
Same. Flashes, reboots. And then a cyclic connection.
Thanks again for your help!

alienatedsec, 2023-01-12:
Please don't give up on this.

It would be worth comparing if any files are missing or even deleting some and reflashing.

Veritas999, 2023-01-12:
I'm ready. But there is no access to the camera, and I am good friends with a soldering iron and screwdrivers, but not in programming.

alienatedsec, 2023-01-13:
Try pairing it with the Chinese YI app - not sure if that will work, but if it does, we may have some progress.

alienatedsec, 2023-03-03:
Try #238 if still experiencing the issue

closed 2023-06-23 by github-actions[bot]

KKnishev, 2023-07-07:
The same here (( in one happy day without any updates with Outdoor 4CUS
working well with hack-v5 and losted wifi connection :/
blinking blue and then reboot (in loop) and also the same in \tmp\sd\no_wifi_dev\log.txt
nothing helps from this topic

```
[./rmm][1/19/10:1:51:867]: rmm got /tmp/got_wpa ssid(MY_WIFI) pwd(********) bind(EU9OPqmgMDQVsSok)
[./dispatch][1/19/10:1:51:884]: invalid msg 0x1005
[./rmm][1/19/10:1:51:885]: msg snd success[./dispatch][1/19/10:1:51:887]: in choose_server, region_id = 16, api_server = https://api.eu.xiaoyi.com, sname = familymonitor-h30, dlproto = mieu
[./rmm][1/19/10:1:51:889]: msg snd success[./rmm][1/19/10:1:51:906]: got RMM_SPEAK_SCAN_OK[./dispatch][1/19/10:1:51:973]: msg snd success[./dispatch][1/19/10:1:52:9]: invalid msg 0x1004
[./rmm][1/19/10:1:52:14]: got RMM_SPEAK_CONNECTTING[./dispatch][1/19/10:1:52:128]: wifi disconnected, now reconnect wifi
[./watch_process][1/19/10:2:1:435]: cloud crashed![./watch_process][1/19/10:2:1:435]: p2p_tnp crashed![./dispatch][1/19/10:2:3:654]: wpa may crashed, now reset
[./dispatch][1/19/10:2:3:717]: no usb dev

```

alienatedsec, 2023-07-07:
> blinking blue and then reboot (in loop) and also the same in \tmp\sd\no_wifi_dev\log.txt

Try the latest pre-release @KKnishev #248

KKnishev, 2023-07-07:
amazing, it's online again with 9.3.8c, thanks a lot!!

alienatedsec, 2023-07-07:
> amazing, it's online again with 9.3.8c, thanks a lot!!

`9.3.8d` is the latest :) https://github.com/alienatedsec/yi-hack-v5/releases/tag/9.3.8d

Veritas999, 2023-07-20:
Unfortunately, it didn't help me.

laverdone, 2023-07-28:
same error here, any news?

alienatedsec, 2023-07-28:
Reset your cameras to defaults and update with the latest 0.3.8

alienatedsec, 2023-08-07:
Chinese versions of the cam will need proxychain setting up to work correctly outside of China - there was a problem with proxychain in general, especially with early versions of `yi-hack-v5`. I got it fixed with [this `0.4.0b` release](https://github.com/alienatedsec/yi-hack-v5/releases/tag/0.4.0b). 

Please, make sure you load the right baseline firmware first.

dskvr, 2023-10-29:
Also having this issue. yi-hack-v3 and yi-hack-v5 worked without issues. But after a week, the camera would noo longer connect to wifi. Upon trying to mitigate the issue, I am stuck in the same wifi loop as described by @Veritas999

alienatedsec, 2023-10-30:
> UPDATE 2: While I have connected to WIFI, and I've done the process again, this time connecting on recovery and THEN flashing `v5`, web interface is still not accessible after 20 minutes. Traceroute to the IP confirms it's the correct device (`yi-hack-v5`) and yes, at port `8080`. Spent enough time going in circles for now.

@dskvr 
Did you copy the `yi-hack-v5` folder to the root of your SD card? #267 

>There are certain complexities with the project and what it was based on; yi-hack-v3, and yi-hack-v4.
>The above versions are optimal for what was required at the time and reside within the device's rootfs and home partitions.

>This has evolved with yi-hack-v5 - the rootfs is too small to hold all binaries, and home is not flashing when packed with more files. As an outcome, the yi-hack-v5 folder is located on the SD Card; however, the yi-hack-v5 needs the initial baseline hack to rootfs and home partitions too.

AnthraX1, 2023-11-10:
I had the same problem. Fixed using original rootfs_h20 and home_h20 files.
Format your sd card to fat32 and copy those two files in there. Plug the sd card in and plug in the usb cable. 
In my case the yellow light will keep on and the camera appears stuck, not even responding to reset button. Unplug the camera then move to the next step:

Now, download v0.38 with the rootfs_h20, home_h20 files and yi-hack-v5 folder. untar the file and copy all contents to your sd card. Make sure all h20 files and yi-hack-v5 folder is in the root dir of your sd card.
Put in the sd card and plug in the camera, it would flash and reboot and now you can pair it with app again.

laverdone, 2023-11-10:
> I had the same problem. Fixed using original rootfs_h20 and home_h20 files. Format your sd card to fat32 and copy those two files in there. Plug the sd card in and plug in the usb cable. In my case the yellow light will keep on and the camera appears stuck, not even responding to reset button. Wait for about 3 minutes then unplug the camera and move to the next step:
> 
> Now, download v0.38 with the rootfs_h20, home_h20 files and yi-hack-v5 folder. untar the file and copy all contents to your sd card. Make sure all h20 files and yi-hack-v5 folder is in the root dir of your sd card. Put in the sd card and plug in the camera, it would flash and reboot and now you can pair it with app again.

Work perfectly for me... Good solution

kknishev, 2023-11-13:
> Fixed using original rootfs_h20 and home_h20 files.

what version of the original files?

alienatedsec, 2023-11-21:
Please try the new pre-release
https://github.com/alienatedsec/yi-hack-v5/releases/tag/0.4.1c

kknishev, 2023-11-28:
for my outdoor unfortunately nothing changes for all last pre-releases
I trying 0.3.7/0.3.8 + pre-release - camera successfully updated, connected to wifi and working absolutelly well (web, onvif, mqtt to HA) during 6-12-24 hours.. and after that going to cycloreboot ;(

alienatedsec, 2023-11-28:
@kknishev 
Which services are you using? e.g., ssh, ftp? whats specific for your case?

alienatedsec, 2023-11-28:
> working absolutelly well (web, onvif, mqtt to HA) 

I just read it again and it says it - is that all? how about RTSP? 
It could be that ONVIF or MQTT are filling the wrong space and killing the camera.

kknishev, 2023-11-28:
> Which services are you using? e.g., ssh, ftp? whats specific for your case?
ssh, rtsp, onvif, mqtt -yes, ftp -no
I use cam in HA with go2rtc addon just for web monitoring and motion detection.
I have a three yi-outdoors cams with the same HW, but only one has that issue...

I read about that mqtt is killing a cam and it seems to me that without it cam working more longer... any solution here?

alienatedsec, 2023-11-28:
> I have a three yi-outdoors cams with the same HW, but only one has that issue...

@kknishev 
Is it possible that it was flashed differently than other two cameras? E.g., it used to run other hack versions before flashing to v5?

kknishev, 2023-11-29:
> Is it possible that it was flashed differently than other two cameras?

Two other cams were flashed not at the same time, can't remember...  but with v5 exactly
Moreover the cam with issue has been working well with v5 for over a year and the problem started with ~ 0.3.7/0.3.8

*my wifi net has a common SID for 2.4 and 5Ghz