alienatedsec/yi-hack-v5 #211: Privacy status unavailable

g1za, 2022-12-05:
Hello,
Has anything changed with version 0.3.5?
I installed "yi-hack_ha_integration" but the privacy switch made available by this custom integration is unavailable since that upgrade.
Thanks

g1za, 2022-12-05:
Also camera entity made available by the same seems to be broken.

alienatedsec, 2022-12-05:
Yes @g1za - https://github.com/alienatedsec/yi-hack-v5/commit/44da7e79c5eb8399e3543eb8df17dcd236be502b

I'll have a look tomorrow when free

alienatedsec, 2022-12-06:
@g1za Can I have a screenshot from your HA? I never used this integration, but happy to make it right.

Mine was added successfully but shows offline - I cannot see any other sensors and functionalities like the `privacy switch` or even the `RTSP` stream.

[image]

alienatedsec, 2022-12-06:
I think I know what I did - let me test it.

g1za, 2022-12-06:
Thanks, just in case please see my screenshots

[image]

[image]

[image]

[image]

[image]

g1za, 2022-12-06:
And also (I disabled some as not applicable for my old Yi Dome or because taking too many resources - disabled on MQTT page FW as well) 

[image]

closed 2022-12-06 by alienatedsec

g1za, 2022-12-06:
Thank you. It's back to normal now.

About swap file:

Short answer: not sure if I need it but yes, I use it.

Long answer: for my platform I think it's enabled by default (20% if I remember correctly), anyway I turn the option on. Some time ago I had an issue and manually increased to 60 Then the problem disappeared (probably some issues with the SD card) and I didn't worry too much about that anymore (I also think that recent updates may have turned it back to default values).
If your question is related to performance on my platform: if I disable anything I do not need (ONVIF, on MQTT I delete the suffixes that are not applicable to me -  I read somewhere this is the bulletproof way to save resources) it is quite stable now. Only from time to time the camera enters a motion detection loop and it starts and keeps recording... if I reboot the camera it goes back to normal (not sure if it a problem of the camera or of the integration I use).
I may try to switch to RTSP high res and see what happens.

Thanks again.