alienatedsec/yi-hack-v5 #212: Changes to Configuration and WiFi credentials are lost after reboot

MichlW, 2022-12-06:
Camera model: Yi Outdoor 4CUS.

1) After upgrading to latest version 0.3.6 BETA any changes made in `page=configurations` are not persistent after a reboot. And of course, i used the save button before restarting the camera! ;-)

2) Also WiFi changes on `page=wifi` are not saved or applied correctly and will be lost after a reboot.

My workaround to change some settings (RTSP streams 'both' in my case) was to edit the file `/tmp/sd/yi-hack-v5/etc/system.conf` manually. That worked and was persistent through a reboot.

My workaround to change WiFi credentials was manually creating file `/tmp/configure_wifi.cfg` and executing the script `/tmp/sd/yi-hack-v5/script/configure_wifi.sh` from shell.

Both worked for me.

alienatedsec, 2022-12-06:
How did you upgrade?

alienatedsec, 2022-12-06:
Weird - it works for me - there is no difference in which camera is used for this function. It takes long for the `saved` message to appear

[image]

15-20 seconds later

[image]

MichlW, 2022-12-06:
That Saved message comes really quick in about 1sec. I wondered about this, because I know that saving normally consumes some time.
Its not a browser issue, I tried another system and other browsers. Also I removed any cookies before...

Are there anything  I can try or test, where it hangs?

alienatedsec, 2022-12-06:
Paste the output for `ps tree` please

MichlW, 2022-12-06:
```
/home/yi-hack-v5 # ps tree
PID   USER     TIME  COMMAND
    1 root      0:02 init
    2 root      0:00 [kthreadd]
    3 root      0:36 [ksoftirqd/0]
    6 root      0:00 [khelper]
    7 root      0:00 [kworker/u:1]
   88 root      0:01 [sync_supers]
   90 root      0:00 [bdi-default]
   91 root      0:00 [kintegrityd]
   93 root      0:00 [kblockd]
  103 root      0:00 [khubd]
  105 root      0:05 [kusbotg]
  199 root      0:00 [kswapd0]
  251 root      0:00 [fsnotify_mark]
  264 root      0:00 [crypto]
  316 root      0:00 [romblock0]
  319 root      0:00 [mtdblock0]
  324 root      0:00 [romblock1]
  327 root      0:00 [mtdblock1]
  332 root      0:00 [romblock2]
  335 root      0:00 [mtdblock2]
  340 root      0:00 [romblock3]
  343 root      0:00 [mtdblock3]
  348 root      0:00 [romblock4]
  351 root      0:00 [mtdblock4]
  356 root      0:00 [romblock5]
  359 root      0:00 [mtdblock5]
  364 root      0:00 [romblock6]
  367 root      0:00 [mtdblock6]
  372 root      0:00 [romblock7]
  375 root      0:00 [mtdblock7]
  398 root      0:00 [cfinteractive]
  403 root      0:00 [kworker/0:1]
  404 root      0:00 [kworker/u:2]
  412 root      0:00 [deferwq]
  413 root      0:00 [jffs2_gcd_mtd4]
  427 root      0:00 udevd --daemon
  470 root      0:00 udevd --daemon
  482 root      0:00 udevd --daemon
  500 root      0:11 [mmcqd/1]
  678 root      0:00 [jffs2_gcd_mtd5]
  712 root      0:03 [flush-179:0]
  773 root      0:00 [cfg80211]
  792 root      0:00 ./log_server
  793 root      2:54 ./dispatch
  973 root      3h11 ./rmm
  988 root      0:00 [hidog]
 1627 root      0:02 httpd -p 8080 -h /tmp/sd/yi-hack-v5/www/ -c /tmp/httpd.conf
 1643 root      0:00 dropbear -R -B
 1648 root      0:00 ipc_multiplexer
 1653 root      0:11 mqttv4
 1672 root      8:47 rRTSPServer -r both -p 554 -a no
 1683 root      5:06 h264grabber -r low -m yi_outdoor -f
 1684 root      5:45 h264grabber -r high -m yi_outdoor -f
 1699 root      0:45 {wd_rtsp.sh} /bin/sh /tmp/sd/yi-hack-v5/script/wd_rtsp.sh
 1736 root      0:00 /usr/sbin/crond -c /var/spool/cron/crontabs/
 1779 root      0:00 ntpd -p de.pool.ntp.org
 2086 root      0:00 crond -c /tmp/sd/yi-hack-v5/etc/crontabs
 4180 root      0:00 [kworker/0:2]
 4217 root      3:19 [RtmpTimerTask]
 4218 root      1:24 [RtmpMlmeTask]
 4219 root      0:00 [RtmpCmdQTask]
 4220 root      0:00 [RtmpWscTask]
 4232 root      0:21 /home/base/tools/wpa_supplicant -c/tmp/wpa_supplicant.conf -g/var/run/wpa_supplicant-global -iwlan0 -B
 4356 root      0:00 /sbin/udhcpc -i wlan0 -b -s /home/app/script/default.script -x hostname:Kamera-Haustuer
10831 root      0:00 dropbear -R -B
10888 root      0:00 sleep 10
10890 root      0:00 -sh
10896 root      0:00 ps tree
```

alienatedsec, 2022-12-06:
Can you kill the httpd service `1627` upload and replace the `www` folder from the hack and restart to see if this helps?

MichlW, 2022-12-06:
I killed httpd, deleted the content in `/tmp/sd/yi-hack-v5/www/` and copied the content from the www folder of the file `yi_outdoor_0.3.6.tgz` into it.
The problem is exactly the same and the save process only consumes less than 1sec. Nothing is saved.

I think i will freshly install the 0.3.6 release to the camera. Any other ideas?

alienatedsec, 2022-12-06:
you can also try
`chmod -R 777 /tmp/sd/yi-hack-v5/etc -v -f >> log.txt` and paste the output of the `log.txt` here

MichlW, 2022-12-06:
```
mode of '/tmp/sd/yi-hack-v5/etc' changed to 0777 (rwxrwxrwx)
mode of '/tmp/sd/yi-hack-v5/etc/mqtt_advertise.conf' changed to 0777 (rwxrwxrwx)
mode of '/tmp/sd/yi-hack-v5/etc/proxychains.conf' changed to 0777 (rwxrwxrwx)
mode of '/tmp/sd/yi-hack-v5/etc/camera.conf' changed to 0777 (rwxrwxrwx)
mode of '/tmp/sd/yi-hack-v5/etc/mqttv4.conf' changed to 0777 (rwxrwxrwx)
mode of '/tmp/sd/yi-hack-v5/etc/TZ' changed to 0777 (rwxrwxrwx)
mode of '/tmp/sd/yi-hack-v5/etc/crontabs' changed to 0777 (rwxrwxrwx)
mode of '/tmp/sd/yi-hack-v5/etc/crontabs/root' changed to 0777 (rwxrwxrwx)
mode of '/tmp/sd/yi-hack-v5/etc/wm_res' changed to 0777 (rwxrwxrwx)
mode of '/tmp/sd/yi-hack-v5/etc/wm_res/high' changed to 0777 (rwxrwxrwx)
mode of '/tmp/sd/yi-hack-v5/etc/wm_res/high/wm_540p_5.bmp' changed to 0777 (rwxrwxrwx)
mode of '/tmp/sd/yi-hack-v5/etc/wm_res/high/wm_540p_0.bmp' changed to 0777 (rwxrwxrwx)
mode of '/tmp/sd/yi-hack-v5/etc/wm_res/high/wm_540p_6.bmp' changed to 0777 (rwxrwxrwx)
mode of '/tmp/sd/yi-hack-v5/etc/wm_res/high/wm_540p_7.bmp' changed to 0777 (rwxrwxrwx)
mode of '/tmp/sd/yi-hack-v5/etc/wm_res/high/wm_540p_2.bmp' changed to 0777 (rwxrwxrwx)
mode of '/tmp/sd/yi-hack-v5/etc/wm_res/high/wm_540p_9.bmp' changed to 0777 (rwxrwxrwx)
mode of '/tmp/sd/yi-hack-v5/etc/wm_res/high/wm_540p_3.bmp' changed to 0777 (rwxrwxrwx)
mode of '/tmp/sd/yi-hack-v5/etc/wm_res/high/wm_540p_10.bmp' changed to 0777 (rwxrwxrwx)
mode of '/tmp/sd/yi-hack-v5/etc/wm_res/high/wm_540p_12.bmp' changed to 0777 (rwxrwxrwx)
mode of '/tmp/sd/yi-hack-v5/etc/wm_res/high/wm_540p_1.bmp' changed to 0777 (rwxrwxrwx)
mode of '/tmp/sd/yi-hack-v5/etc/wm_res/high/wm_540p_8.bmp' changed to 0777 (rwxrwxrwx)
mode of '/tmp/sd/yi-hack-v5/etc/wm_res/high/wm_540p_11.bmp' changed to 0777 (rwxrwxrwx)
mode of '/tmp/sd/yi-hack-v5/etc/wm_res/high/wm_540p_4.bmp' changed to 0777 (rwxrwxrwx)
mode of '/tmp/sd/yi-hack-v5/etc/wm_res/low' changed to 0777 (rwxrwxrwx)
mode of '/tmp/sd/yi-hack-v5/etc/wm_res/low/wm_540p_5.bmp' changed to 0777 (rwxrwxrwx)
mode of '/tmp/sd/yi-hack-v5/etc/wm_res/low/wm_540p_0.bmp' changed to 0777 (rwxrwxrwx)
mode of '/tmp/sd/yi-hack-v5/etc/wm_res/low/wm_540p_6.bmp' changed to 0777 (rwxrwxrwx)
mode of '/tmp/sd/yi-hack-v5/etc/wm_res/low/wm_540p_7.bmp' changed to 0777 (rwxrwxrwx)
mode of '/tmp/sd/yi-hack-v5/etc/wm_res/low/wm_540p_2.bmp' changed to 0777 (rwxrwxrwx)
mode of '/tmp/sd/yi-hack-v5/etc/wm_res/low/wm_540p_9.bmp' changed to 0777 (rwxrwxrwx)
mode of '/tmp/sd/yi-hack-v5/etc/wm_res/low/wm_540p_3.bmp' changed to 0777 (rwxrwxrwx)
mode of '/tmp/sd/yi-hack-v5/etc/wm_res/low/wm_540p_10.bmp' changed to 0777 (rwxrwxrwx)
mode of '/tmp/sd/yi-hack-v5/etc/wm_res/low/wm_540p_12.bmp' changed to 0777 (rwxrwxrwx)
mode of '/tmp/sd/yi-hack-v5/etc/wm_res/low/wm_540p_1.bmp' changed to 0777 (rwxrwxrwx)
mode of '/tmp/sd/yi-hack-v5/etc/wm_res/low/wm_540p_8.bmp' changed to 0777 (rwxrwxrwx)
mode of '/tmp/sd/yi-hack-v5/etc/wm_res/low/wm_540p_11.bmp' changed to 0777 (rwxrwxrwx)
mode of '/tmp/sd/yi-hack-v5/etc/wm_res/low/wm_540p_4.bmp' changed to 0777 (rwxrwxrwx)
mode of '/tmp/sd/yi-hack-v5/etc/defaults.tar.bz2' changed to 0777 (rwxrwxrwx)
mode of '/tmp/sd/yi-hack-v5/etc/system.conf' changed to 0777 (rwxrwxrwx)
mode of '/tmp/sd/yi-hack-v5/etc/dropbear' changed to 0777 (rwxrwxrwx)
mode of '/tmp/sd/yi-hack-v5/etc/dropbear/dropbear_ecdsa_host_key' changed to 0777 (rwxrwxrwx)
mode of '/tmp/sd/yi-hack-v5/etc/hostname' changed to 0777 (rwxrwxrwx)
```

alienatedsec, 2022-12-06:
If saving is not working now, try the SD card status in the Yi app or reflash with a new version - I am about to release `0.3.7` shortly.

closed 2022-12-06 by MichlW