alienatedsec/yi-hack-v5 #215: Nvr dahua doesn't recognize onvif

fraFrankSheriff, 2022-12-07:
Hi,
i have a yi outdoor v0.3.7 and my nvr dahua does't recognize the onvif protocol. The camera works fine with vlc but not with nvr dahua and i think that it's the same for a hikvision nvr.
Maybe the problem is similar to
https://github.com/roleoroleo/yi-hack-MStar/issues/88

Someone could help me to fix?
Thank you
Frank

fraFrankSheriff, 2022-12-07:
No, with dahua nvr has never worked.

alienatedsec, 2022-12-07:
I will look into it after the audio release.

However, I will need your support with the above subject.
Please can I ask for your help and follow the below on the 0.3.7 firmware:
```
cp /tmp/view view1
cp /tmp/view view2
cp /tmp/view view3
```
and send me those three files to  alienatedsec@gmail.com - I don't recommend attaching them here.
Please wait between 5-10 seconds before running each line of commands

fraFrankSheriff, 2022-12-07:
just sent

alienatedsec, 2022-12-07:
Hi again @fraFrankSheriff 

Can you please send me another set of files? Maybe try to save them to SD before transferring.
```
cp /tmp/view /tmp/sd/view1
cp /tmp/view /tmp/sd/view2
cp /tmp/view /tmp/sd/view3
```

fraFrankSheriff, 2022-12-07:
Ok, sent!

fraFrankSheriff, 2022-12-12:
Have you find a solution?

PapaGolf1957, 2023-01-05:
Hi
I have a similar issue with my 15US 1080p DomeCamera and a Xvision NVR. However using the ONVIF Device Tools the profile for the camera shows that for the ONVIF Video Encoding Qulaity, Frame Rate, BitRate are all zero similarly for Frame Rate under the video source. Using 2 other ONVIF cameras and the same tool all of these values are non-zero. By the way, you have made a virtually useless camera with the stock software, into something that I can use, keep up the good work

enprsuw, 2023-03-08:
onvif on my Synology does not work for this v5 release. Please prioritiese this feature. :)

PapaGolf1957, 2023-06-15:
The question was is ONVIF going to be corrected, there are 3 messages above all saying that the ONVIF settings within the hacked firmware are incompatible with the mentioned NVRs. As I stated in my message I thing it is down to missing data within the profile. It would be a shame to go as far as you have in making this series of cameras usable without looking at ONVIF again.

closed 2023-07-24 by github-actions[bot]