alienatedsec/yi-hack-v5 #217: The sound speed is too fast when playing

wangxu8611, 2022-12-07:
Video recorded via rtsp. The sound speed is too fast when playing. The video time is 60 seconds and the sound time is 25 seconds after the sound and picture are separated

wangxu8611, 2022-12-07:
Im sorry。
The video stored in the SD card is recorded by the camera itself, not by rstp

alienatedsec, 2022-12-07:
Which hack version and camera do you use?

wangxu8611, 2022-12-07:
release： yi_home_1080p_0.3.7.tgz
camera model：YYS.2016.CN

Firmware Version | 0.3.7
Base Version | 2.1.0.0E_201809191630
Model Suffix | yi_home_1080p
Hardware ID | 48CN

wangxu8611, 2022-12-07:
If you need my video file, I can send it to you by email

alienatedsec, 2022-12-07:
weird - mine is a few seconds delayed, but I am testing a new version.

Please send to alienatedsec@gmail.com

wangxu8611, 2022-12-07:
file has been sent and one more issue about onvif

alienatedsec, 2022-12-08:
Does it work fine on your mobile?

wangxu8611, 2022-12-10:
手机看回放视频，声音也是断断续续的
Watch the playback on the mobile APP, and the sound is intermittent。
感觉像是芯片性能不够，导致的声音断续。
It seems that the sound is intermittent due to insufficient chip performance.
开启rtsp以后，感觉后端网站打开速度都变慢了很多。
After starting rtsp, I feel that the backend websites are much slower.

alienatedsec, 2022-12-10:
So, the final thought about what SD card is in the camera?

wangxu8611, 2022-12-11:
[image]
class10

alienatedsec, 2022-12-11:
[hdparm.zip](https://github.com/alienatedsec/yi-hack-v5/files/10203088/hdparm.zip)

Please unpack the `hdparm` to `/tmp/sd` and execute it as per the below:
`./tmp/sd/hdparm -Tt /dev/mmcblk0p1`

The below is my reading
```

/tmp/sd # ./hdparm -Tt /dev/mmcblk0p1

/dev/mmcblk0p1:
Timing buffer-cache reads:    12 MB in 0.56 seconds = 21890 kB/s
Timing buffered disk reads:   14 MB in 3.24 seconds = 4422 kB/s
/tmp/sd #

```

closed 2023-06-23 by github-actions[bot]