alienatedsec/yi-hack-v5 #219: Error from stream worker: Error demuxing stream

pobezito, 2022-12-12:
Hi,

I´ve running this integration without any problem until now.

At the moment it still works most of the time, but I´m having unexpected reboots and problems with the streaming on the lovelace card.

I get this from the depuration:

homeassistant.components.stream.stream.camera.yi_hack_v5_b1421e_cam] Error from stream worker: Error demuxing stream while finding first packet: [Errno 110] Operation timed out: 'rtsp://192.168.1.140/ch0_1.h264'

Any thoughts?

Thanks in advance,

alienatedsec, 2022-12-12:
@pobezito Please use the rRTSPServer from the previous version e.g. 0.3.6. Just copy a file and overwrite but keep the rest intact - so the entire 0.3.7 remains on the sd card

pobezito, 2022-12-12:
I was using 0.3.2, sorry.

The 0.3.7 update seems to have improved things. Now the stream works on WebRTC but the reboots keep appearing (every 7-15 minutes). May this be caused byt the rRTSPServer file in bin directory? Still havent tried your idea.

Thanks for help

alienatedsec, 2022-12-12:
> May this be caused byt the rRTSPServer file in bin directory? Still havent tried your idea.

Try first, and please let me know.

closed 2023-06-21 by github-actions[bot]