alienatedsec/yi-hack-v5 #222: Yi Dome 1080P 15US - No video if cloud is disabled

tarcisiocjr, 2022-12-25:
Hey @alienatedsec, if possible, can you help-me to debug this issue?

I have a Yi Dome 1080P 15US and if I try disable the cloud functionality the snapshot and rtsp stop to work. Here is part of my log.txt:

```sh
[./rmm][12/25/3:1:38:336]: ai_get_aacstream: get aenc stream select time out
[./rmm][12/25/3:1:38:363]: (0), failed with 0xa0188005!
[./rmm][12/25/3:1:38:392]: (0), failed with 0xa0188005!
[./rmm][12/25/3:1:38:422]: (0), failed with 0xa0188005!
[./rmm][12/25/3:1:38:452]: (0), failed with 0xa0188005!
[./rmm][12/25/3:1:38:481]: (0), failed with 0xa0188005!
tFrame err, vi chn 0 a0078005
[./rmm][1/1/0:0:29:469]: HI_MPI_VI_GetFrame err, vi chn 0 a0078005
[./rmm][1/1/0:0:29:471]: HI_MPI_VI_GetFrame err, vi chn 0 a0078005
[./rmm][1/1/0:0:29:474]: HI_MPI_VI_GetFrame err, vi chn 0 a0078005
```

Full log attached. Do you have any tips to solve this?

[log.txt](https://github.com/alienatedsec/yi-hack-v5/files/10300620/log.txt)

tarcisiocjr, 2022-12-25:
The process rmn, rRSTPServer, dispatch and h264grabber are running.

[image]

alienatedsec, 2023-03-02:
There is a new code to support a disabled cloud; however, not been tested as of yet. It should be included in the next release

closed 2023-07-25 by github-actions[bot]