alienatedsec/yi-hack-v5 #225: Slow streams and snapshots

domenicoblanco, 2023-01-14:
**Describe the bug**
Probably the same as #223, asking a snapshot or starting the streams requires many seconds, multiple times resulting in a connection timeout.

**To Reproduce**
Steps to reproduce the behaviour:
1. Open http://x.x.x.x:8080/cgi-bin/snapshot.sh?res=low&watermark=no in a browser or `ffmpeg -i rtsp://x.x.x.x/ch0_0.h264` as a stream

**Expected behaviour**
A response in less time, preventing connection timeouts

**Set Up Details (please complete the following information):**
 - Firmware Version: 0.3.7
 - Device: Yi 1080 Outdoor
 - Model: 4CUS

**Additional context**
Sometimes is really slow also the web interface to load.
Connecting via SSH the usage by top is about 57%

P.S. Thanks for your work!

Gera1974, 2023-02-03:
Hi, I have the same issue. Below you have the specs of my camera.
Set Up Details (please complete the following information):
Firmware Version: 0.3.7
Device: Yi Home 1080p
Model: 48US

P.S. Thanks for your work!

kapabel, 2023-02-11:
I think I can +1 this issue. My load average is on 
7.61 7.89 7.84

i have a 4CUS and just updated to 0.3.7.

alienatedsec, 2023-02-13:
Please can you try the [0.3.8](https://drive.google.com/file/d/17s99VxX6F4CxuhA8LiHq3GBgDqb5YkqK/view?usp=sharing) version whenever possible?

domenicoblanco, 2023-02-13:
Just tried and is still slow, also is not able to load the Hi-Res snapshot (both via direct url or from the web)

[image]

alienatedsec, 2023-02-13:
The below is an output of the original busybox from yi_outdoor (I copied it over from the original firmware)

```
/tmp/sd # ./busybox
BusyBox v1.20.2 (2016-01-25 13:29:17 CST) multi-call binary.
Copyright (C) 1998-2011 Erik Andersen, Rob Landley, Denys Vlasenko
and others. Licensed under GPLv2.
See source distribution for full notice.

Usage: busybox [function] [arguments]...
   or: busybox --list[-full]
   or: busybox --install [-s] [DIR]
   or: function [arguments]...

        BusyBox is a multi-call binary that combines many common Unix
        utilities into a single executable.  Most people will create a
        link to busybox for each function they wish to use and BusyBox
        will act like whatever it was invoked as.

Currently defined functions:
        add-shell, addgroup, adduser, arp, arping, ash, awk, bootchartd, cat,
        chmod, chown, chpasswd, cp, cryptpw, cttyhack, cut, date, dd, delgroup,
        deluser, depmod, df, du, echo, egrep, fdisk, fgrep, find,
        flash_eraseall, flashcp, free, fsync, getty, grep, halt, hd, hexdump,
        hush, ifconfig, init, insmod, iostat, kill, killall, killall5, linuxrc,
        ln, login, logname, ls, lsmod, lsof, lsusb, lzcat, lzma, md5sum, mesg,
        mkdir, mknod, mkpasswd, modinfo, modprobe, more, mount, mpstat, mv,
        netstat, passwd, ping, poweroff, ps, pwd, reboot, remove-shell, rm,
        rmdir, rmmod, route, sed, sh, sleep, su, sulogin, sync, sysctl, tail,
        tar, top, tr, udhcpc, umount, unlzma, usleep, vi, vlock, xargs

```
The below is on this hack v0.3.7 and v0.3.8
```
/tmp/sd # busybox
BusyBox v1.31.1 (2022-12-06 12:01:56 GMT) multi-call binary.
BusyBox is copyrighted by many authors between 1998-2015.
Licensed under GPLv2. See source distribution for detailed
copyright notices.

Usage: busybox [function [arguments]...]
   or: busybox --list[-full]
   or: busybox --install [-s] [DIR]
   or: function [arguments]...

        BusyBox is a multi-call binary that combines many common Unix
        utilities into a single executable.  Most people will create a
        link to busybox for each function they wish to use and BusyBox
        will act like whatever it was invoked as.

Currently defined functions:
        [, [[, add-shell, addgroup, adduser, arch, arp, arping, ash, awk,
        base64, basename, bc, bootchartd, bzip2, cat, chgrp, chmod, chown,
        chpasswd, cp, crond, crontab, cryptpw, cttyhack, cut, date, dd,
        delgroup, deluser, depmod, df, dirname, dmesg, du, echo, egrep, env,
        fdisk, fgrep, find, flash_eraseall, flashcp, flock, free, fsync, ftpd,
        ftpget, ftpput, getty, grep, gzip, halt, hd, hdparm, hexdump, hexedit,
        hostname, httpd, i2ctransfer, ifconfig, inetd, init, insmod, iostat,
        kill, killall, killall5, linuxrc, ln, login, logname, ls, lsmod, lsof,
        lsusb, lzcat, lzma, md5sum, mesg, mkdir, mknod, mkpasswd, mkswap,
        modinfo, modprobe, more, mount, mpstat, mv, nc, netstat, nohup, ntpd,
        od, passwd, pidof, ping, poweroff, printenv, ps, pwd, reboot,
        remove-shell, resume, rm, rmdir, rmmod, route, sed, sendmail, setfattr,
        setpriv, sh, sha1sum, sleep, sort, su, sulogin, svok, swapon, sync,
        sysctl, tail, tar, tc, tcpsvd, telnetd, test, top, touch, tr, ts,
        udhcpc, umount, unlzma, uptime, usleep, vi, vlock, wget, xargs
/tmp/sd #
```

I need to find a way to test with the legacy busybox and the upgraded version.

domenicoblanco, 2023-02-13:
I'm trying to compile the original version from source in the cam, let me know if I can do something to help

alienatedsec, 2023-02-13:
> I'm trying to compile the original version from source in the cam

How? In the cam?

domenicoblanco, 2023-02-13:
Sorry, not directly in the cam, the idea is to cross-compile but I'm having some difficulties with the toolchain

alienatedsec, 2023-02-14:
There is something in the initial scripts, which is preventing the camera to run on the legacy busybox. It will be a missing function or something. I will need to compile with the version used in the v4, and separate v5 busybox applets (hardlinks) to `/tmp/sd/yi-hack-v5/[bin,sbin]/usr/[bin/sbin].

Pakstyle, 2023-05-28:
got the same issue, following to see what happends

closed 2023-07-05 by github-actions[bot]