alienatedsec/yi-hack-v5 #226: How to tell which Yi Camera you have? Why not list actual model numbers?

starman2k01, 2023-01-15:
My US cameras do have model numbers that look different. My YI 1030p Home Camera is model number YYS2016.
My Outdoor cam is "Outdoor Camera 1080p" model number  YHS3017. Where is this "27US/47US" number, and why not list the exact model numbers supported? I tried flashing my 1080P home with the y20 firmware and it didn't work.
I also have a "Home Camera 3" model YYS2518. I have no idea if this camera is compatible...

alienatedsec, 2023-01-16:
The first 4 digits of under your QR code are the model number recognised by the hack.

starman2k01, 2023-01-16:
Okay, so what model is "Yi 1080p Dome"? I have a "Yi Dome Camera 1080P" BFUS. Yi has SO many camera models that it's important to always say the model number, right?