alienatedsec/yi-hack-v5 #232: Add support for 0.4.4: yi-hack_ha_integration

oat1, 2023-02-19:
Please upgrade the project, to support the new release of yi-hack Home Assistant integration

https://github.com/roleoroleo/yi-hack_ha_integration

alienatedsec, 2023-02-20:
This integration is already partially supported, and I haven't tested the latest version - looking at the code, there are more motion detection functions and enablers. It also checks for the latest version of the hack (mainly allwinner hack). I'll have a chat with roleo to find out more.

oat1, 2023-02-20:
Right now, I get :
[image]

SO had to came back to 0.4.2 version

roleoroleo, 2023-02-20:
Sorry for the breaking changes.
Stay with the previous version of the integration, I will update the code asap.

roleoroleo, 2023-02-21:
@alienatedsec 
Do you want to test the new branch roleo_testing?

oat1, 2023-03-02:
@roleoroleo Would you be able to give us an approximate releasing date for the new patch?

alienatedsec, 2023-03-02:
@oat1 stay still, I have done some initial testing, but it needs some amendments and confirmation.
Check this out https://github.com/roleoroleo/yi-hack_ha_integration/issues/113

oat1, 2023-03-21:
@alienatedsec any news? maybe I can help with testing?

mnowok, 2023-03-21:
If you need testing Yi 1080p Dome I'd like to do it.

alienatedsec, 2023-04-14:
I know it has been a while - please can you try this @mnowok 
[yi_dome_1080p.zip](https://github.com/alienatedsec/yi-hack-v5/files/11237603/yi_dome_1080p.zip)

It also comes with RTSP audio, RTSP password....

alienatedsec, 2023-04-14:
@oat1 which camera do you have?

oat1, 2023-04-15:
Yi outdoor
[image]

mnowok, 2023-04-15:
> I know it has been a while - please can you try this @mnowok [yi_dome_1080p.zip](https://github.com/alienatedsec/yi-hack-v5/files/11237603/yi_dome_1080p.zip)
> 
> It also comes with RTSP audio, RTSP password....

I upgraded one of my two cameras. I noticed the updated wiped my configuration, but not a problem. Camera works in Home Assistant, RTP stream is ok in HA webrtc-camera component. 

I don't get snapshot on cam's webpage. As I haven't been using MQTT on these cameras I can configure it and check later.
[image]

alienatedsec, 2023-04-15:
@mnowok is the low-resolution snapshot working?

mnowok, 2023-04-15:
Maybe I'm blind - how to get low-res snapshot? I can't see any option in Config/Settings.

alienatedsec, 2023-04-15:
@mnowok links are available on the home page of the cam

[image]

mnowok, 2023-04-15:
Ok, I'm blind :-) Anyway - both snapshot links (low/high) don't produce any image (chrome displays empty white screen).

alienatedsec, 2023-04-15:
Can you run the below from the console and provide an output? Point the camera in a less private direction as it will produce an image.

`imggrabber -r low -m yi_dome_1080p -w -d >> img.jpg`

It will produce the `img.jpg` file and the output like the one below. Can you insert the file and what is produced on the screen?

```
/home/yi-hack-v5 # imggrabber -r low -m yi_dome_1080p -w -d >> img.output
Debug on
Starting program
Resolution low
1681552400374 - mapping file /tmp/view, size 1586752, to b6d2a000
1681552400389 - closing the file /tmp/view
1681552400390 - found latest frame: id 87, frame_counter 1918
1681552400398 - processing frame 87
1681552400417 - processing frame 87
1681552400435 - processing frame 87
1681552400453 - processing frame 87
1681552400474 - processing frame 88
1681552400494 - processing frame 88
1681552400514 - processing frame 88
1681552400535 - processing frame 89
1681552400554 - processing frame 89
1681552400573 - processing frame 90
1681552400595 - processing frame 90
1681552400613 - processing frame 90
1681552400635 - processing frame 91
1681552400663 - processing frame 91
1681552400683 - processing frame 92
1681552400714 - processing frame 92
1681552400735 - processing frame 93
1681552400754 - processing frame 93
1681552400774 - processing frame 94
1681552400794 - processing frame 94
1681552400813 - processing frame 94
1681552400835 - processing frame 95
1681552400854 - processing frame 95
1681552400874 - processing frame 96
1681552400894 - processing frame 96
1681552400913 - processing frame 96
1681552400933 - processing frame 97
1681552400954 - processing frame 97
1681552400974 - processing frame 98
1681552400998 - processing frame 98
1681552401014 - processing frame 98
1681552401035 - processing frame 99
1681552401054 - processing frame 99
1681552401074 - processing frame 100
1681552401095 - processing frame 100
1681552401113 - processing frame 100
1681552401136 - processing frame 101
1681552401136 - writing frame: frame_offset 286124, frame_ptr b6e793ec, frame_length 36
1681552401156 - processing frame 102
1681552401156 - writing frame: frame_offset 286164, frame_ptr b6e79414, frame_length 8
1681552401174 - processing frame 103
1681552401174 - writing frame: frame_offset 286176, frame_ptr b6e79420, frame_length 18548
7658098487526604 - frame found, exit loop
Decoding h264 frame
Starting decode
Decode frame
Writing yuv buffer
Cleaning ffmpeg memory
Adding watermark
Encoding jpeg image
Unmapping file /tmp/view, size 1586752, from b6d2a000
```

mnowok, 2023-04-15:
Segmenation fault:

/home/yi-hack-v5 # imggrabber -r low -m yi_dome_1080p -w -d >> img.output
Debug on
Starting program
Resolution low
1681552764075 - mapping file /tmp/view, size 1586752, to b6d22000
1681552764099 - closing the file /tmp/view
1681552764099 - found latest frame: id 236, frame_counter 64627
1681552764103 - processing frame 236
1681552764122 - processing frame 236
1681552764142 - processing frame 236
1681552764162 - processing frame 236
1681552764183 - processing frame 236
1681552764209 - processing frame 237
1681552764222 - processing frame 237
1681552764242 - processing frame 238
1681552764264 - processing frame 238
1681552764291 - processing frame 238
1681552764313 - processing frame 239
1681552764333 - processing frame 239
1681552764353 - processing frame 240
1681552764372 - processing frame 240
1681552764392 - processing frame 241
1681552764413 - processing frame 241
1681552764433 - processing frame 241
1681552764463 - processing frame 242
1681552764483 - processing frame 242
1681552764502 - processing frame 243
1681552764522 - processing frame 243
1681552764542 - processing frame 244
1681552764563 - processing frame 244
1681552764583 - processing frame 244
1681552764602 - processing frame 245
1681552764622 - processing frame 245
1681552764651 - processing frame 246
1681552764673 - processing frame 246
1681552764692 - processing frame 247
1681552764714 - processing frame 247
1681552764733 - processing frame 247
1681552764753 - processing frame 248
1681552764773 - processing frame 248
1681552764793 - processing frame 249
1681552764813 - processing frame 249
1681552764833 - processing frame 249
1681552764852 - processing frame 250
1681552764873 - processing frame 250
1681552764901 - processing frame 250
1681552764922 - processing frame 251
1681552764942 - processing frame 252
1681552764962 - processing frame 252
1681552764983 - processing frame 252
1681552765002 - processing frame 253
1681552765022 - processing frame 253
1681552765042 - processing frame 254
1681552765062 - processing frame 254
1681552765082 - processing frame 254
1681552765102 - processing frame 255
1681552765122 - processing frame 255
1681552765142 - processing frame 256
1681552765163 - processing frame 256
1681552765182 - processing frame 256
1681552765202 - processing frame 257
1681552765222 - processing frame 257
1681552765242 - processing frame 258
1681552765262 - processing frame 258
1681552765283 - processing frame 258
1681552765308 - processing frame 259
1681552765322 - processing frame 259
1681552765342 - processing frame 260
1681552765362 - processing frame 260
1681552765382 - processing frame 260
1681552765402 - processing frame 261
1681552765422 - processing frame 261
1681552765449 - processing frame 262
1681552765463 - processing frame 262
1681552765482 - processing frame 262
1681552765502 - processing frame 263
1681552765523 - processing frame 263
1681552765542 - processing frame 263
1681552765562 - processing frame 264
1681552765563 - writing frame: frame_offset 92516, frame_ptr b6e41fa4, frame_length 36
1681552765583 - processing frame 265
1681552765583 - writing frame: frame_offset 92556, frame_ptr b6e41fcc, frame_length 8
1681552765608 - processing frame 266
1681552765609 - writing frame: frame_offset 92568, frame_ptr b6e41fd8, frame_length 33026
7658098487526604 - frame found, exit loop
Decoding h264 frame
Starting decode
Segmentation fault

alienatedsec, 2023-04-15:
Try replacing with this imggrabber
[imggrabber.zip](https://github.com/alienatedsec/yi-hack-v5/files/11239055/imggrabber.zip)

mnowok, 2023-04-15:
Where is imggrabber located?

alienatedsec, 2023-04-15:
@mnowok 

`/tmp/sd/yi-hack-v5/bin`

mnowok, 2023-04-15:
Now it works fine, both links (low and high) and snapshot page.

/home/yi-hack-v5 # imggrabber -r low -m yi_dome_1080p -w -d >> img.output
Debug on
Starting program
Resolution low
1681553878206 - mapping file /tmp/view, size 1586752, to b6c9e000
1681553878221 - closing the file /tmp/view
1681553878227 - found latest frame: id 11, frame_counter 22866
1681553878236 - processing frame 11
1681553878255 - processing frame 11
1681553878275 - processing frame 11
1681553878294 - processing frame 11
1681553878315 - processing frame 11
1681553878342 - processing frame 11
1681553878365 - processing frame 12
1681553878385 - processing frame 12
1681553878404 - processing frame 13
1681553878425 - processing frame 13
1681553878444 - processing frame 14
1681553878465 - processing frame 14
1681553878485 - processing frame 14
1681553878504 - processing frame 15
1681553878525 - processing frame 15
1681553878551 - processing frame 15
1681553878575 - processing frame 16
1681553878575 - writing frame: frame_offset 219212, frame_ptr b6ddce8c, frame_length 36
1681553878594 - processing frame 17
1681553878595 - writing frame: frame_offset 219252, frame_ptr b6ddceb4, frame_length 8
1681553878616 - processing frame 18
1681553878616 - writing frame: frame_offset 219264, frame_ptr b6ddcec0, frame_length 37761
7658098487526604 - frame found, exit loop
Decoding h264 frame
Starting decode
Decode frame
Writing yuv buffer
Cleaning ffmpeg memory
Adding watermark
Encoding jpeg image
Unmapping file /tmp/view, size 1586752, from b6c9e000

Melectro1, 2023-04-15:
Hi,

I have Yi Dome camera and also Yi Dome cloud 1080.
Both are not working with HA integration 0.4.5, Please, is there anything I should do, I'm using FW 0.3.7?

oat1, 2023-04-17:
@alienatedsec 
Could you send a test file for yi outdoor ? 
Please also describe how I can install it wirelessly, my cam is hanging very high, and removing the sd card is problematic :)

alienatedsec, 2023-04-17:
@oat1 Have a look and monitor this [pre-release discussion](https://github.com/alienatedsec/yi-hack-v5/discussions/248)

I need to update it with the latest - not worth testing as of yet - another pre-release `9.3.8a` to follow soon

Melectro1, 2023-04-17:
@alienatedsec
 is it possible for me to test a new file for the Yi Dome camera and Yi Dome cloud 1080p?

alienatedsec, 2023-04-17:
> @oat1 Have a look and monitor this [pre-release discussion](https://github.com/alienatedsec/yi-hack-v5/discussions/248)
> 
> I need to update it with the latest - not worth testing as of yet - another pre-release `9.3.8a` to follow soon

@Melectro1 Use the same pre-release link for any camera you currently have

Melectro1, 2023-04-17:
@alienatedsec 
Here is my Feedback: I have flashed both types of cameras with version 9.3.8. The web interface is working very slow and both cameras are rebooting very often.

alienatedsec, 2023-04-18:
> @alienatedsec Here is my Feedback: I have flashed both types of cameras with version 9.3.8. The web interface is working very slow and both cameras are rebooting very often.

Thanks; a polite reminder that the guidance says to describe any feedback as part of discussion #248

grungeguitarist, 2023-08-11:
Not working with yi-hack-v5 v0.4.0 and a Yi Outdoor camera.

Added with HACS in Home Assistant.
All switches are off in Home Assistant and turning them on has no effect. Also cam and motion_detection_cam entities do not show anything.

alienatedsec, 2023-08-12:
> Not working with yi-hack-v5 v0.4.0 and a Yi Outdoor camera.

I know there are issues but it's not like not working at all. Can you share more info? E.g., can you see anything when you run mqtt discovery

Melectro1, 2023-08-12:
I don't know if I should interfere with the latest posts, but I have just seen something about only being able to get a connection to the mqtt mosquitto broker to work in Home assistant. Maybe that would explain why my YI camera doesn't work when I also use a mqtt connection for my solar installation.

grungeguitarist, 2023-08-13:
> > Not working with yi-hack-v5 v0.4.0 and a Yi Outdoor camera.
> 
> I know there are issues but it's not like not working at all. Can you share more info? E.g., can you see anything when you run mqtt discovery

Do I have to change something in the web interface to make it work? I have RTSP with both streams and the Swap File enabled, but otherwise, everything is on default (besides a username and password).

So what I did:
- Flashed yi-hack-v5 with 0.3.8 files. Updated to 0.4.0 in the web interface.
- In Home Assistant 2023.7.3 (running hassio on a Raspberry Pi), I installed "yi-hack Home Assistant integration" using HACS.
- Then, "Add Integration" -> "Yi Home Cameras with yi-hack"
- Config (the IP is the one to connect with the web interface of my Yi Outdoor camera):
[image]
- Now the added device looks like this (all switches off, switching on has no effect. FRITZ!BOX 7590 is my router.)
[image]
- I guess this should show the stream but it is not working:
[image]
[image]
- MQTT (everything on default, auto-configured):
[image]

Melectro1, 2023-08-15:
@grungeguitarist
Have you connected other devices with MQTT in your Homeassistant setup?

g1za, 2023-08-17:
-- yi dome camera

I updated to the firmware v.0.4.0 and yi hack integration v.0.4.5 and I'm experiencing the same issue: the integration is not working and any switch is uneffective.
I also have webRTC integration installed and I can see the stream from there, but not from the yi hack integration sensor.
Below some screenshots:

[image]

This is the preview I get if I click on the camera sensor (nothing) but if I feed this to the WebRTC integration I correctly see the stream (I'm not posting here a picture of my home ;) )
[image]
[image]

g1za, 2023-08-17:
I deleted my previous comment as I noticed MQTT was not running on the camera (shame on me) but I can share a bit my experience.
The entities from the yi hack integration that are supposed to show the stream are NOT working out of the box. 
[image]
[image]
[image]
BUT I can see the stream by using the WebRTC integration as suggested by roleoroleo and feeding it with my "telecamera (nascosta)" hidden sensor (not pasting a picture of my home here ;) )

About the switches: I thought they were not working at the beginning but after switching them on/off once now they are working (maybe they needed to synch with the settings I already set from the yi app). I could easily test IR and rotation, I need to test motion detection and recording but I expect they work as well.

I think older switches sensors are not available anymore (but please correct me):
- privacy switch -> replaced by "yi_hack_v5_0aca57_switch_switch_on"
- motion detection sensor -> I can't get it, it may be a problem if you were using it to notify you about detections
- connection status -> can't get it either

g1za, 2023-08-19:
I have a quick update: it’s the second time I find off the switches I turned on (all or most of them). But this condition had no effect on the camera (for example rotation was off but the image still correct; the same for night view).

alienatedsec, 2023-08-19:
@g1za it will be the MQTT module on the camera, and how it interacts with the [`ipc_multiplex`](https://github.com/alienatedsec/yi-hack-v5/tree/master/src/ipc_cmd/ipc_cmd)

@roleoroleo and I are working together to fix this (and other issues) at some point - it is the holiday season and it will be hard to sacrifice good weather and time with the family for coding ;-)

However, we are always happy to review pull requests.

g1za, 2023-08-19:
Of course and I would not ask for that :-)
Take your time, I'm just tracking what I see and likewise to you I do that when I can.
I would be happy to help with pull requests but the problem is that I do not have any idea where to put my hands and I am not even familiar with this coding language :'-)
So what I can do is to report what I see and if needed perform some debug on my camera model.
Thank you and enjoy some time off ;-)

g1za, 2023-08-29:
Thanks for the additional work that has been done so far with the yi hack integration v.0.4.6.   
An update from my Yi dome camera:
- I found in another thread that the privacy sensor is not available anymore, it is a breaking chance -> done
- The motion detection switch has become unavailable
- The motion detection sensor is back but I have the feeling it is not actually showing when motion is detected. I will check further. 
- camera status is always “disconnected” while the camera is online and streaming. 
- I quickly tested some other switches: they are retained, synchronised with the camera settings and effective (for example rotation and led). Related to my previous point I still need to see if “record on motion detection” switch works properly.

alienatedsec, 2023-08-29:
@g1za are you on `0.4.1a` pre-release?

g1za, 2023-08-29:
Ah no… I didn’t notice there was a new pre-  
I will update and let you know how it goes.

g1za, 2023-08-29:
I confirm the above (following the camera and HA reboot). 
this time I also had to re-sync some switches status (camera switch and rotation). 
Events are being recorded (still UTC time displayed) (regardless of the unavailability of the integration switch - I think it relies on internal camera settings)

g1za, 2023-08-29:
About connection status: at first sight it seems it can’t retain it
[image]

roleoroleo, 2023-09-05:
Please, update to 0.4.6.

g1za, 2023-09-10:
I was already on 0.4.6

roleoroleo, 2023-09-20:
I'm working on it.

g1za, 2024-01-13:
Hello everybody,
as we have now release 0.4.8 I just wanted to share that the issues I previously reported (camera availability, frequent recording, ...) seem to be fixed.
The only one still pending is the switch that should reflect the option to record video on motion detection that is still shown as unavailable. 
Thanks for the hard work
[image]

roleoroleo, 2024-01-13:
Dit you try to remove the cam and add it again?

g1za, 2024-01-13:
Not yet. I'll try :)

g1za, 2024-01-13:
I removed the camera and added it again but now that entity is not listed anymore (maybe you didn't map that setting? maybe on purpose?)

Also a couple of additional entities are not available anymore (probably they were missing also before I added the camera again, but as not available for my model I already disabled them - so not relevant for my setup but maybe for somebody else, so mentioning just for the records).

[image]

roleoroleo, 2024-01-14:
Here https://github.com/roleoroleo/yi-hack_ha_integration/blob/main/custom_components/yi_hack/switch.py you can find the list of the switch for the v5.
Here https://github.com/roleoroleo/yi-hack_ha_integration/blob/main/custom_components/yi_hack/binary_sensor.py the list of the binary sensor.
Please check the lists and let me know if your model supports other unlisted entities.

Motion detection swith is commented (probably a mistake)
`YiHackSwitch(hass, config_entry, "motion_detection", "Motion Detection"),`
ths is why it's not working. I will fix the problem.

g1za, 2024-01-14:
My camera is a Yi dome, so a pretty old one and not all the latest features are supported but I can see all of them in the Yi app interface even if disabled.

In my app I see the following switches ( (remotely?) IA enabled - require Kami/Yi subscription, for me at least):
* human detection
* vehicles detection
* animals detection
* other detections
[image]

I also see the following switch:
* motion tracking (not related to recording, the camera will follow the movement) - this is performed locally
[image]

In addition to what already mentioned, the "motion detected recording" switch, currently commented (available before hi-hack-integration v.0.4.4

Please let me know if this covers your question.
Thanks!

roleoroleo, 2024-01-15:
I will check these options and update the integration.
About AI features, for your model they are not available locally. Just through a frame analysis done in the cloud.
So, it's not possible to enable them.

g1za, 2024-01-15:
Yes, that's clear.
About the "missing" motion detection I tried to uncomment and this made it to be available again.
I was checking the app and it would reflect the following section (which was then expanded with the IA features - not available for me locally).
[image]

But I was also cross checking with the hacked firmware and I can't see the option... so maybe that's the reason why you commented it, to make it available in a future release of the firmware as soon as it is supported?

[image]

roleoroleo, 2024-01-15:
> About the "missing" motion detection I tried to uncomment and this made it to be available again.
> I was checking the app and it would reflect the following section (which was then expanded with the IA features - not available for me locally).

Is it fully working?
When you check the option in the app, does the ha switch change?
And vice versa?

g1za, 2024-01-15:
This specific switch doesn't have any effect on the app and viceversa (I think there is no corresponding setting in the camera web-interface).

About other switches: the sync of the settings is bidirectional (including the camera web-interface). For the Yi app it may be needed to force close and re-open it if you change the setting using the entity switch.

roleoroleo, 2024-01-16:
> This specific switch doesn't have any effect on the app and viceversa (I think there is no corresponding setting in the camera web-interface).

I will check it.

alienatedsec, 2024-02-04:
Please follow and test if needed https://github.com/alienatedsec/yi-hack-v5/releases/tag/0.4.1e

alienatedsec, 2024-02-14:
Support is there - the latest https://github.com/alienatedsec/yi-hack-v5/releases/tag/0.4.1i 

Please raise a new issue if needed

closed 2024-02-14 by alienatedsec