alienatedsec/yi-hack-v5 #239: Camera Wont Save Recors files anymore

Dranoeleo, 2023-03-06:
Hi Guys,
i have had the phenomenon, that multiple Cameras wont save the Records anymore as MP4 files, not on the SD card and also not to the FTP Server configured under "Events".

After a little research i found out that there was an blackout before this bug on all cameras.
I took the sd card, copied it and formatted it in Windows, took it back to the camera but it doesnt seem to work.
Then i took a look in the Yi App and the SD Status still said: SD Card needs to be formatted.

The only solution was to take a brand new SD Card and copy all the stuff on it: Now the Records saving works perfectly :)

I havent found this trouble in any of the Yi-Hacks so i decided to report this to the community. I know the problem must have been the broken SD Card but maybe i am not the only one worrying about this.

alienatedsec, 2023-03-06:
@Dranoeleo it would be great if you could follow the template and provide all the necessary information.

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behaviour:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behaviour**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Set Up Details (please complete the following information):**
 - Firmware Version: [e.g. 0.3.7]
 - Device: [e.g., YI HOME 1080P]
 - Model: [e.g., 48US]

**Additional context**
Add any other context about the problem here. If any other systems are involved, please complete the following:
 - Name of the system [e.g., Home Assistant]
 - Version [e.g., 2022-12-xx]

**Technical Details**
Please add any technical details (e.g., log files)

Dranoeleo, 2023-03-06:
Solution is in the text

closed 2023-03-06 by Dranoeleo