alienatedsec/yi-hack-v5 #245: Home assistant mqtt wifi_strength should be in dB or dBm, not %

pkishino, 2023-04-12:
**Describe the bug**
As per a few versions back, the unit of measure for wifi signals was finalized as dB or dBm.
The config script here still uses % which triggers an issue with home assistant during discovery.
The fix would be to stick to using dB which is, what I believe, is the default for the camera (the % uses a conversion formula)

**To Reproduce**
current example wifi_strength config:
```
{
  "availability_topic": "yicam/status",
  "payload_available": "online",
  "payload_not_available": "offline",
  "device": {
    "identifiers": [
      "yi-cam"
    ],
    "manufacturer": "YI",
    "model": "YI Hack",
    "name": "Yi Camera",
    "sw_version": "0.3.7"
  },
  "qos": 1,
  "device_class": "signal_strength",
  "icon": "mdi:wifi",
  "state_topic": "yicam/telemetry",
  "name": "Yi Camera Wlan Strengh",
  "unique_id": "yi-cam-wlan_strength",
  "value_template": "{{ ((value_json.wlan_strength|int) * 100 / 70 )|int }}",
  "unit_of_measurement": "%",
  "platform": "mqtt"
}
```

**Expected behaviour**
the above details should be something as below:
  "unit_of_measurement": "dB",

**Screenshots**
N/A

**Set Up Details (please complete the following information):**
 - Firmware Version: 0.3.7
 - Device: YI Dome 1080p
 - Model: 45US

**Additional context**
Add any other context about the problem here. If any other systems are involved, please complete the following:
 - Name of the system [e.g., Home Assistant]
 - Version [e.g., 2022-12-xx]

**Technical Details**
Please add any technical details (e.g., log files)

pkishino, 2023-06-14:
Still active, can try to look into creating a pr..

pkishino, 2023-07-16:
Haven’t gotten around to this yet…

alienatedsec, 2023-08-10:
@pkishino Looking closer into recent fixes, this would not be relevant and the current unit is correct

Please check this https://github.com/alienatedsec/yi-hack-v5/issues/42#issuecomment-1672692347

closed 2023-08-10 by alienatedsec