alienatedsec/yi-hack-v5 #251: RTSP not working on 47US

scruben, 2023-05-25:
Installation was fine (Firmware is 0.3.7 on a Yi Home 720p 47US).

Config webpage can be accessed and snapshots can be done. 

RTSP is enabled but once I copy the URL to VLC or kaffeine it doesn't work.

Tried all the combinations with/out password but no luck. 

Any idea? anything I can ssh and log to help debugging? Thanks!

noz1380, 2023-06-24:
I think I'm finding the same with 9.3.8c

Tried on phone with vlc, but cannot get it to work with HA.

alienatedsec, 2023-06-25:
@noz1380, please execute this script twice (option kill first, and second-time launch) and run it in debug mode. I will need the to see the output.

[launch.zip](https://github.com/alienatedsec/yi-hack-v5/files/11859434/launch.zip)

scruben, 2023-06-30:
I don't get any debug messages even after trying to open the stream in VLC... could be those invalid options in rRTSPServer? Thanks

```
/home/yi-hack-v5/script # ./launch.sh 
Using model: yi_home
Specify the operation (k for kill, l for launch): l
Specify the settings (high, low, or both): high
Enable audio? (y/n): n
Execute in debug mode? (y/n): y
Using fifo as output
Resolution high
rRTSPServer: invalid option -- a
rRTSPServer: invalid option -- u
rRTSPServer: invalid option -- w
debug on

"ch0_0.h264" stream, from the file "/tmp/h264_high_fifo"
Play this stream using the URL "rtsp://192.168.0.29/ch0_0.h264"
```

alienatedsec, 2023-06-30:
Works for me

```
/home/yi-hack-v5 # ./launch.sh
Using model: yi_dome
Specify the operation (k for kill, l for launch): l
Specify the settings (high, low, or both): high
Enable audio? (y/n): n
Execute in debug mode? (y/n): y
Using fifo as output
Resolution high

"ch0_0.h264" stream, from the file "/tmp/h264_high_fifo"
Play this stream using the URL "rtsp://[REDACTED]/ch0_0.h264"
```

alienatedsec, 2023-06-30:
Have you executed this on `0.3.7` or `9.3.8c`? @scruben

scruben, 2023-06-30:
0.3.7. Should I try flash and test with 9.3.8c?

alienatedsec, 2023-06-30:
@scruben please try this new version of the launch - I will be adding it to the repo too. There are some additional checks useful to me which are to be printed on the screen

[launch.zip](https://github.com/alienatedsec/yi-hack-v5/files/11917427/launch.zip)

scruben, 2023-06-30:
```
/home/yi-hack-v5/script # ./launch.sh 
cat: can't open '/home/yi-hack-v5/version': No such file or directory
Using model: yi_home
Version on cam: 
Version on sd: 0.3.7
Specify the operation (k for kill, l for launch): k
/home/yi-hack-v5/script # ./launch.sh 
cat: can't open '/home/yi-hack-v5/version': No such file or directory
Using model: yi_home
Version on cam: 
Version on sd: 0.3.7
Specify the operation (k for kill, l for launch): l
Specify the settings (high, low, or both): high 
Enable audio? (y/n): n
Execute in debug mode? (y/n): y
/home/yi-hack-v5/script # Using fifo as output
Resolution high
rRTSPServer: invalid option -- a
rRTSPServer: invalid option -- u
rRTSPServer: invalid option -- w
debug on

"ch0_0.h264" stream, from the file "/tmp/h264_high_fifo"
Play this stream using the URL "rtsp://192.168.0.29/ch0_0.h264
```
I'll try this weekend with 9.3.8c, out of curiosity... it wasn't available yet when I originally flashed the cam.

noz1380, 2023-06-30:
> @scruben @noz1380, please execute this script twice (option kill first, and second-time launch) and run it in debug mode. I will need to see the output.
> 
> [REMOVED - an new script [below](https://github.com/alienatedsec/yi-hack-v5/issues/251#issuecomment-1614746870)]

Sorry to be a noob, I'd like to assist, where do I run the script, from ssh terminal on the camera?

alienatedsec, 2023-06-30:
@noz1380 the camera terminal, please. You may need to enable the execution with:
chmod +x launch.sh

Execute: 
./launch.sh

alienatedsec, 2023-06-30:
> I'll try this weekend with 9.3.8c, out of curiosity... it wasn't available yet when I originally flashed the cam.

Its a manual effort - more https://github.com/alienatedsec/yi-hack-v5/discussions/248 or https://github.com/alienatedsec/yi-hack-v5/discussions/257 - I am to release another version `9.3.8d` soon

alienatedsec, 2023-06-30:
Here you go https://github.com/alienatedsec/yi-hack-v5/releases/tag/9.3.8d

scruben, 2023-07-03:
No luck... no debug messages when trying to open the stream in VLC or kaffeine
```
/home/yi-hack-v5/script # ./launch.sh 
Using model: yi_home
Version on cam: 9.3.8d
Version on sd: 9.3.8d
Specify the operation (k for kill, l for launch): k
/home/yi-hack-v5/script # ./launch.sh 
Using model: yi_home
Version on cam: 9.3.8d
Version on sd: 9.3.8d
Specify the operation (k for kill, l for launch): l
Specify the settings (high, low, or both): high
Enable audio? (y/n): n
Execute in debug mode? (y/n): y
/home/yi-hack-v5/script # Using fifo as output
Resolution high
Audio fifo does not exist, disabling audio.
"ch0_0.h264" stream, from the file "/tmp/h264_high_fifo"
Play this stream using the URL "rtsp://192.168.0.29/ch0_0.h264"
```

scruben, 2023-07-03:
Some more info... I disabled RTSP from UI and run the script but with audio... now it doesn't fail immediately on opening the stream, but missing some messages
```/home/yi-hack-v5/script # ./launch.sh 
Using model: yi_home
Version on cam: 9.3.8d
Version on sd: 9.3.8d
Specify the operation (k for kill, l for launch): k
killall: wd_rtsp.sh: no process killed
killall: rRTSPServer: no process killed
killall: h264grabber: no process killed
/home/yi-hack-v5/script # ./launch.sh 
Using model: yi_home
Version on cam: 9.3.8d
Version on sd: 9.3.8d
Specify the operation (k for kill, l for launch): l
Specify the settings (high, low, or both): high
Enable audio? (y/n): y
Execute in debug mode? (y/n): y
/home/yi-hack-v5/script # Using fifo as output
Using fifo as output
AAC audio
Resolution high
Starting aac replicator
```

alienatedsec, 2023-07-03:
Try 0_0 stream

alienatedsec, 2023-07-03:
@scruben 

I have the same camera and it works perfectly, and started giving debug info when the VLC streaming started.

```
/home/yi-hack-v5/script # ./launch.sh
Using model: yi_home
Version on cam: 9.3.8d
Version on sd: 9.3.8d
Specify the operation (k for kill, l for launch): k
killall: wd_rtsp.sh: no process killed
killall: rRTSPServer: no process killed
killall: h264grabber: no process killed
/home/yi-hack-v5/script # ./launch.sh
Using model: yi_home
Version on cam: 9.3.8d
Version on sd: 9.3.8d
Specify the operation (k for kill, l for launch): l
Specify the settings (high, low, or both): high
Enable audio? (y/n): y
Execute in debug mode? (y/n): y
/home/yi-hack-v5/script # Using fifo as output
Using fifo as output
AAC audio
Resolution high
Starting aac replicator
Read first frame: profile 1, sampling_frequency_index 11 => samplingFrequency 8000, channel_configuration 1

"ch0_0.h264" stream, from the file "/tmp/h264_high_fifo"
AAC audio enabled
Play this stream using the URL "rtsp://[REDACTED]/ch0_0.h264"
Read frame: syncword 0xfff, protection_absent 1, frame_length 490
Read frame: syncword 0xfff, protection_absent 1, frame_length 505
Read frame: syncword 0xfff, protection_absent 1, frame_length 516
Read frame: syncword 0xfff, protection_absent 1, frame_length 499
Read frame: syncword 0xfff, protection_absent 1, frame_length 518
Read frame: syncword 0xfff, protection_absent 1, frame_length 507
Read frame: syncword 0xfff, protection_absent 1, frame_length 468
Read frame: syncword 0xfff, protection_absent 1, frame_length 510
Read frame: syncword 0xfff, protection_absent 1, frame_length 512
AudioFramedMemorySource found at x = 0
fReplicator->inputSource() = 0x110fb18
createStreamReplica completed successfully
Sampling frequency: 8000, Num channels: 1, Config string: 1588
Read frame: syncword 0xfff, protection_absent 1, frame_length 535
AudioFramedMemorySource found at x = 0
fReplicator->inputSource() = 0x110fb18
createStreamReplica completed successfully
Sampling frequency: 8000, Num channels: 1, Config string: 1588
Read frame: syncword 0xfff, protection_absent 1, frame_length 529
Read frame: syncword 0xfff, protection_absent 1, frame_length 472
Read frame: syncword 0xfff, protection_absent 1, frame_length 559
Read frame: syncword 0xfff, protection_absent 1, frame_length 526
Read frame: syncword 0xfff, protection_absent 1, frame_length 480
```

roleoroleo, 2023-07-03:
> I noticed that enabling audio is still a problem @roleoroleo
> 
> That is after applying the latest patches for the rRTSPServer.

Ok, I will check the code.

closed 2023-08-10 by github-actions[bot]