alienatedsec/yi-hack-v5 #261: Recording without Cloud not working

cela96, 2023-07-03:
**Describe the bug**
I have enabled the "Disable Cloud" option and enabled the "Recording without Cloud" option, but despite this, the camera is not recording on the SD card.

**Expected behaviour**
The camera should be recording the videos onto the SD card.

**Screenshots**
[image]

**Set Up Details:**
 - Firmware Version: [Pre-Release 9.3.8d]
 - Device: [yi_outdoor]
 - Model: [4CUS]

**Additional context**
I am fairly certain that the issue is also present with the firmware [0.3.7 - BETA], but I cannot test it at the moment.

I'm not sure if it can be related, but the RTSP stream randomly interrupts for about 5 seconds every 2-3 minutes. This happens with both the cloud enabled and disabled.

alienatedsec, 2023-07-03:
@cela96 thanks for testing. I'll need to look into it a bit closer.

alienatedsec, 2023-07-12:
@cela96 Can you provide an output of the `fdisk -l` ? Is the recording working when the cloud is enabled?

cela96, 2023-07-15:
Yes, the recording works correctly when cloud is enabled.

```/home/yi-hack-v5 # fdisk -l
Disk /dev/romblock0: 0 MB, 262144 bytes, 512 sectors
0 cylinders, 255 heads, 63 sectors/track
Units: sectors of 1 * 512 = 512 bytes

Disk /dev/romblock0 doesn't contain a valid partition table
Disk /dev/mtdblock0: 0 MB, 262144 bytes, 512 sectors
0 cylinders, 255 heads, 63 sectors/track
Units: sectors of 1 * 512 = 512 bytes

Disk /dev/mtdblock0 doesn't contain a valid partition table
Disk /dev/romblock1: 0 MB, 65536 bytes, 128 sectors
0 cylinders, 255 heads, 63 sectors/track
Units: sectors of 1 * 512 = 512 bytes

Disk /dev/romblock1 doesn't contain a valid partition table
Disk /dev/mtdblock1: 0 MB, 65536 bytes, 128 sectors
0 cylinders, 255 heads, 63 sectors/track
Units: sectors of 1 * 512 = 512 bytes

Disk /dev/mtdblock1 doesn't contain a valid partition table
Disk /dev/romblock2: 0 MB, 65536 bytes, 128 sectors
0 cylinders, 255 heads, 63 sectors/track
Units: sectors of 1 * 512 = 512 bytes

Disk /dev/romblock2 doesn't contain a valid partition table
Disk /dev/mtdblock2: 0 MB, 65536 bytes, 128 sectors
0 cylinders, 255 heads, 63 sectors/track
Units: sectors of 1 * 512 = 512 bytes

Disk /dev/mtdblock2 doesn't contain a valid partition table
Disk /dev/romblock3: 1 MB, 1638400 bytes, 3200 sectors
0 cylinders, 255 heads, 63 sectors/track
Units: sectors of 1 * 512 = 512 bytes

Disk /dev/romblock3 doesn't contain a valid partition table
Disk /dev/mtdblock3: 1 MB, 1638400 bytes, 3200 sectors
0 cylinders, 255 heads, 63 sectors/track
Units: sectors of 1 * 512 = 512 bytes

Disk /dev/mtdblock3 doesn't contain a valid partition table
Disk /dev/romblock4: 1 MB, 1310720 bytes, 2560 sectors
0 cylinders, 255 heads, 63 sectors/track
Units: sectors of 1 * 512 = 512 bytes

Disk /dev/romblock4 doesn't contain a valid partition table
Disk /dev/mtdblock4: 1 MB, 1310720 bytes, 2560 sectors
0 cylinders, 255 heads, 63 sectors/track
Units: sectors of 1 * 512 = 512 bytes

Disk /dev/mtdblock4 doesn't contain a valid partition table
Disk /dev/romblock5: 12 MB, 13303808 bytes, 25984 sectors
1 cylinders, 255 heads, 63 sectors/track
Units: sectors of 1 * 512 = 512 bytes

Disk /dev/romblock5 doesn't contain a valid partition table
Disk /dev/mtdblock5: 12 MB, 13303808 bytes, 25984 sectors
1 cylinders, 255 heads, 63 sectors/track
Units: sectors of 1 * 512 = 512 bytes

Disk /dev/mtdblock5 doesn't contain a valid partition table
Disk /dev/romblock6: 0 MB, 65536 bytes, 128 sectors
0 cylinders, 255 heads, 63 sectors/track
Units: sectors of 1 * 512 = 512 bytes

Disk /dev/romblock6 doesn't contain a valid partition table
Disk /dev/mtdblock6: 0 MB, 65536 bytes, 128 sectors
0 cylinders, 255 heads, 63 sectors/track
Units: sectors of 1 * 512 = 512 bytes

Disk /dev/mtdblock6 doesn't contain a valid partition table
Disk /dev/romblock7: 0 MB, 65536 bytes, 128 sectors
0 cylinders, 255 heads, 63 sectors/track
Units: sectors of 1 * 512 = 512 bytes

Disk /dev/romblock7 doesn't contain a valid partition table
Disk /dev/mtdblock7: 0 MB, 65536 bytes, 128 sectors
0 cylinders, 255 heads, 63 sectors/track
Units: sectors of 1 * 512 = 512 bytes

Disk /dev/mtdblock7 doesn't contain a valid partition table
Disk /dev/mmcblk0: 59 GB, 63864569856 bytes, 124735488 sectors
244579 cylinders, 255 heads, 2 sectors/track
Units: sectors of 1 * 512 = 512 bytes

Device       Boot StartCHS    EndCHS        StartLBA     EndLBA    Sectors  Size Id Type
/dev/mmcblk0p1    0,1,1       1023,254,2           2  124735487  124735486 59.4G 83 Linux```

cela96, 2023-07-15:
I found this error in the root of the SD card, but the card has always worked formatted this way. It was formatted directly by the camera.

```
/tmp/sd # cat fat32error.txt
Your SD card is not FAT32 to load the latest firmware correctly. Go to https://github.com/alienatedsec/yi-hack-v5#getting-started
```

I have also noticed that there is a huge file called wd_rtsp.log
It's probably not relevant to the issue, but in the long run, it could consume a few gigabytes.

`-rwxr-xr-x    1 root     root      116.9M Jul 15 11:30 wd_rtsp.log`

Porkenstien, 2023-07-19:
Howdy,
Camera Details-
Yi Outdoor Cam(circa 2017)
Firmware: 3.0.0.0D
Hardware ID: 9CUS
Yi-Hack-V5 Ver: 9.3.8c

would like to add to this thread as I am seeing this issue also. As near as I can tell it was 2 weeks after updating to Pre Release 9.3.8c and the recording to sd stopped.

Output of fdisk -l:
`Disk /dev/romblock0: 0 MB, 262144 bytes, 512 sectors
0 cylinders, 255 heads, 63 sectors/track
Units: sectors of 1 * 512 = 512 bytes

Disk /dev/romblock0 doesn't contain a valid partition table
Disk /dev/mtdblock0: 0 MB, 262144 bytes, 512 sectors
0 cylinders, 255 heads, 63 sectors/track
Units: sectors of 1 * 512 = 512 bytes

Disk /dev/mtdblock0 doesn't contain a valid partition table
Disk /dev/romblock1: 0 MB, 65536 bytes, 128 sectors
0 cylinders, 255 heads, 63 sectors/track
Units: sectors of 1 * 512 = 512 bytes

Disk /dev/romblock1 doesn't contain a valid partition table
Disk /dev/mtdblock1: 0 MB, 65536 bytes, 128 sectors
0 cylinders, 255 heads, 63 sectors/track
Units: sectors of 1 * 512 = 512 bytes

Disk /dev/mtdblock1 doesn't contain a valid partition table
Disk /dev/romblock2: 0 MB, 65536 bytes, 128 sectors
0 cylinders, 255 heads, 63 sectors/track
Units: sectors of 1 * 512 = 512 bytes

Disk /dev/romblock2 doesn't contain a valid partition table
Disk /dev/mtdblock2: 0 MB, 65536 bytes, 128 sectors
0 cylinders, 255 heads, 63 sectors/track
Units: sectors of 1 * 512 = 512 bytes

Disk /dev/mtdblock2 doesn't contain a valid partition table
Disk /dev/romblock3: 1 MB, 1638400 bytes, 3200 sectors
0 cylinders, 255 heads, 63 sectors/track
Units: sectors of 1 * 512 = 512 bytes

Disk /dev/romblock3 doesn't contain a valid partition table
Disk /dev/mtdblock3: 1 MB, 1638400 bytes, 3200 sectors
0 cylinders, 255 heads, 63 sectors/track
Units: sectors of 1 * 512 = 512 bytes

Disk /dev/mtdblock3 doesn't contain a valid partition table
Disk /dev/romblock4: 1 MB, 1310720 bytes, 2560 sectors
0 cylinders, 255 heads, 63 sectors/track
Units: sectors of 1 * 512 = 512 bytes

Disk /dev/romblock4 doesn't contain a valid partition table
Disk /dev/mtdblock4: 1 MB, 1310720 bytes, 2560 sectors
0 cylinders, 255 heads, 63 sectors/track
Units: sectors of 1 * 512 = 512 bytes

Disk /dev/mtdblock4 doesn't contain a valid partition table
Disk /dev/romblock5: 12 MB, 13303808 bytes, 25984 sectors
1 cylinders, 255 heads, 63 sectors/track
Units: sectors of 1 * 512 = 512 bytes

Disk /dev/romblock5 doesn't contain a valid partition table
Disk /dev/mtdblock5: 12 MB, 13303808 bytes, 25984 sectors
1 cylinders, 255 heads, 63 sectors/track
Units: sectors of 1 * 512 = 512 bytes

Disk /dev/mtdblock5 doesn't contain a valid partition table
Disk /dev/romblock6: 0 MB, 65536 bytes, 128 sectors
0 cylinders, 255 heads, 63 sectors/track
Units: sectors of 1 * 512 = 512 bytes

Disk /dev/romblock6 doesn't contain a valid partition table
Disk /dev/mtdblock6: 0 MB, 65536 bytes, 128 sectors
0 cylinders, 255 heads, 63 sectors/track
Units: sectors of 1 * 512 = 512 bytes

Disk /dev/mtdblock6 doesn't contain a valid partition table
Disk /dev/romblock7: 0 MB, 65536 bytes, 128 sectors
0 cylinders, 255 heads, 63 sectors/track
Units: sectors of 1 * 512 = 512 bytes

Disk /dev/romblock7 doesn't contain a valid partition table
Disk /dev/mtdblock7: 0 MB, 65536 bytes, 128 sectors
0 cylinders, 255 heads, 63 sectors/track
Units: sectors of 1 * 512 = 512 bytes

Disk /dev/mtdblock7 doesn't contain a valid partition table
Disk /dev/mmcblk0: 30 GB, 31914983424 bytes, 62333952 sectors
3880 cylinders, 255 heads, 63 sectors/track
Units: sectors of 1 * 512 = 512 bytes

Device       Boot StartCHS    EndCHS        StartLBA     EndLBA    Sectors  Size Id Type
/dev/mmcblk0p1    0,130,3     1023,254,63       8192   62333951   62325760 29.7G  c Win95 FAT32 (LBA)`

However, I do not have fat32error.txt file which is expected as my card shows Win95 FAT32 formatting.
I do have wd_rtsp.log in /tmp/sd  and this file is around 95mg in size.

Hope this helps to resolve issue.

alienatedsec, 2023-07-19:
Please read through #267 for more details on the video recording problem

young747, 2023-07-28:
Camera Details-

Firmware Version 	0.3.8
Baseline Firmware Version 	0.3.8
Base Version 	2.1.0.0E_201809191630
Model Suffix 	yi_home_1080p
Hardware ID 	48US

I have enabled the "Disable Cloud" option and enabled the "Recording without Cloud" option,
 but despite this, the camera is not recording on the SD card.

I fixed such as the following.

in /tmp/sd/yi-hack-v5/script/system.sh  (line 170 ~ 185)
in /home/yi-hack-v5/script/system.sh  (line 170 ~ 185)

replace
```c
if [[ $(get_config DISABLE_CLOUD) == "yes" ]] ; then
    (
        cd /home/app
        killall dispatch
        LD_PRELOAD=/home/yi-hack-v5/lib/ipc_multiplex.so ./dispatch &
        sleep 2
        LD_LIBRARY_PATH="/home/yi-hack-v5/lib:/lib:/home/lib:/home/app/locallib:/home/hisiko/hisilib" ./rmm &
        sleep 4
        ./cloud &
    )
elif [[ $(get_config REC_WITHOUT_CLOUD) == "yes" ]] ; then
    (
        cd /home/app
        ./mp4record &
    )
fi
``` 

with
```c
if [[ $(get_config DISABLE_CLOUD) == "yes" ]] ; then
    (
        cd /home/app
        killall dispatch
        LD_PRELOAD=/home/yi-hack-v5/lib/ipc_multiplex.so ./dispatch &
        sleep 2
        LD_LIBRARY_PATH="/home/yi-hack-v5/lib:/lib:/home/lib:/home/app/locallib:/home/hisiko/hisilib" ./rmm &
        sleep 4
        ./cloud &

        if [[ $(get_config REC_WITHOUT_CLOUD) == "yes" ]] ; then
            cd /home/app
            ./mp4record &
        fi
    )
fi
``` 

and then reboot.

alienatedsec, 2023-07-30:
`0.3.9` released

closed 2023-07-30 by alienatedsec