alienatedsec/yi-hack-v5 #262: Problems with yi dome 1080p 45US

nordes85, 2023-07-03:
Firmware does not work on Yi Dome 1080p 45USY7E7I4.
on the camera it is flashed, but nothing works.
in web gui it works only the first time, but then that's it. when I restart I no longer see the web gui.
I returned to the V4 with problems with the PTZ.

alienatedsec, 2023-07-03:
> Firmware does not work on Yi Dome 1080p 

Which firmware? Can you show me how it does not work, please?

nordes85, 2023-07-03:
Firmware yi_dome_1080p_0.3.7.tgz

tell me how I can show you and proceed. screenshot?

alienatedsec, 2023-07-03:
> tell me how I can show you and proceed. screenshot?

Whichever way you believe is the best to show.

BeastleeUK, 2023-07-15:
I think the problem is that there's insufficient space in the home partition so this bins out part way through, the total content is 14Mb and the partition is 12.7MB. . I've tried to make it work by copying in the relevant parts to home and can get the web service to run but nothing works once in there.

Is the SD Card supposed to mount to /tmp/sd? Mine doesn't and I tried copying the content there to but unfortunately that's also too small to work ad causes the cam to reboot and lose the content again.

alienatedsec, 2023-07-22:
@nordes85 @BeastleeUK try the latest firmware

BeastleeUK, 2023-07-23:
> @nordes85 @BeastleeUK try the latest firmware

Sorry, same issue as the Yi-hack-v5 folder is still bigger than the partition. Iwas able to get to connected via FTP but no SSH or web. As soon as I try to load the yi-hack-v5 folder to /tmp/sd it copies a few files then crashes and the blue wifi light starts to flash. Eventually it reboots and reconnects but the /tmp/sd is empty again.

 [image]

Pol259, 2023-08-06:
The same problem here. YI cam dome 1080p 45USYBPHB version yi-hack-v5 - 0.3.9 (0.3.8 was the same)
Only ssh works. Ftp does not. Web interface also doesn't work.

[image]

alienatedsec, 2023-08-06:
@Pol259 what is the output of the `ps`?

BeastleeUK, 2023-08-08:
@alienatedsec here's mine, if it helps:
[image]

alienatedsec, 2023-08-08:
@BeastleeUK your hack is not installed properly. It looks like the yi-hack-v5 folder does not exist on your SD card

Can you run ls command inside /tmp folder?

alienatedsec, 2023-08-08:
@Pol259 try http://ip:8080

BeastleeUK, 2023-08-08:
> @BeastleeUK your hack is not installed properly. It looks like the yi-hack-v5 folder does not exist on your SD card
> 
> Can you run ls command inside /tmp folder?

That seems to be exactly the problem but I can't find logs to see why. Output of ls -al is 
[image]

The SD card is 8GB formatted to FAT32, it must be seeing it intially to start the install but something is blocking the mount of it once it reboots.

Pol259, 2023-08-08:
@BeastleeUK <https://github.com/BeastleeUK> I had exactly the same issue,
the only way to make the camera mount the as properly is to format the card
through the App. Then mount it on windows/parallels (on macOS I was not
able to initialise). Copy the folder yi-hack-v5 without formatting the
card. After that everything should work.

BeastleeUK, 2023-08-08:
> @BeastleeUK <https://github.com/BeastleeUK> I had exactly the same issue, the only way to make the camera mount the as properly is to format the card through the App. Then mount it on windows/parallels (on macOS I was not able to initialise). Copy the folder yi-hack-v5 without formatting the card. After that everything should work. W dniu wt., 8.08.2023 o 21:22 BeastleeUK ***@***.***> napisał(a):
> -- Sincerely, *Piotr Grzymski*

@Pol259 you've just fixed it for me!  I used my original 64GB card, reformatted it in the app, added the folder to it and it works perfectly.

@alienatedsec it looks like on this device we should use the original card and not swap to a <16GB FAT32 as instructed.

Thanks for your help, time to get testing :)

alienatedsec, 2023-08-08:
@nordes85 please can you check as well?

alienatedsec, 2023-08-09:
> it looks like on this device we should use the original card and not swap to a <16GB FAT32 as instructed.

@BeastleeUK as long as you have the baseline version 0.3.8 then it is ok. The camera may not see the SD for many reasons but it needs FAT32 to load the 0.3.8 baseline firmware. Whatever you do after, it's to ensure the card and camera can record the content.

nordes85, 2023-08-09:
> @nordes85 please can you check as well?

@alienatedsec I'm doing some tests in these days and I'll keep you updated
Ty

closed 2023-09-16 by github-actions[bot]