alienatedsec/yi-hack-v5 #276: How to update Basline, its not shown in WebUI.

Rushmed, 2023-08-03:
Hi.

i updated to 3.8.0 but in WebUI no Basline version is shown.
The script launch.sh is not present in /scripts.

What to do now?

How to updtae Baseline?

Thanks in further.

Camversion: Yi Outdoor

alienatedsec, 2023-08-03:
@Rushmed please follow #267

YoYoost, 2023-08-04:
Just to be sure: my Hack (SD) Firmware Version is:0.4.0a. Should my baseline be 0.3.8 regardless of what (pre)release I am using?

alienatedsec, 2023-08-04:
@YoYoost that is correct

closed 2023-08-04 by Rushmed

alienatedsec, 2023-08-11:
> Is the empty Baseline Firmware Version in the web interface a problem?

It is. Please load 0.3.8 first and then upgrade to 0.4.0 via the web interface.

grungeguitarist, 2023-08-11:
Please be a bit more detailed... What should I do if "you find fat32error.txt in /tmp/sd then your version file does not exist and your firmware was flashed incorrectly."?

> It is. Please load 0.3.8 first and then upgrade to 0.4.0 via the web interface.

What I tried now is copying both 0.3.8 files and 0.3.8 "yi-hack-v5" to the SD and repeated the procedure.

Now the UI says:

Firmware Version | 0.3.8
Baseline Firmware Version

So again. Baseline Firmware is empty in the web interface. If this is not correct then I should fix this before uprading to 0.4.0 via the web interface. But how?

grungeguitarist, 2023-08-11:
Ok, I think I (hopefully) solved the problem by re-initiating the flashing process using a downgrade to 0.3.7, then back to 0.3.8:

1.) I did a downgrade to 0.3.7 (copied both 0.3.7 files and 0.3.7 "yi-hack-v5" folder to the formatted SD, then booted).
2.) Then I upgraded to 0.3.8 (copied both 0.3.8 files and 0.3.8 "yi-hack-v5" folder to the formatted SD, then booted).
3.) Then I updated to 0.4.0 via web interface.
4.) SD card format in Yi App
5.) Copied 0.4.0 "yi-hack-v5" folder to the SD via ftp to "/tmp/sd", then booted.

Now the web interface says correctly:

Hack (SD) Firmware Version	0.4.0
Baseline Firmware Version	0.3.8