alienatedsec/yi-hack-v5 #277: Recording without cloud still not possible 0.3.9

Rushmed, 2023-08-04:
Hi Sir.

I succesfully updated Baseline to 0.3.8 and YiHackV5 to 0.3.9 but recording only works with enabled cloud connection.
[image]

Any idea?

alienatedsec, 2023-08-04:
Have you enabled `NTPD` and restarted the cam afterwards?

Occasionally, I found it does not record after restarting if the time is not synced before the `mp4record` execution. We could move `ntpd` further up the chain.
https://github.com/alienatedsec/yi-hack-v5/blob/08385ac5c234d8e68063bb51364d0d878274a059/src/static/static/home/yi-hack-v5/script/system.sh#L216-L219

You can try yourself by moving the above code in the `system.sh` to line L152
https://github.com/alienatedsec/yi-hack-v5/blob/08385ac5c234d8e68063bb51364d0d878274a059/src/static/static/home/yi-hack-v5/script/system.sh#L152

Rushmed, 2023-08-05:
Yes NTPD is enabled and many restarts are done.
Edited system.sh as you recomented but still no recording without cloud. With cloud enabled recording works.
Of course recording without cloud is enabled.
Also checked on page=snapshot that the time is correctly shown.

alienatedsec, 2023-08-05:
Another thing you could try is to free up more space on the SD card. Set it to 30% and restart again.

Rushmed, 2023-08-05:
Tried without success. Its a complete fresh SD installed. No files in record folder.

alienatedsec, 2023-08-05:
I'll check on mine to confirm but it may not be today

Rushmed, 2023-08-05:
Ur welcome Sir!

Rushmed, 2023-08-08:
@alienatedsec Any result?

alienatedsec, 2023-08-08:
I found a problem with the NTPD. It stops updating the time and currently my local time stopped on a day in July. My camera stopped recording without cloud on the 1st August. Will try to fix as soon as I can.

alienatedsec, 2023-08-08:
The `ntpd` was a false flag - it turns out that `./cloud &` is not executing or crashing shortly after for some reason.

I got the `cloudAPI_fake` amended slightly, which seems to respond well - `./cloud &` works ok when executed manually from the console, but the `system.sh` needs more work to check on the `./cloud &` execution steps.

alienatedsec, 2023-08-09:
Replace the files within `/tmp/sd/yi-hack-v5` with those [yi-hack-v5.zip](https://github.com/alienatedsec/yi-hack-v5/files/12297009/yi-hack-v5.zip) and restart the camera. Let me know and I will follow up with another pre-release.

I also committed those changes here https://github.com/alienatedsec/yi-hack-v5/commit/35b1a8c4a720613736b1d767666d532ad28f03d7

Porkenstien, 2023-08-09:
Ok,
did update and uploaded the files in previous post by @alienatedsec. No change in recording, did the dance of turning Cloud  features On/Off, recordings still only work with Cloud turned on.

FYI, after uploading the updates I noticed that the date stamp on most of the files  are 12-31-1979 ???

alienatedsec, 2023-08-09:
@Porkenstien it has been working for me out of the box. 

Please can you post any screenshots? The config and cloudAPI_fake files?

Porkenstien, 2023-08-09:
Ok,
Screenshots showing date stamp irregularities, latest files uploaded and Events Page which shows nothing even after dancing and clapping in front of camera:
[image]
[image]
[image]

Porkenstien, 2023-08-09:
One more screenshot of my Config Page:
[image]

alienatedsec, 2023-08-09:
@Porkenstien you copied files to a wrong location

Please make sure you replace files in /tmp/sd

Porkenstien, 2023-08-09:
My apologies @alienatedsec !
It is working after putting the files in the proper directories. (A reminder that I shouldn't mess with tech' until after the my first pot of coffee!)
Thanks for your hard work!

Rushmed, 2023-08-09:
Reinstalled your new release 0.4.0c over WebUi and still everything fine.
Than you Sir!

alienatedsec, 2023-08-10:
While it is working better than before, I need to tune the `cloudAPI_fake` and update the `wd_rtsp.sh` to monitor the `./cloud &` process as it crashed overnight and there is nothing to make it recover.

As of now - it is work in progress

alienatedsec, 2023-08-10:
Following `0.4.0d` the `./cloud &` monitoring is working correctly and I can see some anomalies already. The time and date keep changing at the restart or the `NTPD` reliability is not there. It could be the reason why `./cloud` is crashing. I will keep monitoring.
```
2023-08-10 07:59:09 - Starting RTSP watchdog...
2023-08-10 07:59:11 - No running processes, restarting ./cloud & ...
2023-08-10 08:43:23 - No running processes, restarting ./cloud & ...
2023-08-10 08:48:48 - No running processes, restarting ./cloud & ...
2023-08-10 08:50:25 - No running processes, restarting ./cloud & ...
2023-08-08 22:59:16 - Starting RTSP watchdog...
2023-08-08 23:00:43 - No running processes, restarting ./cloud & ...
```

Porkenstien, 2023-08-10:
Are we seeing a buggy clock on the camera? Maybe a heat or cpu load giving the onboard clock issues.
Has anyone other than myself seen the date stamp on files or directories show year 1969?

alienatedsec, 2023-08-10:
> Has anyone other than myself seen the date stamp on files or directories show year 1969?

@Porkenstien 
I don't think that is an issue - my `/tmp/sd/yi-hack-v5` folder is dated 1980. It's likely because of the build process and steps to change ownership to avoid permission issues.

[image]

alienatedsec, 2023-08-11:
Whenever I restart the camera (the line `Starting RTSP watchdog...`), the time is going back to 8th August. Still investigating but the `./cloud &` process is not crashing as of now.

```
2023-08-10 07:59:11 - No running processes, restarting ./cloud & ...
2023-08-10 08:43:23 - No running processes, restarting ./cloud & ...
2023-08-10 08:48:48 - No running processes, restarting ./cloud & ...
2023-08-10 08:50:25 - No running processes, restarting ./cloud & ...
2023-08-08 22:59:16 - Starting RTSP watchdog...
2023-08-08 23:00:43 - No running processes, restarting ./cloud & ...
2023-08-08 22:59:14 - Starting RTSP watchdog...
```

g1za, 2023-08-11:
Just updated to regular v. 0.4.0. I can tell local recording is working ok, even with cloud disabled. 
BUT the time is wrong; it is 2 hours behind (Timezone correctly set in the settings).
Date is correct.

Porkenstien, 2023-08-11:
> Whenever I restart the camera (the line `Starting RTSP watchdog...`), the time is going back to 8th August. Still investigating but the `./cloud &` process is not crashing as of now.
>
> ```
> 2023-08-10 [...]

Would 08/08/2023 be the last reboot of the camera? Is the process or caching for the process retaining a date stamp? Could calling to kill ./cloud or RTSP process(es) before restarting clear any previous time stamps?

closed 2023-09-25 by github-actions[bot]