alienatedsec/yi-hack-v5 #278: Proxychain-ng - config and lib unable to locate - v0.4.0a

alienatedsec, 2023-08-06:
It works - at least partially - so far it shows results for `Test without proxy:` 

I need to fix the proxychain execution as it feels like the final compilation is wrong - leave it with me, but you should see something after pressing `save and test configuration` button

_Originally posted by @alienatedsec in https://github.com/alienatedsec/yi-hack-v5/discussions/273#discussioncomment-6648851_

alienatedsec, 2023-08-06:
@voquangminh I found an issue, but at least it works now
```
/home/yi-hack-v5 # cat /tmp/sd/yi-hack-v5/etc/proxychains.conf
random_chain

chain_len = 1

tcp_read_time_out 15000
tcp_connect_time_out 8000

[ProxyList]
socks4 61.171.50.169 6688
socks4 218.78.65.202 6688
socks5 1.180.0.242 7302
socks5 60.190.195.146 10800
/home/yi-hack-v5 #  proxychains4 wget -O- -q http://ipinfo.io
[proxychains] config file found: /tmp/sd/yi-hack-v5/etc/proxychains.conf
[proxychains] preloading /tmp/sd/yi-hack-v5/lib/libproxychains4.so
[proxychains] DLL init: proxychains-ng 4.16
[proxychains] Random chain  ...  61.171.50.169:6688  ...  timeout
[proxychains] Random chain  ...  218.78.65.202:6688  ...  timeout
[proxychains] Random chain  ...  60.190.195.146:10800  ...  34.117.59.81:80  ...  OK
{
  "ip": "60.190.195.146",
  "city": "Shaoxing",
  "region": "Zhejiang",
  "country": "CN",
  "loc": "30.0024,120.5786",
  "org": "AS4134 CHINANET-BACKBONE",
  "timezone": "Asia/Shanghai",
  "readme": "https://ipinfo.io/missingauth"
}/home/yi-hack-v5 #
/home/yi-hack-v5 #
```

Please try this package and edit your `proxychains.conf` manually for now
[yi-hack-v5.zip](https://github.com/alienatedsec/yi-hack-v5/files/12270303/yi-hack-v5.zip)

alienatedsec, 2023-08-06:
[image]

Possible to work with the WebUI with this package - version `0.4.0b` to be released tomorrow
[yi-hack-v5.zip](https://github.com/alienatedsec/yi-hack-v5/files/12270798/yi-hack-v5.zip)

voquangminh, 2023-08-07:
@alienatedsec thank u for support, I will try new ver and test the config.

alienatedsec, 2023-08-07:
@voquangminh version `0.4.0b` released

closed 2023-08-07 by alienatedsec