alienatedsec/yi-hack-v5 #279: Releases assets missing firmware files

S0NN1, 2023-08-08:
It appears that in latest pre-releases and latest release the firmware files (rootfs/home) are not bundled in the asset tgz file.
Is this intended? 
Should I use firmwares from older releases? 
If so, shouldn't it be specified in the documentation?
[image]

alienatedsec, 2023-08-08:
@S0NN1 Apologies if that is not clear at first. Some details are in the release notes here https://github.com/alienatedsec/yi-hack-v5/releases and here #267 

In short, `0.3.8` is the last version to provide `rootfs` and `home` files, which is to formulate the baseline firmware and the SD-based firmware. Therefore, versions `0.3.9` onwards will have the `yi-hack-v5` only. However, I recommend upgrading to the pre-release `0.4.0b` after the baseline `0.3.8`.

alienatedsec, 2024-02-04:
Please follow and test if needed https://github.com/alienatedsec/yi-hack-v5/releases/tag/0.4.1e