alienatedsec/yi-hack-v5 #283: [47US] from 0.3.7 to 0.38-9 or 0.4.0 camera/RTSP freezes when motion is detected.

franciscot, 2023-08-11:
**Describe the bug**
Upgraded from 0.3.7 to 0.3.8 0.3.9 or 0.4.0, RTSP freezes when motion is detected.  Maybe when the camera starts recording?

**To Reproduce**
Steps to reproduce the behaviour:
1. Format SD with Yi app. Install firmware yi hack 5. Config by default.
2. Only activate RTSP in low mode without audio.
3. Play RSTP in a video broswer or frigate for example. The video is fine. Without cuts.
4. If any motion is detected, after 3 seconds the video freezes and there is no access to the camera (connected by ssh). After 30 seconds it returns to normal and continues streaming video (maybe it restarts?).

**Expected behaviour**
Like with 0.3.7 firmware I can watch the streaming video when motion is detected without any problem likes cuts, freezes or restarts.

**Set Up Details (please complete the following information):**
- Hack (SD) Firmware Version | 0.4.0
- Baseline Firmware Version | 0.3.8
- Base Version | 1.8.7.0F_201809191400
- Model Suffix | yi_home
- Hardware ID | 47US

**Technical Details**
I used launch.sh to try to debug but I can't get anything. I dont know because it freezes or there are no logs.

alienatedsec, 2023-08-22:
@franciscot please can you try with this file?

[h264grabber.gz](https://github.com/alienatedsec/yi-hack-v5/files/12408351/h264grabber.gz)

franciscot, 2023-08-25:
> @franciscot please can you try with this file?
> 
> [h264grabber.gz](https://github.com/alienatedsec/yi-hack-v5/files/12408351/h264grabber.gz)

I did a clean install with 0.40 and 0.38 firmware with  h264grabber file replaced  on SD Card/yi-hack-v5/bin/
Same problem.

franciscot, 2023-08-25:
New test **without enabling RSTP**:
- I did a clean install 0.4.0/0.38.
- I just changed hostname and enabled swapfile.
- I connected via ssh
- I started a move event.
- The ssh was slow and froze.
- The http connection was lost.

New test **disabling cloud with video recording enabled**.
It didn't freeze when any motion was detected and video was recorded to SD card.

Then I tried enabling RSTP without cloud and video recording enable. No freezes again.

It seems to be a problem when starting a motion event with the cloud enabled.

P.S. With 0.3.7 I use cloud enbale without problem.

alienatedsec, 2023-08-29:
Please update to `0.4.1a`

franciscot, 2023-09-09:
> Please update to `0.4.1a`

Same problem. If I disable the cloud there are no freezes.

I could see that the system load is about 4. When it freezes the load goes up to about 15. 

`17:33:14 up 12 min,  load average: 4.25, 4.11, 2.56`