alienatedsec/yi-hack-v5 #286: SSH - authorized_keys support

lenisko, 2023-08-18:
SSH keys support would be great

pmirshad, 2023-08-20:
SSH `authorized_keys` support works for me on [release/0.4.0](https://github.com/alienatedsec/yi-hack-v5/releases/tag/0.4.0).

```jsonc
# Device info.
"name"           : "yi-hack-v5",
"fw_version"     : "0.4.0",
"basefw_version" : "0.3.8",
"home_version"   : "2.1.0.0E_201809191630",
"model_suffix"   : "yi_home_1080p",
"hardware_id"    : "18US",
```

```bash
# Login without SSH keys.
$ ssh root@192.168.0.13
The authenticity of host '192.168.0.13 (192.168.0.13)' can't be established.
ECDSA key fingerprint is SHA256:FWpGpO65psqIse5sf9iZGmQxm52hwy8dZti9+XBoDCw.
Are you sure you want to continue connecting (yes/no/[fingerprint])? yes
Warning: Permanently added '192.168.0.13' (ECDSA) to the list of known hosts.
root@192.168.0.13's password:

      _     _           _           ___
  _ _|_|___| |_ ___ ___| |_ ___ _ _|  _|
 | | | |___|   | .'|  _| '_|___| | |__ |
 |_  |_|   |_|_|__,|___|_,_|    \_/|___|
 |___|
 -----------------------------------------------------
  yi-hack-v5 - 0.4.0
 -----------------------------------------------------
 WARNING: Cannot check for updates from GitHub.

 The camera might not be connected to the internet
 or, if it has just been turned on, the updater
 hasn't contacted the server yet.
 -----------------------------------------------------

/home/yi-hack-v5 # exit
Connection to 192.168.0.13 closed.

# Add SSH key to camera.
$ ssh-copy-id -i ~/.ssh/id_rsa.pub root@192.168.0.13
/usr/bin/ssh-copy-id: INFO: Source of key(s) to be installed: "/home/<user>/.ssh/id_rsa.pub"
/usr/bin/ssh-copy-id: INFO: attempting to log in with the new key(s), to filter out any that are already installed
/usr/bin/ssh-copy-id: INFO: 1 key(s) remain to be installed -- if you are prompted now it is to install the new keys
root@192.168.0.13's password:

Number of key(s) added: 1

Now try logging into the machine, with:   "ssh 'root@192.168.0.13'"
and check to make sure that only the key(s) you wanted were added.

# Login with SSH keys.
$ ssh root@192.168.0.13

      _     _           _           ___
  _ _|_|___| |_ ___ ___| |_ ___ _ _|  _|
 | | | |___|   | .'|  _| '_|___| | |__ |
 |_  |_|   |_|_|__,|___|_,_|    \_/|___|
 |___|
 -----------------------------------------------------
  yi-hack-v5 - 0.4.0
 -----------------------------------------------------
 WARNING: Cannot check for updates from GitHub.

 The camera might not be connected to the internet
 or, if it has just been turned on, the updater
 hasn't contacted the server yet.
 -----------------------------------------------------

/home/yi-hack-v5 #

/home/yi-hack-v5 # ls -lhR ~/.ssh
/home/yi-hack-v5//.ssh:
total 1K
-rw-------    1 root     root         744 Aug 20 10:30 authorized_keys

/home/yi-hack-v5 # cat ~/.ssh/authorized_keys
ssh-rsa AAAAB3NzaC1y...yg+KjQ== username@hostname
```

lenisko, 2023-08-20:
Oh, I put them in a wrong directory. Thanks for hint!

lenisko, 2023-08-20:
Hmm.. weird, not working to me

`ssh-copy-id` not working like intended:
```
➜  ~ ssh-copy-id -i ~/.ssh/id_rsa.pub root@192.168.1.177
/usr/bin/ssh-copy-id: INFO: Source of key(s) to be installed: "/home/<user>/.ssh/id_rsa.pub"
/usr/bin/ssh-copy-id: INFO: attempting to log in with the new key(s), to filter out any that are already installed
/usr/bin/ssh-copy-id: INFO: 1 key(s) remain to be installed -- if you are prompted now it is to install the new keys
root@192.168.1.177's password:

Number of key(s) added: 1

Now try logging into the machine, with:   "ssh 'root@192.168.1.177'"
and check to make sure that only the key(s) you wanted were added.

➜  ~ ssh 'root@192.168.1.177'
root@192.168.1.177's password:
```

Manually created dir and `authorized_keys`
```
/home/yi-hack-v5 # ls -lah .ssh/authorized_keys ^C
/home/yi-hack-v5 # ls -lhR ~/.ssh
/home/yi-hack-v5//.ssh:
total 1K
-rw-------    1 root     root         743 Aug 20 13:15 authorized_keys
/home/yi-hack-v5 # cat ~/.ssh/authorized_keys
ssh-rsa AAAAB...
➜  ~ ssh root@192.168.1.177 -i ~/.ssh/id_rsa
root@192.168.1.177's password:
```

pmirshad, 2023-08-20:
Can you try this and see why the key is being rejected ? Which release are you on ?
```bash
$ ssh -v root@192.168.1.177 -i ~/.ssh/id_rsa
```

lenisko, 2023-08-22:
Looks like I had to use 
```
ssh -o "PubkeyAcceptedAlgorithms=+ssh-rsa" -v root@192.168.1.177 -i .ssh/id_rsa
```

closed 2023-08-22 by lenisko