alienatedsec/yi-hack-v5 #287: Yi 1080p Home - Not recording

lenisko, 2023-08-18:
**Describe the bug**
Camera is not recording events. Tried with both disabled/enabled cloud (though I prefer disabled). Moved from 0.4.0d where I couldn't get it working as well.

**To Reproduce**
```
Config:
Disable Cloud: ON
Recording without Cloud: ON
RTSP: ON
NTPD: ON
NTP Server: pool.ntp.org
Swap File: ON

Settings:
Switch on/off the camera: ON
Save video when a motion is detected: ON
Detection sensitivity: Medium
Status led: OFF
IR led: OFF
```

**Expected behaviour**
Recording saved to microSD card

**Set Up Details (please complete the following information):**
Firmware Version | 9.3.8e
Base Version | 2.1.0.0E_201809191630
Hardware ID | 48US

```
# ps aux
PID   USER     TIME  COMMAND
    1 root      0:00 init
    2 root      0:00 [kthreadd]
    3 root      0:00 [ksoftirqd/0]
    4 root      0:00 [kworker/0:0]
    5 root      0:00 [kworker/u:0]
    6 root      0:00 [khelper]
    7 root      0:00 [kworker/u:1]
   88 root      0:00 [sync_supers]
   90 root      0:00 [bdi-default]
   91 root      0:00 [kintegrityd]
   93 root      0:00 [kblockd]
  103 root      0:00 [khubd]
  105 root      0:00 [kusbotg]
  199 root      0:00 [kswapd0]
  251 root      0:00 [fsnotify_mark]
  264 root      0:00 [crypto]
  316 root      0:00 [romblock0]
  319 root      0:00 [mtdblock0]
  324 root      0:00 [romblock1]
  327 root      0:00 [mtdblock1]
  332 root      0:00 [romblock2]
  335 root      0:00 [mtdblock2]
  340 root      0:00 [romblock3]
  343 root      0:00 [mtdblock3]
  348 root      0:00 [romblock4]
  351 root      0:00 [mtdblock4]
  356 root      0:00 [romblock5]
  359 root      0:00 [mtdblock5]
  364 root      0:00 [romblock6]
  367 root      0:00 [mtdblock6]
  372 root      0:00 [romblock7]
  375 root      0:00 [mtdblock7]
  398 root      0:00 [cfinteractive]
  403 root      0:00 [kworker/0:1]
  404 root      0:00 [kworker/u:2]
  412 root      0:00 [deferwq]
  413 root      0:00 [jffs2_gcd_mtd4]
  427 root      0:00 udevd --daemon
  483 root      0:00 udevd --daemon
  485 root      0:00 udevd --daemon
  511 root      0:00 [mmcqd/1]
  681 root      0:00 [jffs2_gcd_mtd5]
  717 root      0:00 [flush-31:2]
  718 root      0:00 [flush-179:0]
  777 root      0:01 [RtmpTimerTask]
  778 root      0:01 [RtmpMlmeTask]
  779 root      0:00 [RtmpCmdQTask]
  780 root      0:00 [RtmpWscTask]
  783 root      0:00 ./log_server
  951 root      0:00 [hidog]
 1438 root      0:01 ./dispatch
 1451 root      0:43 ./rmm
 1501 root      0:00 ./cloud
 1517 root      0:00 httpd -p 8080 -h /tmp/sd/yi-hack-v5/www/ -c /tmp/httpd.conf
 1532 root      0:00 dropbear -R -B
 1558 root      0:01 rRTSPServer -r high -a no -p 554 -u -w
 1568 root      0:01 h264grabber -r high -m yi_home_1080p -f
 1573 root      0:00 {wd_rtsp.sh} /bin/sh /tmp/sd/yi-hack-v5/script/wd_rtsp.sh
 1609 root      0:00 /home/base/tools/wpa_supplicant -c/tmp/wpa_supplicant.conf -g/var/run/wpa_supplicant-global -iwlan0 -B
 1617 root      0:00 /usr/sbin/crond -c /var/spool/cron/crontabs/
 1662 root      0:00 ntpd -p pool.ntp.org
 1796 root      0:00 /sbin/udhcpc -i wlan0 -b -s /home/app/script/default.script -x hostname:two
 2126 root      0:00 dropbear -R -B
 2212 root      0:00 -sh
 2219 root      0:00 crond -c /tmp/sd/yi-hack-v5/etc/crontabs
 2220 root      0:00 /bin/login --
 2222 root      0:00 [flush-mtd-unmap]
 2514 root      0:00 sleep 10
 2517 root      0:00 ps aux

/home/yi-hack-v5 # df -h
Filesystem                Size      Used Available Use% Mounted on
/dev/root                 1.3M      1.2M     72.0K  94% /
tmpfs                    15.0M      4.0K     15.0M   0% /dev
/dev/mtdblock5           12.7M      9.2M      3.4M  73% /home
tmpfs                    16.0M      1.6M     14.4M  10% /tmp
/dev/mmcblk0p1           14.7G    173.4M     14.5G   1% /tmp/sd
/dev/mmcblk0p1           14.7G    173.4M     14.5G   1% /tmp/sd/yi-hack-v5/www/record

/home/yi-hack-v5 # free -mh
              total        used        free      shared  buff/cache   available
Mem:             30          15           2           0          12           0
-/+ buffers/cache:           15          14
Swap:            63           0          63
```

alienatedsec, 2023-08-19:
The latest version is `0.4.0` and it records correctly without cloud if all conditions are met. Please don't use those pre-releases prior to `0.4.0` anymore.

raleighlittles, 2023-08-20:
@lenisko I have the same camera as you. Can you explain exactly how you got the firmware installed? The instructions on the README aren't really helpful.

I have:

* Yi 1080p Home camera (S/N starts with 48US)
* 32 GB microSD card, FAT32 formatted
* A download of `yi_home_1080p_0.4.0.tgz`

Would I just copy these files to the root directory of the microSD card?

[image]

lenisko, 2023-08-20:
@raleighlittles 
> You need version 0.3.8 as a minimum to load this release - you can find out more [here ](https://github.com/alienatedsec/yi-hack-v5/releases/tag/0.3.8)and https://github.com/alienatedsec/yi-hack-v5/discussions/267 and https://github.com/alienatedsec/yi-hack-v5/issues/279

alienatedsec, 2023-08-20:
> Would I just copy these files to the root directory of the microSD card?

@raleighlittles 
I would start by reading the release notes if something is not clear. Then, its worth noting that you need the baseline version `0.3.8` prior to installing any later versions. Finally, any versions after `0.3.8` can be loaded from the WebUI.

1. Install `0.3.8` by extracting all files - `rootfs_xx` and `home_xx` and the `yi-hack-v5` folder from the archive to a root of the FAT32 SD Card - this will load the baseload correctly - insert it into the camera and boot up.
2. After flashing and normal operation, you may need to format the SD Card in the YI app - this will make sure your recording is working fine, but it will wipe your `yi-hack-v5` folder.
3. After formatting, it will reboot the camera without a hack folder on your SD but will leave `Telnet` and `FTP` enabled if the baseload was loaded correctly. You could either use it to transfer the `yi-hack-v5` folder or remove the SD card and transfer the folder using your PC.
4. You can load a pre-release or the latest release using WebUI - `0.4.0` is the latest as of today.
5. The below is what is in the root directory of my SD card.
[image]

lenisko, 2023-08-20:
@alienatedsec thanks for great go-through! I had to format microSD using application. Was actually going to ask if it is possible to setup whole camera without touching cloud thing [like this on allwinner-v2](https://github.com/roleoroleo/yi-hack-Allwinner-v2/wiki/Change-WiFi-credentials-(deprecated)) but as I can see some bits are still missing.

alienatedsec, 2023-08-20:
@lenisko please try #193 

Create a file in `/tmp/sd/recover/configure_wifi.cfg`

The format needs to be as follows:
```
wifi_ssid=[your_wifi_essid]
wifi_psk=[your_wifi_password]
```

Reboot

lenisko, 2023-08-20:
It won't work without initial setup of camera in yi-app ?

Tried on fresh camera, and it's blinking blue
[image]

alienatedsec, 2023-08-20:
Has it created an `mtdblock2_xxxxxx.bin` on your SD?

alienatedsec, 2023-08-20:
@lenisko I just found a bug, so need to address it, but if you have a file then we can recover your previous settings if needed.

lenisko, 2023-08-20:
@alienatedsec Yes I saw this file 👍 I don't have any settings in there nor care about previous ones. Unless, they are required to get WiFi and recording working :) I don't plan to use this little with Cloud features.

alienatedsec, 2023-08-20:
@lenisko 
1. Replace the content of `system.sh` and `configure_wifi.sh` in `/tmp/sd/yi-hack-v5/script/` folder - those were amended by this commit https://github.com/alienatedsec/yi-hack-v5/commit/c82ff436b86f67756e74d7159e515621cc73a6bc
2. Make a copy of and rename your file to `mtdblock2_recover.bin`  
3. Move it to `/tmp/sd/recover` folder
4. Reboot
5. The new script will recover your partition and rename to `mtdblock2_recover_done.bin`

I need to investigate the previous process.

> The format needs to be as follows:
> 
> ```
> wifi_ssid=[your_wifi_essid]
> wifi_psk=[your_wifi_password]
> ```

I am about to find out which encoding needs to be used as the manual process is not formatting the partition content correctly :)

lenisko, 2023-08-20:
@alienatedsec dd dump got renamed to `mtdblock2_recover_done.bin` and camera spoke "waiting to connect".

Leaving this one in current state so I will wait for your solution to test cloud-less setup.

Big thanks for your time!