alienatedsec/yi-hack-v5 #298: Disable alarm sound

danielhainich, 2023-09-30:
**Describe the bug**
After Update to 0.4.0, all my cams Are playing an alarm Sound. I cant find anything to disable this behaviour. 

**Expected behaviour**
Disable sound

**Set Up Details (please complete the following information):**
 - Firmware Version: 0.4.0
 - Device: different
 - Model: different

alienatedsec, 2023-09-30:
@danielhainich what is `different` :)

Also, try `0.4.1a` instead

danielhainich, 2023-09-30:
it seems that the alarm sound will be triggered by motion detection. alarm is only triggered when someone is in range of cameraview.

Minims, 2023-09-30:
I have this behavior too, but only if "record to cloud" is disabled.
Would be great to be able to disable this sound on motion detection if cloud is disabled.

Minims, 2023-11-03:
checked in 0.4.1b, I confirmed it's still present. Tested on Yi Outdoor

Minims, 2023-11-03:
I don’t have the issue on yi home 720p might be  related to yi outdoor only

Minims, 2023-11-30:
how can I read logs on the camera when motion is detected to check why alarm sound is on ?

Minims, 2023-12-04:
I just need to launch this command to get logs or check if I get the ALARM Sound ? 
```
ipc_cmd IPC_MOTION_START
```

Minims, 2024-01-04:
Issue still present

Minims, 2024-01-25:
It seems 
ipc_cmd -o on 
did the job.
It was on or off i don’t remember i have tested both

@danielhainich can you give it a try ?

Minims, 2024-01-26:
Hum it does not work. The sound alarm only appears after a motion start only if there were no motion for many minutes.

Minims, 2024-02-08:
Upgraded to 0.4.1e and i still the issue when I Disable Cloud

Minims, 2024-02-08:
Hum seems related to this function in app
« Camera Alarm »

It was ON,  i set to OFF, disabled Cloud, reboot.
Seems better, will check tomorrow.

[image]

ClementMarrast, 2024-02-10:
> Hum seems related to this function in app « Camera Alarm »
> 
> It was ON, i set to OFF, disabled Cloud, reboot. Seems better, will check tomorrow.
> 
> ![IMG_1009](https://private-user-images.githubusercontent.com/1724785/303500045-766206ee-8b69-4baa-8a42-a9bfd0a63602.jpeg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDc1NjA3ODksIm5iZiI6MTcwNzU2MDQ4OSwicGF0aCI6Ii8xNzI0Nzg1LzMwMzUwMDA0NS03NjYyMDZlZS04YjY5LTRiYWEtOGE0Mi1hOWJmZDBhNjM2MDIuanBlZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAyMTAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMjEwVDEwMjEyOVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTgwYWQ5YjUxM2MzMzc0ZDAxZTMxNzg2NzJjNzA2NDVkYWUzM2I5YWFiNWRhZmU0MDk3YTAzOTUzYTBmMzZhMjcmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.0PtuswbId2_5PwycQ3GhqNJCNgKo_YIwVAaPg6WAOL0)

I confirm, it works also for me

databoy2k, 2024-03-12:
I'll just make a comment, partially to keep the issue alive and partially to trigger it for alienatedsec to close. 

Is Yi-Hack-v5 intended to fully replace the use of Yi Cloud? If so, then this should be on the to-do list. If it's just supplemental, then probably close the issue.

My preference, whatever good a non-dev's preference is, would be to replace Yi Cloud. I currently have the camera, even running this firmware, on my IOT network where the devices are fully firewalled from my main network but also the internet. It's actually a royal pain to punch the hole through the firewall to let the camera out long enough to flip the switch, not to mention the fact that I don't have a yi account anymore. I assume I'll have to reset the whole thing to add it to a new account, and I commented on the discord that my cameras aren't in the most accessible positions to begin with.

Either way, the issue can be closed if it's out of scope.

closed 2024-04-19 by github-actions[bot]

johol, 2024-06-16:
I was able to stop beeping by removing the acodec kernel module

# diff -au system.sh.orig system.sh
--- system.sh.orig      2024-06-16 16:45:55.246021590 +0200
+++ system.sh   2024-06-16 16:47:02.250021241 +0200
@@ -471,3 +471,7 @@
 $YI_HACK_PREFIX/script/check_update.sh
 
 crond -c $YI_HACK_PREFIX/etc/crontabs
+
+# Disable audio out. Stop beeping
+rmmod acodec
+

dreimer1986, 2024-08-27:
Ran into the same issue. I got this cam today and started tinkering. First I configured the bare minimum in the cloud app, for better safe than sorry formatted the SD inside the camera via app, too and then flashed the releaase firmware. Btw for me the BEEP was there before any mod was flashed already. I hoped that I can disable that after modding and went on. Then I went to the prerelease as I did not find any way for disabling it. Btw... The j prerelease does not seem to reenable the cloud anymore, but even if it would, this one is a nasty one that should be added to webUI IMO. I now disabled the same kernel module @johol did to skip a full reset just to make it stfu.