alienatedsec/yi-hack-v5 #301: I don't see/can't upgrade to 0.4.1b

g1za, 2023-10-05:
**Describe the bug**
In the service pane the new pre-release is not identified hence it's not possible to upgrade form the camera interface

**To Reproduce**
Steps to reproduce the behaviour:
1. Go to Service tab
2. Check the pre-release version is not recognized/made available for upgrade

**Expected behaviour**
See the new pre-release available and be able to click on the "Upgrade to pre-release" button

**Screenshots**
[image]
[image]

**Set Up Details (please complete the following information):**
 - Hack (SD) Firmware Version: 0.4.1a
 - Baseline Firmware Version: 0.3.8
 - Device: yi_dome
 - Model: 63US

alienatedsec, 2023-10-05:
@g1za I had the same on yi_dome for some reason - download the package and save it in your root folder. Change the name of the file by replacing numbers with `x.x.x`

[yi_dome_0.4.1b.tgz](https://github.com/alienatedsec/yi-hack-v5/releases/download/0.4.1b/yi_dome_0.4.1b.tgz) needs to be renamed to `yi_dome_x.x.x.tgz` - then go back to the upgrade page and press any upgrade button.

g1za, 2023-10-05:
Ok thank you. Maybe we can keep this open for reference till the next release and then close it.
I mentioned your solution in the release discussion for cross-reference.
Thank you

closed 2023-11-12 by github-actions[bot]

g1za, 2023-11-22:
I’m experiencing again the same with 0.4.1c

alienatedsec, 2023-11-22:
> I’m experiencing again the same with 0.4.1c

@g1za are you sure it's not your firewall or DNS playing up?

g1za, 2023-11-22:
I'm not but before 0.4.1b I never had this problem and I changed nothing in my configuration.
I use cloudflare as DNS (1.1.1.1; 1.1.1.2; security.cloudflare-dns.com; one.one.one.one).
Can you please confirm the path being used to download the new release, so I can try to ping it?

g1za, 2023-11-22:
Never mind, you already provided the basic path previously
[https://github.com/alienatedsec/yi-hack-v5/releases/download/0.4.1c/yi_dome_0.4.1c.tgz](url)