alienatedsec/yi-hack-v5 #304: [47US] "Disable cloud" enabled causes mqtt state is always offline (firmware 0.41b)

franciscot, 2023-10-09:
**Describe the bug**
With Disable cloud enabled mqtt's state is always offline. Tested with version 0.40 0.41a & 0.41b

**To Reproduce**
Steps to reproduce the behaviour:
1. Reset to YI-HACK-V5 defaults
2. Config hostname, timezone, user and password, swapfile on. Nothing else.
3. Cofig MQTT. Just changed "General config", "Authentication" and topics-prefix. Nothing else.
4. Reboot
5. The mqtt status of the camera is online.
6. Activate "Disable cloud"
7. Reboot
8. The mqtt status of the camera is offline always.

**Expected behaviour**
Disable cloud doesn't affect to status in mqtt. In Home Assistant mqtt the camera is "not available" because is offline. 

**Set Up Details (please complete the following information):**
- Hack (SD) Firmware Version | 0.4.1b
- Baseline Firmware Version | 0.3.8
- Base Version | 1.8.7.0F_201809191400
- Model Suffix | yi_home
- Hardware ID | 47US

**Additional context**
mqtt message:
yicam/status/offline

dskvr, 2023-10-13:
This might be the problem I am having. However, deactivating "Disable Cloud" has no effect.

franciscot, 2023-11-14:
I can confirm: In MQTT QoS is always in "0".

franciscot, 2023-12-19:
Testing with [0.4.1d](https://github.com/alienatedsec/yi-hack-v5/releases/tag/0.4.1d) 
now it's working the state (online/offline) in mqtt, with or without "disable cloud".

closed 2023-12-19 by franciscot