alienatedsec/yi-hack-v5 #311: Cannot flash on Yi Home 1080 48US

bqtphan, 2023-11-09:
I've tried flashing on two of these cameras, but both get stuck with a yellow light for over 20 mins before I gave up. However, I was successful with V4.

I have the 48US. 
- I formatted a 2GB microsd card
- Dropped the files in (home_y20, rootfs_y20, yi-hack-v5)
- Plug in power
- Yellow flashing light
- Constant yellow flashing light

Can anyone help me with this?

alienatedsec, 2023-11-09:
@bqtphan It looks like the files are corrupted - please can you redownload and try again? Use this version whenever possible https://github.com/alienatedsec/yi-hack-v5/releases/tag/0.3.8

bqtphan, 2023-11-09:
YOU'RE RIGHT!
Maybe the extraction using WinZip messed it up. I extracted using CMD instead.
I'm embarrassed.

closed 2023-11-09 by bqtphan