alienatedsec/yi-hack-v5 #312: Rootfs corruption after a few months

AnthraX1, 2023-11-10:
**Describe the bug**
The rootfs on yi dome 1080p camera would get corrupted and stuck in a yellow light reboot loop. Can be reset but won't connect to wifi. Log says no wifi dev and wpa_supplicant crashed.

The only way to fix is to flash with an older version of rootfs_h20 and home_h20 files then load the sdcard with yi-hack-v5 0.38 files again. Flashing home_h20 alone won't solve the problem. 

**To Reproduce**
Steps to reproduce the behaviour:
All two of my long running yi dome 1080p cameras had the same problem after maybe 2 months.

**Expected behaviour**
Camera doesn't crash

**Screenshots**
None

**Set Up Details (please complete the following information):**
 - Firmware Version: 0.38
 - Device: YI Dome 1080p
 - Model: 45US

**Technical Details**
It's exactly like this issue: https://github.com/alienatedsec/yi-hack-v5/issues/204

alienatedsec, 2023-11-20:
@AnthraX1 I get closer to the latest 'wpa_supplicant' release - I found that different cameras have different versions. Please can you share which version is on your camera? Ideally, run the `wpa_supplicant` command from the console and share the content here.

alienatedsec, 2023-11-21:
@AnthraX1 New release with the latest `wpa_supplicant`

https://github.com/alienatedsec/yi-hack-v5/releases/tag/0.4.1c

K13PTo, 2023-11-30:
I've been having the same issue but cannot go beyond version 0.4.1b to see if it resolves  for me. If i try versions 0.4.1c and 0.4.1d, i go into boot loop.  Currently at 4.0

Firmware Version 	0.4.0
Base Version 	1.9.2.0I_201812141405
Model Suffix 	yi_dome_1080p
Hardware ID 	65US
wpa_supplicant v2.0-devel

AnthraX1, 2023-12-05:
> @AnthraX1 I get closer to the latest `wpa_supplicant` release - I found that different cameras have different versions. Please can you share which version is on your camera? Ideally, run the `wpa_supplicant` command from the console and share the content here.

I was using 0.40, not sure what version was in it. Isn't it compiled in and have binary file in yi-hack-v5/wpa/?

AnthraX1, 2023-12-05:
for the new firmware, I'm stuck at the Retrieving Pairing Status page on the app after the cam successfully connected to wifi. the last step in the app just times out while the camera is actually connected to the network.

mariosc, 2023-12-18:
the issue is not fixed with 0.4.1d. I thought it was ok but the multi reboot issue came back on. The only way to get the camera to work again is to run the yi-hack-v4 first, then run the v5 scripts.

alienatedsec, 2023-12-19:
> the issue is not fixed with 0.4.1d. I thought it was ok but the multi reboot issue came back on. The only way to get the camera to work again is to run the yi-hack-v4 first, then run the v5 scripts.

I confirm the same - both of my cameras are in the loop after reboot - I am investigating now.

RidingTheSpiral, 2023-12-19:
I had the same problem twice in a week. First time with the last beta version, to restore it I've flashed back the stock firmware. Second time I didn't update and it boot looped with 038 beta. Now I've restore it with 037 beta (like someone suggested in the other discussion), let's see how it goes! I've keept all there was on the sdcard, there some log that can be useful??

g1za, 2024-01-22:
today after a power outage my yi dome camera went into a boot loop.
I'm now trying to re-flash the latest base firmware and then update to the latest pre-release.

g1za, 2024-01-22:
no luck. the files seem to be deployed correctly on the SD card but it still goes into boot loop.
I upload what seems to contain some log files and now I'm trying to go back to stock firmware.
[log_no_wifi_dev.tar.gz](https://github.com/alienatedsec/yi-hack-v5/files/14010081/log_no_wifi_dev.tar.gz)

alienatedsec, 2024-01-22:
> the files seem to be deployed correctly on the SD card but it still goes into boot loop.

@g1za please remember to use FAT32

g1za, 2024-01-22:
@alienatedsec always

alienatedsec, 2024-02-04:
Please follow and test if needed https://github.com/alienatedsec/yi-hack-v5/releases/tag/0.4.1e

alienatedsec, 2024-02-11:
So I need to rework `busybox`, strip it of some tools, and move them to the `/tmp/sd/yi-hack-v5/.` folder.

Simply, it is too big and is unlikely to make cameras work with as little as 32kB space on the `rootfs` partition.

#291 is another discussion covering this subject in many directions - I'll start now and will make you aware of any prorgess

alienatedsec, 2024-02-12:
Some positive news. I found the `rootfs` partition needs around 80KB of free space to operate. I managed to increase the free space up to 124KB. The `Use%` increases up to around 95% over time and drops to 90% after a while. It means that some processes are still writing onto the `/dev/root` filesystem.

If that will work fine for the rest of the day, and I'll manage the other issue with the wget on the SD card, it will be released as the final baseline very soon.

```
# df -h
Filesystem                Size      Used Available Use% Mounted on
/dev/root                 1.3M      1.1M    124.0K  90% /
tmpfs                    15.0M      4.0K     15.0M   0% /dev
/dev/mtdblock5           12.7M      9.1M      3.6M  72% /home
tmpfs                    16.0M      1.9M     14.1M  12% /tmp
/dev/mmcblk0p1            7.4G     82.1M      7.3G   1% /tmp/sd
/dev/mmcblk0p1            7.4G     82.1M      7.3G   1% /tmp/sd/yi-hack-v5/www/record
```

Mentioning other issues for awareness #338

alienatedsec, 2024-02-13:
The new release `0.4.1h` should resolve the boot loop issue.

alienatedsec, 2024-02-19:
This is fixed in the latest pre-release - I appreciate it could take weeks or months to reproduce, but then please raise another issue if needed

closed 2024-02-19 by alienatedsec

ad-on-is, 2024-10-12:
I'm having exactly the same issue with 0.4.1. Camera works for a day or two, and then suddenly starts bootlooping.

Hack (SD) Firmware Version 	0.4.1
Baseline Firmware Version 	0.4.1
Base Version 	1.8.7.0F_xxx
Model Suffix 	yi_home
Hardware ID 	47US
Serial Number 	Y9FYxxxx

alienatedsec, 2024-10-12:
> I'm having exactly the same issue with 0.4.1. Camera works for a day or two, and then suddenly starts bootlooping.
> 
> 
> 
> Hack (SD) Firmware Version 	0.4.1
> 
> Baseline Firmware Version 	0.4.1
> 
> Base Version 	1.8.7.0F_xxx
> 
> Model Suffix 	yi_home
> 
> Hardware ID 	47US
> 
> Serial Number 	Y9FYxxxx

@ad-on-is this hasn't been an issue for several months now. I guess without a terminal output we cannot find what would be the problem. Furthermore, the issue with boot looping was partition-related. The affected partition has been modified, so it has enough space for camera operations.

ad-on-is, 2024-10-12:
@alienatedsec Unfortunately I can't SSH into the cam when it happens, since it doesn't connect to the wifi.

Is there a way to get to the latest logs when this happens?

Also, I've increased the SD size to 30%, since I read in other threads that it helped for some people.